espressif/arduino-esp32 #1467: ESPNOW Duplex Mode

Mahmoudomar91, 2018-06-04:
### Hardware:
Board:							?ESP32 Dev Module?
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
I am trying to send and receive data through ESPNOW between two ESP32 boards, lets call them ESP1 & ESP2. 
The thing is that if i want to send data from ESP1 to ESP2 i have to initialize ESP1 as master and ESP2 as slave and when i want to get the reply from the first transmission I set ESP2 as master and ESP1 as a slave and so on. 
so maybe sometimes a timing issue occurs when the two devices are both set as masters or as slaves at the same time. This causes a problem as they both halt and wait for data. 
Is there a way to make this process easier or make both devices master and slave at the same time so that they can send and receive data simultaneously ?

lbernstone, 2018-06-05:
There is nothing preventing slaves from sending data.  You will need to modify the code to have a send function.  https://esp-idf.readthedocs.io/en/latest/api-reference/wifi/esp_now.html

Mahmoudomar91, 2018-06-05:
Then how can i get the peer information to send the data from the slave to the master?
the slave is configured as AP so it has SSID that the master can search for. but the master doesn't have a SSID.

lbernstone, 2018-06-05:
Sorry, I don't know.  I probably should have kept my mouth shut.

eg-astrouka, 2018-06-05:
Slave can send the data to a predefined MAC address of the master. SSID is not used in this case. You can set the master's MAC using the function _esp_base_mac_addr_set_

Mahmoudomar91, 2018-06-05:
@eg-astrouka  then should i set the wifi mode to (WIFI_AP_STA) or (WIFI_AP)??

eg-astrouka, 2018-06-05:
@Mahmoudomar91 for the master it works on my side like that
```
uint8_t new_mac[8] = {0x30,0xAE,0xA4,0x11,0x11,0x11};
esp_base_mac_addr_set(new_mac);
WiFi.mode(WIFI_STA);
```
Slave has `WiFi.mode(WIFI_AP_STA);`
But if you want to test it with the latest arduino-esp32 version, take a look into #1425 issue

Mahmoudomar91, 2018-06-05:
@eg-astrouka  thanks for the code i tried it and it worked. Also i saw the issue you mention and i wrote its solution. 
The thing is i still don't know how to make the slave send data. could you please share the code lines with me. 
All the examples i can find use the esp-idf  and for ESP8266 but i am using Arduino IDE to program ESP32 so most code doesn't work.

eg-astrouka, 2018-06-06:
@Mahmoudomar91 slave code looks like
```
esp_now_peer_info_t slave;
uint8_t mac_master[] = {0x30,0xae,0xa4,0x11,0x11,0x11};
void InitESPNow() 
{
  if (esp_now_init() == ESP_OK)
    Serial.println(F("ESPNow Init Success"));
  else 
  {
    Serial.println(F("ESPNow Init Failed"));
    ESP.restart();
  }
  
  memset(&slave, 0, sizeof(slave));
  for (int i = 0; i < 6; ++i)
    slave.peer_addr[i] = (uint8_t)mac_master[i];
  slave.channel = CHANNEL; // pick a channel
  slave.encrypt = 0; // no encryption
  const esp_now_peer_info_t *peer = &slave;
  const uint8_t *peer_addr = slave.peer_addr;
  esp_err_t addStatus = esp_now_add_peer(peer);
}

void sendData(uint8_t* data, uint8_t data_len) 
{
  const uint8_t *peer_addr = slave.peer_addr;
  esp_err_t result = esp_now_send(peer_addr, data, data_len);
...
}
```
Slave and master should use the same CHANNEL

lbernstone, 2018-06-06:
If you have solved your issue, please close it.

Mahmoudomar91, 2018-06-08:
@lbernstone  well i haven't tried the code yet. but i will close it and if any issue happened i will reopen it

closed 2018-06-08 by Mahmoudomar91

Jerteddybear, 2019-04-30:
Hey @Mahmoudomar91, where you able to work this out? I want to solve the same problem (having ESP1 and ESP2 send data both ways) but I'm having a hard time understanding what to do from the code pasted here. 

I'm having no trouble with the classic Master to Slave approach; having one ESP32 with an IR sensor communicating to another ESP32 with a LED. But what I want is for both to have an IR sensor and a LED so they can communicate both ways.

Let me know!