espressif/arduino-esp32 #1471: ESP32 DevKit V4 dont run after power cycle

ivanbar, 2018-06-05:
Board:							       ESP32 Dev Module V4
Core Installation/update date:			4/06/2018
IDE name:							Arduino IDE  1.8.3
Flash Frequency:					       80Mhz
Upload Speed:						115200

Hello,

I ve bought several ESP32 devkit V4 from expressif, and I have the same problem:

After a power cycle, the program STOPS.

I need to reset to re start program execution.

Same problem in Windows 10 and Linux, tried several upload velocities, and frecuencies.

Only if I put GPIO 12 to GND and GPIO 2 to 3,3V  program runs ok after power cycle.

Any idea to solve this ??

Thank you very much.

### Sketch:

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(26, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(26, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(26, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

### Debug Messages:
```
No Error is shown, because if I start serial comunication, reset is done, and program runs ok.

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13088
entry 0x40078a5c

```

lbernstone, 2018-06-05:
https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#note-about-gpio12

ivanbar, 2018-06-07:
I have esp32devkit V2 and has not this problem.

Thanks for your reply, but i would like to know if:

A)This problem is a bug in Esp32devkit v4 board form espressif ??
B) Is there a software solution, diferent to modify pin connections GPIO2 and GPIO 12 ??

Thank you

lbernstone, 2018-06-07:
This is a physical bug.  You will need to look at the schematic to determine if you have this problem.  I do not know if your particular board has this issue, but many boards using SD cards present the problem.
Burning efuses is a software solution.  How would you change pin connections through software?  There are physical wires attached there, which affect the chip behavior.

ivanbar, 2018-06-08:
Hello Ibernstone!

thanks for the reply. We are not using any SD card. We have just bought several units of ESP32 DevKit v4 directly from ESPRESSIF company.

Our issue is that after uploading the program using Arduino IDE and, then, esptool.py, our program runs perfectly. The problem comes when we unplug the power provider (micro USB) and plug in it again. After this action, the program does not executes anymore and to force it to run, we have to press the EN button (reset) to make it work again. 

As we are using these boards for a project, we can not manually press the EN button after each re-power of the board. So we need a solution for that. If you consider it is a physical bug, is our only solution ask to the retailer to change us the product?
Thanks for your further insights on that.

BR,

ivanbar, 2018-06-08:
Solved.

Its a hardware bug. 

You must  remove C15 from esp32devkit V4.

lbernstone, 2018-06-08:
Please close the issue.

virat92, 2018-09-06:
i am new in ESP32 
in ESP32 DevkitC V4 is not showing in Arduino IDE
please help
i am using Arduino 1.8.6
ESP Bored installation date-06/09/2018

nietaki, 2021-12-25:
I'm having similar issues to @ivanbar and it seems like [his comment](https://github.com/espressif/arduino-esp32/issues/1471#issuecomment-395713280) is referring to this issue here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc.html#note-on-c15

Unfortunately the board I'm using already has the C15 capacitor removed / not installed so that's not a possible fix for me.

On the other hand I am using an SD card with the board, and I don't yet understand how this impacts the situation.

nietaki, 2021-12-31:
Update: In my case one working solution was this: https://github.com/espressif/arduino-esp32/issues/1274#issuecomment-643171871

From what I understand the problem is related to the signal timing on IO0 and EN pins - EN goes to +3V3 too fast and causes an undesired state of the microcontroller (not sure what state exactly). Adding a capacitor between EN and GND slows the rise of the voltage on the EN pin and rectifies the situation, or at least that's how I'm interpreting the situation

bantugam, 2022-03-21:
See also for a particular solution for ESP32-D1-R32
https://github.com/espressif/arduino-esp32/issues/1274#issuecomment-1072978693