espressif/arduino-esp32 #1473: ledcAttachPin ledcSetup ledcWrite. It is not normal when pin is greater than 6

dtboy1995, 2018-06-06:
```c++
uint8_t ledRRight = 27;
uint8_t ledGRight = 14;
uint8_t ledBRight = 12;

uint8_t ledRLeft = 33;
uint8_t ledGLeft = 25;
uint8_t ledBLeft = 26;

uint8_t ledRBottom = 23;
uint8_t ledGBottom = 22;
uint8_t ledBBottom = 32;

void setup()
{
  Serial.begin(9600);
  delay(10);

  ledcAttachPin(ledRRight, 7);
  ledcAttachPin(ledGRight, 8);
  ledcAttachPin(ledBRight, 9);

  ledcAttachPin(ledRLeft, 4);
  ledcAttachPin(ledGLeft, 5);
  ledcAttachPin(ledBLeft, 6);

  ledcAttachPin(ledRBottom, 1);
  ledcAttachPin(ledGBottom, 2);
  ledcAttachPin(ledBBottom, 3);

  ledcSetup(1, 12000, 8);
  ledcSetup(2, 12000, 8);
  ledcSetup(3, 12000, 8);

  ledcSetup(4, 12000, 8);
  ledcSetup(5, 12000, 8);
  ledcSetup(6, 12000, 8);

  ledcSetup(7, 12000, 8);
  ledcSetup(8, 12000, 8);
  ledcSetup(9, 12000, 8);
}
void loop()
{
  delay(2000);
  ledcWrite(7, 255);
  ledcWrite(8, 0);
  ledcWrite(9, 255);
  delay(2000);
  ledcWrite(7, 0);
  ledcWrite(8, 0);
  ledcWrite(9, 255);
  delay(2000);
  ledcWrite(7, 10);
  ledcWrite(8, 200); 
  ledcWrite(9, 255);
}

```
I test:
**
The led lights on pin foot 1, 2, 3, 4, 5 and 6 all work normally, but the led lights working on 7, 8, 9, 10, 11, 12, 13, 14 and 15 do not display normally
**

lbernstone, 2018-06-06:
The is a [pinmap](https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap) on the front page.  You cannot use the gpio pins attached to the SPI flash.

beegee-tokyo, 2018-06-06:
And in the [Wiki](https://desire.giesecke.tk/index.php/2018/01/30/esp-wroom-32-pin-definitions/) is a description of all pins and their usage.

dtboy1995, 2018-06-07:
@lbernstone  @beegee-tokyo  thank you very much.
I am very sorry for not having read the document in detail. Thank you for your help

lbernstone, 2018-06-07:
Please close the issue.

closed 2018-06-07 by dtboy1995