espressif/arduino-esp32 #1474: hey i am switching frequency between 240 MHZ and 2 MHZ but i am unable to find how to adjust my baudrate

usmanshahid001, 2018-06-06:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

SoftwareGuy, 2018-06-07:
Baudrate for what? Serial?

timkoers, 2018-06-09:
You can use Serial.begin(baudrate) 
You should check the datasheet of the serial to USB controller, if you have one, for the baudrate extremes.

stickbreaker, 2019-05-19:
code to support CPU frequency changes has been implemented since V1.0.2

closed 2019-05-19 by stickbreaker