espressif/arduino-esp32 #1484: Slow RTT on Ping to ESP32

janneslr, 2018-06-11:
Hi, is there any reason why I have such a slow Round Trip Time on a ping to my ESP32.  It varies from 1ms to 700ms and sometime timeouts.  Anyone else with this problem?
Reply from 10.168.3.181: bytes=32 time=119ms TTL=255
Reply from 10.168.3.181: bytes=32 time=25ms TTL=255
Reply from 10.168.3.181: bytes=32 time=67ms TTL=255
Reply from 10.168.3.181: bytes=32 time=87ms TTL=255
Reply from 10.168.3.181: bytes=32 time=103ms TTL=255
Reply from 10.168.3.181: bytes=32 time=102ms TTL=255
Reply from 10.168.3.181: bytes=32 time=115ms TTL=255
Reply from 10.168.3.181: bytes=32 time=24ms TTL=255
Reply from 10.168.3.181: bytes=32 time=43ms TTL=255
Reply from 10.168.3.181: bytes=32 time=49ms TTL=255
Reply from 10.168.3.181: bytes=32 time=63ms TTL=255
Reply from 10.168.3.181: bytes=32 time=84ms TTL=255
Reply from 10.168.3.181: bytes=32 time=112ms TTL=255
Reply from 10.168.3.181: bytes=32 time=94ms TTL=255
Reply from 10.168.3.181: bytes=32 time=142ms TTL=255
Reply from 10.168.3.181: bytes=32 time=21ms TTL=255

See the averages:
Approximate round trip times in milli-seconds:
    Minimum = 2ms, Maximum = 1037ms, Average = 64ms

If I compare to my ESP8266:
Reply from 10.168.3.114: bytes=32 time=4ms TTL=128
Reply from 10.168.3.114: bytes=32 time=13ms TTL=128
Reply from 10.168.3.114: bytes=32 time=6ms TTL=128
Reply from 10.168.3.114: bytes=32 time=3ms TTL=128
Reply from 10.168.3.114: bytes=32 time=3ms TTL=128
Reply from 10.168.3.114: bytes=32 time=4ms TTL=128
Reply from 10.168.3.114: bytes=32 time=8ms TTL=128
Reply from 10.168.3.114: bytes=32 time=5ms TTL=128
Reply from 10.168.3.114: bytes=32 time=17ms TTL=128
Reply from 10.168.3.114: bytes=32 time=4ms TTL=128

Approximate round trip times in milli-seconds:
    Minimum = 3ms, Maximum = 17ms, Average = 6ms

enly1, 2018-06-12:
Just started playing with an ESP32 and have similar performance ( 50 cm from wifi router )

64 bytes from 192.168.1.97: icmp_seq=57 ttl=255 time=194 ms
64 bytes from 192.168.1.97: icmp_seq=58 ttl=255 time=115 ms
64 bytes from 192.168.1.97: icmp_seq=59 ttl=255 time=36.0 ms
64 bytes from 192.168.1.97: icmp_seq=60 ttl=255 time=266 ms
64 bytes from 192.168.1.97: icmp_seq=61 ttl=255 time=184 ms
64 bytes from 192.168.1.97: icmp_seq=62 ttl=255 time=102 ms
64 bytes from 192.168.1.97: icmp_seq=63 ttl=255 time=22.2 ms
64 bytes from 192.168.1.97: icmp_seq=64 ttl=255 time=250 ms
64 bytes from 192.168.1.97: icmp_seq=65 ttl=255 time=170 ms
64 bytes from 192.168.1.97: icmp_seq=66 ttl=255 time=89.6 ms
64 bytes from 192.168.1.97: icmp_seq=67 ttl=255 time=319 ms
64 bytes from 192.168.1.97: icmp_seq=68 ttl=255 time=245 ms
64 bytes from 192.168.1.97: icmp_seq=69 ttl=255 time=164 ms
64 bytes from 192.168.1.97: icmp_seq=70 ttl=255 time=81.0 ms
64 bytes from 192.168.1.97: icmp_seq=71 ttl=255 time=311 ms
64 bytes from 192.168.1.97: icmp_seq=72 ttl=255 time=230 ms
64 bytes from 192.168.1.97: icmp_seq=73 ttl=255 time=154 ms
64 bytes from 192.168.1.97: icmp_seq=74 ttl=255 time=71.6 ms
64 bytes from 192.168.1.97: icmp_seq=75 ttl=255 time=299 ms
64 bytes from 192.168.1.97: icmp_seq=76 ttl=255 time=220 ms
64 bytes from 192.168.1.97: icmp_seq=77 ttl=255 time=141 ms
64 bytes from 192.168.1.97: icmp_seq=78 ttl=255 time=62.3 ms
64 bytes from 192.168.1.97: icmp_seq=79 ttl=255 time=288 ms
64 bytes from 192.168.1.97: icmp_seq=80 ttl=255 time=209 ms
64 bytes from 192.168.1.97: icmp_seq=81 ttl=255 time=131 

Running Simple Wifi Server example sketch on Arduino.

1163167506, 2018-06-13:
ye  
You and I have the same problem.

I and espressif official trench admit that esp32 has this problem through them.

But I don't have time to wait for the official solution.

Can anyone offer good advice

1163167506, 2018-06-19:
I solved this problem
Because esp32 enters the power saving mode by default, just set it up.

WiFi.mode (WIFI_STA);

Esp_wifi_set_ps (WIFI_PS_NONE);

WiFi.begin (ssid3, password3);

lbernstone, 2018-06-21:
If it is fixed, please close the issue.

me-no-dev, 2018-06-21:
it worked but it consumes more power ;) in essence you are keeping the esp awake and wifi on all of the time

lbernstone, 2018-07-29:
If you use WiFi.setSleep(false), you will need no additional libraries.

Uksa007, 2018-07-31:
I noticed I have this issue also, confirmed that turning off the power save fixes the issue, but it seems like a bit of a hack. 

I note that on my ESP8266 the Wifi power save is automatically turned off when the ESP8266 is communitcing over the WIFI and then turned back on again, which make sense to me, and is how most Wifi devices work. 
Why does the ESP32 WIFI not fuction in this way?

vinot, 2019-07-12:
Hi all,

thank you a lot for the solution, was frustrating those 200ms ping...

yrs

toni

dao89, 2019-07-13:
> I noticed I have this issue also, confirmed that turning off the power save fixes the issue, but it seems like a bit of a hack.
>
> I note that on my ESP8266 the Wifi power save is automatically turned [...]

One year past, but ESP32 still very slow when power saving not disabled. When disabled, it consumes twice more power! esp8266 doen't have such problem... It is very sad.

closed 2019-09-25 by stale[bot]

YordanYanakiev, 2020-01-13:
It turns out that **this resoled actually ALL of the issues connected to the WebServer random connection drops**. So - link this topic to all of such issues. It most likely if not solve totally it will at lease easy the problem.

LockonS, 2021-09-03:
> If you use WiFi.setSleep(false), you will need no additional libraries.

WORKS LIKE A CHARM! Thannk you so much.

mbratch, 2022-03-28:
The best case I'm getting from pinging 3 different ESP32's on my network is in the low 100's of ms. Worst case, it's taking a few seconds! Everything else on my network response within 20ms. There's something dreadfully wrong with the ESP32 response.

I have wifi power save disabled.

I'm using ESP-IDF 4.4.

mbratch, 2022-03-28:
I am running ESP-IDF 4.4 and I have Wifi powersave turned off. Pings average about 200ms. Pretty slow.

mbratch, 2022-03-29:
The ping response on my ESP32 running with ESP-IDF 4.4 seems to be relatively long. Unclear why this is the case.
[image]

gonzabrusco, 2022-04-29:
The correct way to do this is `WiFi.setSleep(WIFI_PS_NONE);`

The wifi energy saving options are:
```
 WIFI_PS_NONE,        /**< No power save */
 WIFI_PS_MIN_MODEM,   /**< Minimum modem power saving. In this mode, station wakes up to receive beacon every DTIM period */
 WIFI_PS_MAX_MODEM,   /**< Maximum modem power saving. In this mode, interval to receive beacons is determined by the listen_interval parameter in wifi_sta_config_t */
```

mbratch, 2022-04-29:
> The correct way to do this is `WiFi.setSleep(WIFI_PS_NONE);`

That's using the Arduino WiFi library. I am not using Arduino, but rather ESP-IDF.

For ESP-IDF, the correct call is `esp_wifi_set_ps(WIFI_PS_NONE);` This is what I am using and I still observe slower ping responses than I would have expected.

gonzabrusco, 2022-04-30:
> 

> That's using the Arduino WiFi library. I am not using Arduino, but rather ESP-IDF.
>
> For ESP-IDF, the correct call is `esp_wifi_set_ps(WIFI_PS_NONE);` This is what I am using and I still observe [...]

Well, this is the arduino-esp32 repository, thus my comment is directed to Arduino users. If you have a problem with IDF, you better submit an issue in the IDF repository (https://github.com/espressif/esp-idf)

mbratch, 2022-04-30:
@gonzabrusco ha indeed. I've been working with several esp32 related repositories lately and got my wires crossed a little! My apologies.

gonzabrusco, 2022-04-30:
> @gonzabrusco ha indeed. I've been working with several esp32 related repositories lately and got my wires crossed a little! My apologies.
> 
> Under the hood, as it were, the Arduino library probably calls the esp-idf API. Definitely not an Arduino specific issue

No need to apologize! Yes, actually Arduino calls the IDF function you menctioned. https://github.com/espressif/arduino-esp32/blob/ce2cd111a150ca87ab733ae434d77962a7efe96f/libraries/WiFi/src/WiFiGeneric.cpp#L1162

But now that I see, the function that accepts a boolean (as mentioned previously) actually exists too! https://github.com/espressif/arduino-esp32/blob/ce2cd111a150ca87ab733ae434d77962a7efe96f/libraries/WiFi/src/WiFiGeneric.cpp#L1148
But it let's you only chose between two types of energy savings (off or minimal).