espressif/arduino-esp32 #1487: WifiClientSecure connect(IPAddress ip, hostport) not working !!

iamneo2416, 2018-06-12:
Hi all, please help.

All WifiClientSecure searched on internet sample code use host name in connect, but no one single use IPAddress ipv4 to do connect. Recently, I built a web server(**apache2**) + **php** + **mysql** + phpMyAdmin (ubuntu 16.04 apache2 w/TSL). I can use **https://192.168.1.210/phpMyAdmin** to login to mysql web management page successfully, BUT I cannot use following code to do connect. PLEASE HELP. Can anyone give me a working sample? I can also use command line to do 
**C:\> telnet 192.168.1.210 443**   that also works. means 443 is listening.

P.S. I followed exactly the same steps to built this web server.
[https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-apache-in-ubuntu-16-04](https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-apache-in-ubuntu-16-04)

======================================
```
const char* host = "192.168.1.210";
IPAddress host_ip(ipStrToNum(host ));
Wificlientsecure client;

wifi.begin(user_name, passwd);
client.connect(host_ip, 443);  //<== this line always return failed !

int WiFiClientSecure::connect(IPAddress ip, uint16_t port)
{
    return connect(ip, port, _CA_cert, _cert, _private_key);
}
I found this API in here line(88).
[https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp)

....
I saw there should be a api for wificlientsecure.

// I found this utility function working great to transfer char ipv4 to uint32.
uint32_t ipStrToNum(const char* ipStr)
{
  const int SIZE_OF_NUMS = 4;
  union
  {
    uint8_t bytes[SIZE_OF_NUMS];  // IPv4 address
    uint32_t dword;
  } _address;
  _address.dword = 0; // clear return

  int i = 0;
  uint8_t num = 0; // start with 0
  while ((*ipStr) != '\0')
  {
    // while not end of string
    if ((*ipStr == '.') || (*ipStr == ','))
    {
      // store num and move on to next position
      _address.bytes[i++] = num;
      num = 0;
      if (i>=SIZE_OF_NUMS)
      {
        break; // filled array
      }
    }
    else
    {  
      if (*ipStr != ' ')
      {      // skip blanks
        num = (num << 3) + (num << 1); // *10 = *8+*2
        num = num +  (*ipStr - '0');
      }  
    }  
    ipStr++;
  }  
  if (i<SIZE_OF_NUMS)
  {
    // store last num
    _address.bytes[i++] = num;
  }
  return _address.dword; 
}  
```

iamneo2416, 2018-06-12:
Here is my actual file attached. Please someone to help me. I need to create a web server within INTRANET not in INTERNET. Therefore, I use private IP stack (192.168.1.x)

[WiFiClientSecureTest.txt](https://github.com/espressif/arduino-esp32/files/2093079/WiFiClientSecureTest.txt)

iamneo2416, 2018-06-12:
I tried 5 sample code original from host based like "google.com" , "api.github.com" , .... all working fine with connect(ipv4, 443), but once I changed to connect( IPAddress, 443) and all get fails !

iamneo2416, 2018-06-12:
here is another sample code I tried. I only add ipStrToNum function() and change host to host_ipv4 and put that into connect(ipv4, 443).

=============================
```
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = "mywifissid";
const char* password = "password";

//const char* host = "api.github.com";  <== use this to connect. it works !
const char* host = "192.168.1.210";  <== change to this. it does not work !
const int httpsPort = 443;

// Use web browser to view and copy
// SHA1 fingerprint of the certificate
const char* fingerprint = "CF 05 98 89 CA FF 8E D8 5E 5C E0 C2 E4 F7 E6 C3 C7 50 DD 5C";

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  // Use WiFiClientSecure class to create TLS connection
  WiFiClientSecure client;
 
  IPAddress host_ipv4(ipStrToNum(host));
  Serial.print("connecting to ");
  Serial.println(host_ipv4);
  
  if (!client.connect(host, httpsPort))
  {
    Serial.println(">>> connection failed");
    return;
  }

  if (client.verify(fingerprint, host))
  {
    Serial.println("certificate matches");
  }
  else
  {
    Serial.println("certificate doesn't match");
  }

  String url = "/repos/esp8266/Arduino/commits/esp8266/status";
  Serial.print("requesting URL: ");
  Serial.println(url);

  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: BuildFailureDetectorESP8266\r\n" +
               "Connection: close\r\n\r\n");

  Serial.println("request sent");
  while (client.connected())
  {
    String line = client.readStringUntil('\n');
    if (line == "\r")
    {
      Serial.println("headers received");
      break;
    }
  }
  String line = client.readStringUntil('\n');
  if (line.startsWith("{\"state\":\"success\""))
  {
    Serial.println("esp8266/Arduino CI successfull!");
  }
  else
  {
    Serial.println("esp8266/Arduino CI has failed");
  }
  Serial.println("reply was:");
  Serial.println("==========");
  Serial.println(line);
  Serial.println("==========");
  Serial.println("closing connection");
}

uint32_t ipStrToNum(const char* ipStr)
{
  const int SIZE_OF_NUMS = 4;
  union
  {
    uint8_t bytes[SIZE_OF_NUMS];  // IPv4 address
    uint32_t dword;
  } _address;
  _address.dword = 0; // clear return

  int i = 0;
  uint8_t num = 0; // start with 0
  while ((*ipStr) != '\0')
  {
    // while not end of string
    if ((*ipStr == '.') || (*ipStr == ','))
    {
      // store num and move on to next position
      _address.bytes[i++] = num;
      num = 0;
      if (i>=SIZE_OF_NUMS)
      {
        break; // filled array
      }
    }
    else
    {  
      if (*ipStr != ' ')
      {      // skip blanks
        num = (num << 3) + (num << 1); // *10 = *8+*2
        num = num +  (*ipStr - '0');
      }  
    }  
    ipStr++;
  }  
  if (i<SIZE_OF_NUMS)
  {
    // store last num
    _address.bytes[i++] = num;
  }
  return _address.dword; 
}  

void loop() {
}
```
=============================

lbernstone, 2018-06-12:
Wow.  Way to overcomplicate things.  Set the IPAddress like so:
```
IPAddress host_ip = {192.168.1.210};
```
Then, set Core Debug Level to Verbose in the Tools menu.
```
#include <WiFiClientSecure.h>
void setup() {
  WiFi.begin();
  while (WiFi.status() != WL_CONNECTED) delay(500);
  delay(500);
  IPAddress host = {172,217,12,4}; //www.google.com
  WiFiClientSecure client;
  client.connect(host, 443);
}
void loop() {}
```

iamneo2416, 2018-06-13:
Let me try to use 192,168,1,210 and feedback to you in a minute !

iamneo2416, 2018-06-13:
Can you help me to see what's going on when I switch from google.com back to 192.168.1.210, it's still not working !!

I followed following link to set up my server. Not change anything.

[https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-apache-in-ubuntu-16-04] (https://www.digitalocean.com/community/tutorials/how-to-create-a-self-signed-ssl-certificate-for-apache-in-ubuntu-16-04)

=======================
```

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

String wifi = "hify";
String pass = "12345678";
void setup()
{
  Serial.begin(115200);
  
  WiFi.begin(wifi.c_str(), pass.c_str());
  while (WiFi.status() != WL_CONNECTED) { Serial.print("."); delay(500); }
  delay(500);
  //IPAddress host = {172,217,12,4}; //www.google.com; this working !
  IPAddress host = {192,168,1,210}; //www.google.com; this NOT working !
  WiFiClientSecure client;
  if(!client.connect(host, 443))
  {
    Serial.println(">>> not connect");
    return;
  }
  Serial.println("WiFiClientSecure connected !!");
}
void loop() {}
```
=======================

lbernstone, 2018-06-13:
Google will be a much better resource than me for this.  wget -d is a much better testing tool than a browser, which will hide lots of errors from you.

iamneo2416, 2018-06-14:
DEBUG output created by Wget 1.17.1 on linux-gnu.

Reading HSTS entries from /root/.wget-hsts
URI encoding = âUTF-8â
--2018-06-14 18:58:55--  http://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Caching localhost => ::1 127.0.0.1
Connecting to localhost (localhost)|::1|:80... connected.
Created socket 3.
Releasing 0x000055fed6214610 (new refcount 1).

---request begin---
GET / HTTP/1.1
User-Agent: Wget/1.17.1 (linux-gnu)
Accept: */*
Accept-Encoding: identity
Host: localhost
Connection: Keep-Alive

---request end---
HTTP request sent, awaiting response...
---response begin---
HTTP/1.1 302 Found
Date: Thu, 14 Jun 2018 10:58:55 GMT
Server: Apache/2.4.18 (Ubuntu)
Strict-Transport-Security: max-age=63072000; includeSubdomains
X-Frame-Options: DENY
X-Content-Type-Options: nosniff
Location: https://192.168.1.210/
Content-Length: 281
Keep-Alive: timeout=5, max=100
Connection: Keep-Alive
Content-Type: text/html; charset=iso-8859-1

---response end---
302 Found
Registered socket 3 for persistent reuse.
Parsed Strict-Transport-Security max-age = 63072000, includeSubDomains = true
Updated HSTS host: localhost:80 (max-age: 63072000, includeSubdomains: true)
URI content encoding = âiso-8859-1â
Location: https://192.168.1.210/ [following]
Skipping 281 bytes of body: [<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>302 Found</title>
</head><body>
<h1>Found</h1>
<p>The document has moved <a href="https://192.168.1.210/">here</a>.</p>
<hr>
<address>Apache/2.4.18 (Ubuntu) Server at localhost Port 80</address>
</body></html>
] done.
URI content encoding = None
--2018-06-14 18:58:55--  https://192.168.1.210/
**Connecting to 192.168.1.210:443... connected.**
Created socket 4.
Releasing 0x000055fed6235ff0 (new refcount 0).
Deleting unused 0x000055fed6235ff0.
Initiating SSL handshake.
**Handshake successful**; connected socket 4 to SSL handle 0x000055fed6236080
certificate:
  subject: emailAddress=andy.lofer@bakeme.com,CN=192.168.1.210,OU=MIS,O=BakeMe CO.,L=Singaphore,ST=Singaphore,C=SP
  issuer:  emailAddress=andy.lofer@bakeme.com,CN=192.168.1.210,OU=MIS,O=BakeMe CO.,L=Singaphore,ST=Singaphore,C=SP
ERROR: cannot verify 192.168.1.210's certificate, issued by âemailAddress=andy.lofer@bayestek.com,CN=192.168.1.210,OU=MIS,O=BakeMe CO.,L=Singaphore,ST=Singaphore,C=TWâ:
  **Self-signed certificate encountered.**
To connect to 192.168.1.210 insecurely, use `--no-check-certificate'.
Closed 4/SSL 0x000055fed6236080
Saving HSTS entries to /root/.wget-hsts

iamneo2416, 2018-06-14:
```
DEBUG output created by Wget 1.17.1 on linux-gnu.

Reading HSTS entries from /root/.wget-hsts
URI encoding = aUTF-8a
--2018-06-14 18:58:55--  http://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Caching localhost => ::1 127.0.0.1
Connecting to localhost (localhost)|::1|:80... connected.
Created socket 3.
Releasing 0x000055fed6214610 (new refcount 1).

---request begin---
GET / HTTP/1.1
User-Agent: Wget/1.17.1 (linux-gnu)
Accept: */*
Accept-Encoding: identity
Host: localhost
Connection: Keep-Alive

---request end---
HTTP request sent, awaiting response...
---response begin---
HTTP/1.1 302 Found
Date: Thu, 14 Jun 2018 10:58:55 GMT
Server: Apache/2.4.18 (Ubuntu)
Strict-Transport-Security: max-age=63072000; includeSubdomains
X-Frame-Options: DENY
X-Content-Type-Options: nosniff
Location: https://192.168.1.210/
Content-Length: 281
Keep-Alive: timeout=5, max=100
Connection: Keep-Alive
Content-Type: text/html; charset=iso-8859-1

---response end---
302 Found
Registered socket 3 for persistent reuse.
Parsed Strict-Transport-Security max-age = 63072000, includeSubDomains = true
Updated HSTS host: localhost:80 (max-age: 63072000, includeSubdomains: true)
URI content encoding = aiso-8859-1a
Location: https://192.168.1.210/ [following]
Skipping 281 bytes of body: [<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>302 Found</title>
</head><body>
<h1>Found</h1>
<p>The document has moved <a href="https://192.168.1.210/">here</a>.</p>
<hr>
<address>Apache/2.4.18 (Ubuntu) Server at localhost Port 80</address>
</body></html>
] done.
URI content encoding = None
--2018-06-14 18:58:55--  https://192.168.1.210/
Connecting to 192.168.1.210:443... connected.
Created socket 4.
Releasing 0x000055fed6235ff0 (new refcount 0).
Deleting unused 0x000055fed6235ff0.
Initiating SSL handshake.
Handshake successful; connected socket 4 to SSL handle 0x000055fed6236080
certificate:
  subject: emailAddress=andy.lofer@BakeMe.com,CN=192.168.1.210,OU=MIS,O=BakeMe CO.,L=Taipei,ST=Singaphore,C=SP
  issuer:  emailAddress=andy.lofer@BakeMe.com,CN=192.168.1.210,OU=MIS,O=BakeMe CO.,L=Taipei,ST=Singaphore,C=SP
ERROR: cannot verify 192.168.1.210's certificate, issued by aemailAddress=andy.lofer@BakeMe.com,CN=192.168.1.210,OU=MIS,O=BakeMe CO.,L=Taipei,ST=Singaphore,C=SPa:
  Self-signed certificate encountered.
To connect to 192.168.1.210 insecurely, use `--no-check-certificate'.
Closed 4/SSL 0x000055fed6236080
Saving HSTS entries to /root/.wget-hsts

```

iamneo2416, 2018-06-14:
what does that mean "The document has moved "https://192.168.1.210/" ??
Of cause, use WiFiClientSecure to connect 443 port with SSL, 

Did you find anything here? for logger ?

copercini, 2018-06-15:
@iamneo2416 please report this issue on ESP8266 forum, here is for ESP32

closed 2018-06-15 by copercini

brnyza, 2019-05-13:
Just include in your code:
client.setInsecure();

nospam2k, 2020-02-02:
No, you can't. setinsecure() is ESP8266 not ESP32

Sarcares, 2021-07-02:
Your application is trying to verify the certificate through a Certification Authority but, due to the fact that is self-signed, this operation fails.
You have to avoid this verification for having a positive answer from the web server.