espressif/arduino-esp32 #1488: Read spiffs crash in hw_timer

HelloDB, 2018-06-12:
### Hardware:
Board:ESP32 Dev Module
Core Installation/update date:20180612
IDE name:Platform.io
Flash Frequency:40Mhz
Upload Speed:115200

### Description:
Read spiffs crash in hw_timer

### Sketch:
```cpp

/*-------------------------------------------------------------------------------------------
	hw_timer test.

-------------------------------------------------------------------------------------------*/
#include <Arduino.h>

hw_timer_t *timer = NULL;

#include "FS.h"
#include "SPIFFS.h"

File g_file;

void IRAM_ATTR onTimer(void) {
	// Serial.println("test.");
	// Serial.println(g_file.size());
	Serial.println(g_file.position());
	// Serial.println(g_file.read());
	// if(g_file.available()){
	//     Serial.println(g_file.read());
	// }
}

void setup() {

	Serial.begin(115200);
	Serial.println("Start_>>");

	if (!SPIFFS.begin()) {
		Serial.println("SPIFFS initialisation failed!");
	}
	Serial.println("\r\nInitialisation done.");

	g_file = SPIFFS.open("/p1.jpg", "r");
	// g_file.seek(0);
	Serial.printf("size:%u\n", g_file.size());
	Serial.printf("position:%u\n", g_file.position());

	// Use 1st timer of 4 (counted from zero).
	// Set 80 divider for prescaler (see ESP32 Technical Reference Manual for more info).
	timer = timerBegin(0, 80, true);
	timerAttachInterrupt(timer, &onTimer, true);
	timerAlarmWrite(timer, 1000000, true);
	timerAlarmEnable(timer);
}

void loop() {
}

```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078a28
Start_>>

Initialisation done.
size:54990
position:0
abort() was called at PC 0x40083ec5 on core 1

Backtrace: 0x4008755c:0x3ffc0ab0 0x4008765b:0x3ffc0ad0 0x40083ec5:0x3ffc0af0 0x40083fe5:0x3ffc0b20 0x400ec527:0x3ffc0b40 0x400ec4dd:0x3ffc0b60 0x400e9916:0x3ffc0b80 0x400ec80d:0x3ffc0ba0 0x40080d72:0x3ffc0bc0 0x40080e99:0x3ffc0be0 0x4008160d:0x3ffc0c00 0x4000bfed:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078a28
Start_>>

Initialisation done.
size:54990
position:0
abort() was called at PC 0x40083ec5 on core 1

Backtrace: 0x4008755c:0x3ffc0ab0 0x4008765b:0x3ffc0ad0 0x40083ec5:0x3ffc0af0 0x40083fe5:0x3ffc0b20 0x400ec527:0x3ffc0b40 0x400ec4dd:0x3ffc0b60 0x400e9916:0x3ffc0b80 0x400ec80d:0x3ffc0ba0 0x40080d72:0x3ffc0bc0 0x40080e99:0x3ffc0be0 0x4008160d:0x3ffc0c00 0x4000bfed:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078a28
Start_>>

Initialisation done.
size:54990
position:0
...
```

liebman, 2018-06-12:
First guess would be that g_file.position() is not in IRAM so not callable from a timer/interrupt.....

HelloDB, 2018-06-12:
@liebman Thank you for your reply and found it feasible in 8266

lbernstone, 2018-06-12:
ESP32 != ESP8266.  There is a watchdog timer that will choke if an ISR takes too long.  Set a flag in your ISR, and process it in loop.
```
/*-------------------------------------------------------------------------------------------
	hw_timer test.

-------------------------------------------------------------------------------------------*/
#include <Arduino.h>

hw_timer_t *timer = NULL;

#include "FS.h"
#include "SPIFFS.h"

File g_file;
bool test_flag = false;

void IRAM_ATTR onTimer(void) {
  test_flag = true;
}

void setup() {

	Serial.begin(115200);
	Serial.println("Start_>>");

	if (!SPIFFS.begin()) {
		Serial.println("SPIFFS initialisation failed!");
	}
	Serial.println("\r\nInitialisation done.");

	g_file = SPIFFS.open("/p1.jpg", "r");
	// g_file.seek(0);
	Serial.printf("size:%u\n", g_file.size());
	Serial.printf("position:%u\n", g_file.position());

	// Use 1st timer of 4 (counted from zero).
	// Set 80 divider for prescaler (see ESP32 Technical Reference Manual for more info).
	timer = timerBegin(0, 80, true);
	timerAttachInterrupt(timer, &onTimer, true);
	timerAlarmWrite(timer, 1000000, true);
	timerAlarmEnable(timer);
}

void loop() {
  if (test_flag) {
    test_flag = false;
	// Serial.println("test.");
	// Serial.println(g_file.size());
	Serial.println(g_file.position());
	// Serial.println(g_file.read());
	// if(g_file.available()){
	//     Serial.println(g_file.read());
	// }
  }
}

closed 2019-08-15 by stale[bot]