espressif/arduino-esp32 #1489: WPA2 Enterprise PEAP-MSCHAPv2 not connect with static IP

andrefilipin, 2018-06-12:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			20180612
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
WPA2 Enterprise PEAP-MSCHAPv2 not connect with static IP(my AP doesn't have DHCP enabled), I Have other AP with same auth type and DHCP enabled, and it works normaly, please help

### Sketch:
```cpp
#include "esp_wpa2.h"
#include <WiFi.h>

const char *ssid = "***";
#define EAP_ID "user"
#define EAP_USERNAME "user"
#define EAP_PASSWORD "password"

void setup() {
    Serial.begin(115200);
    delay(10);

    
    IPAddress ip(10,21,123,142);   
    IPAddress gateway(10,21,122,1);   
    IPAddress subnet(255,255,254,0);
    IPAddress dns(10,21,71,19);
    
    WiFi.config(ip, gateway, subnet, dns);

    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.disconnect(true);

    esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EAP_ID, strlen(EAP_ID));
    esp_wifi_sta_wpa2_ent_set_username((uint8_t *)EAP_USERNAME, strlen(EAP_USERNAME));
    esp_wifi_sta_wpa2_ent_set_password((uint8_t *)EAP_PASSWORD, strlen(EAP_PASSWORD));
    esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT();
    esp_wifi_sta_wpa2_ent_enable(&config);
    
    WiFi.begin(ssid);    

    while (WiFi.status() != WL_CONNECTED) {
        delay(2000);
        Serial.print(".");        
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

void loop() {
    // Your code here.
}
```

### Debug Messages:
```
rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
[D][WiFiGeneric.cpp:304] _eventCallback(): Event: 2 - STA_START

Connecting to ****
.........................
```
nothing happen more

lbernstone, 2018-06-12:
Take a look at https://github.com/espressif/arduino-esp32/issues/1262

andrefilipin, 2018-06-13:
I removed the comment referenced  in #1262 and doesn't work, results:
`
Connecting to ****
.........................
`

closed 2019-08-15 by stale[bot]

bkrajendra, 2019-09-27:
@andrefilipin any update on this.
Im also working on ESP WPA2-Enterprise connectivity but no success.
I tried a lot with ESP-8266 also but getting error for static IP. Not sure about DHCP.
https://github.com/tzapu/WiFiManager/issues/309#issuecomment-535318990_