espressif/arduino-esp32 #1492: sta model problem

1163167506, 2018-06-13:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Setting esp32 as a sta mode and connecting the router, using the router /PC Ping esp32 to find the delay is especially large (the distance is very close, the quality of the signal is very good by printing). At 60ms, it is beyond my use, and I want to be able to reduce the delay. We also tested the esp8266 discovery (in the same environment, in the same environment, The demo procedure is almost the same) the delay is very stable and relatively low, less than 10 milliseconds. We want to know if esp32 is delayed as Sta mode.

Why does this happen?

### Sketch:
#include "WiFi.h"  
const char* ssid ="TP-LINK_BE5F";  
const char*password =  "test123456@";  
void setup()  
{  
  Serial.begin(115200);  
  delay(10);

  WiFi.mode(WIFI_STA);
  
  WiFi.begin(ssid, password);  
  while (WiFi.status()!= WL_CONNECTED) {  
    delay(500);  
    Serial.println("Connectingto WiFi..");  
  }  
  
  Serial.println("Connectedto the WiFi network");  
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP()); 
}  
void loop() { 
  delay(1000);
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());
  
  Serial.print("channel:");
  Serial.print(WiFi.channel());

  Serial.print("rssi:");
  Serial.print(WiFi.RSSI());
   
} 

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 

![image](https://user-images.githubusercontent.com/28136687/41328850-dbde7fca-6efc-11e8-8a51-4674909a3c7a.png)

![image](https://user-images.githubusercontent.com/28136687/41328855-e5bfc6de-6efc-11e8-83f8-63f141b45dc8.png)

```

1163167506, 2018-06-19:
I solved this problem
Because esp32 enters the power saving mode by default, just set it up.

WiFi.mode (WIFI_STA);

Esp_wifi_set_ps (WIFI_PS_NONE);

WiFi.begin (ssid3, password3);

lbernstone, 2018-06-21:
If solved, please close the issue.

closed 2019-08-15 by stale[bot]