espressif/arduino-esp32 #1494: Diferent behaviour if powered from 5V pin or from micro USB connector

ivanbar, 2018-06-13:
Board:							ESP32 Devkit V4
Core Installation/update date:	        11/jul/2018
IDE name:					        Arduino IDE 1.8.3
Flash Frequency:					40Mhz
Upload Speed:				        115200

Diferent behaviour if esp32devkit V4 powered from 5V pin or from micro USB connector.

Im controlling some ws2812B rgbs with the esp32, you must use especific libraries for esp32 and ws2812, but if I power the board from micro usb connector, all goes OK, but If I power the board from  5v pin, from a regulated source, RGBs goes wrong.

I control a 3 led power strip on GPIO 27.

PINOUT       Esp32devkit v4       5v Power source        WS2812 strip
                                      V5       to  +5VCC          to           VCC
                                     GND   to     GND            to          GND
                                     GPIO 27             To                       Din
Any idea ? Thank you.

Complete code example from: https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/blob/master/arduino-esp32/demo2/demo2.ino

### Sketch:

#include "esp32_digital_led_lib.h"
/*
 * Simplest possible example shows a single strand of NeoPixels. 
 */

strand_t pStrand = {.rmtChannel = 0, .gpioNum = 27, .ledType = LED_WS2812B_V3, .brightLimit = 32, .numPixels = 3,
   .pixels = nullptr, ._stateVars = nullptr};

int stepper = 0;
int colord = 0;

void setup()
{
  Serial.begin(115200);
  Serial.println("Initializing...");

  pinMode (27, OUTPUT);
  digitalWrite (27, LOW);

  if (digitalLeds_initStrands(&pStrand, 1)) {
    Serial.println("Init FAILURE: halting");
    while (true) {};
  }
  
  digitalLeds_resetPixels(&pStrand);
}

void loop()
{
    pStrand.pixels[stepper] = pixelFromRGBW(55, colord, 0, 0);

    stepper++;
    
    if(stepper > 240) {
      stepper = 0;
      colord += 10;
    }

    if(colord > 60) 
      colord = 0;
    
    digitalLeds_updatePixels(&pStrand);
}
```

### Debug Messages:
```
Cant obtain Degug messages because no serial communication is done.
```

stickbreaker, 2018-06-13:
@ivanbar Usually these symptoms point to a weak 5V power supply, When the ESP32 is booting, it can use 500mA of 3.3v.  So the onboard 3.3V regulator needs extra power, make sure your 5v supply can supply ~750mA to 1A.

Chuck.

fpovoski, 2019-01-05:
The cause of this problem appears that some ESP32 boards the USB chip is not powered from the Vin (VDD5V) pin (19) but from the VDDUSB with a diode between.  This is unlike the ESP8266 boards which are the opposite and is correct to enable the USB chip to be powered at 5v whether plugged into the USB or powered solely by the 5v input pin.  It would then cause DTR and RTS to be in an indeterminate voltage affecting EN and IO0 throwing the ESP into a reboot when the higher current surge happens during WIFI communication pulls the rail down. IT happens on an older Makerfocus ESP32 but I will try another ESP32 board to see if occurs there.  Maybe its been fixed in newer boards, nevertheless, I use 8266s powered from the Vin pin all the time with no problems.

shadowkrazee, 2019-03-09:
I am having this same issue, and my experience is limited to arduino, where everything works fine when powering via 5v pin. If i'm understanding @fpovoski correctly, i just need to provide power to another pin in addition to the +5v? If so, which pin is this, and if not, what can i do to resolve this, aside from USB powering the ESP32?

closed 2019-08-15 by stale[bot]