espressif/arduino-esp32 #1495: ArduinoOTA library

chegewara, 2018-06-13:
Ive been asking on forum, but no response. Is it possible to update with this library like with esp-idf ota or is it only to update with esptool.py/exe and from arduino-ide?

lbernstone, 2018-06-13:
https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino

chegewara, 2018-06-13:
I know that example, but there is no info how to push file from internet or any other way. Anyway thanks.

lbernstone, 2018-06-13:
I am sorry there is no example that does exactly what you want for your particular purpose.  There are a couple of examples in updater as well, and I'm sure google knows something about sketches with uploading to Webserver, but since that is not included in the codebase atm, it will not have an example here... Please close this issue.

chegewara, 2018-06-13:
Im not asking about particular example, what i am asking is: is it possible or not?

lbernstone, 2018-06-13:
Yes. ArduinoOTA is specifically for ArduinoOTA.  Update is a generalized library for upgrading firmware.

dronecz, 2018-06-25:
You can change and use AWS S3 OTA Update, if you want to update your ESP32 from server on Internet..

chegewara, 2018-06-25:
Sadly i cant make it work with this library:
https://github.com/me-no-dev/ESPAsyncWebServer
so most likely i will use esp-idf ota.

closed 2018-06-25 by chegewara