espressif/arduino-esp32 #1496: scanCount is uint16_t but scanNetworks returns only int8_t

theproxy, 2018-06-14:
### Hardware:
Board:							ESP32 Dev Module
IDE name:						Currently using Platform.io with ESP32
Flash Frequency:					?40Mhz?
Upload Speed:						?921600?
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)

### Description:
When scanning networks with many many access points, scanNetworks returns negative number (not -1 or -2) and you can still read the output of scanNetworks but now have to deal with overflowed number returned from scanNetworks.

theproxy, 2018-06-14:
in file https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiScan.cpp

lbernstone, 2018-06-15:
libraries/WiFi/src/WiFiType.h:#define WIFI_SCAN_RUNNING   (-1)
libraries/WiFi/src/WiFiType.h:#define WIFI_SCAN_FAILED    (-2)

theproxy, 2018-06-15:
Hi thank you for the reply. I saw the -1 and -2 in the code. What I am seeing (still gathering data on this) is that I am getting a number like -67 or so. I have structured my code that if I see scanNetworks return < -2, then I can ask for at least 127 WiFi.SSID(i) and I get valid data from that call up through that many indices.

It's like walking a tightrope though because I am guessing that there is that much information available in the WiFI data structure, but I know it's a hack to get that data by blindly hardcoding the for loop.

copercini, 2018-07-01:
Solved on https://github.com/espressif/arduino-esp32/pull/1514

Nice PR =)

closed 2018-07-01 by copercini

YordanYanakiev, 2021-06-29:
The issue is not resolved. Only the first time it is working, after then I am getting negative numbers like -2.