espressif/arduino-esp32 #1497: Failed to connect to ESP32: Timed out waiting for packet header

aniellod, 2018-06-14:
### Hardware:
Board:							ESP32 LOLIN w/OLED - ESP-WROM32
Core Installation/update date:		        (see below)
IDE name:						Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Upload Speed:						Tried various

### Description:
Unable to upload anything to this module. 

Have another identical one and same sketch uploads and runs fine. Rules out cable, connection or computer. I tried to upload using esptool.py directly and also using FTDI but same problem.

How can I test if the board damaged? Any tips on how to revive it?

### Debug Messages:
(/dev/ttyUSB1)
ets Jun  8 2016 00:22:57

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

Sketch uses 638418 bytes (48%) of program storage space. Maximum is 1310720 bytes.
Global variables use 38948 bytes (13%) of dynamic memory, leaving 255964 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.3.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```

mrmonteith, 2018-06-14:
I take it the port shows and that it just doesn't upload.  I've bought a couple of dev boards that they didn't design well. I have to hold down the reset but after esptool starts to try to send for about a second or two and then release and it loads.   All my other boards load with no hand holding.  Hope that is what the problem is.

aniellod, 2018-06-14:
Sadly it's not helping in this case. Any variations of pressing the boot/flash button or holding it, releasing it seems to fail. Oddly enough the problem was gradual. The board was working fine without needing to press "boot", then it stopped working and I had to press "boot" for it to accept the code. Now nothing works.  I had a look and tried to diagnose using [this](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection) information but no luck.

CaptIgmu, 2018-06-16:
Make sure certain pins are not connected when uploading programs! In my case, pin 12 was connected to an INT signal from a sensor and could not upload. There may be other ESP32 pins that are sensitive. Also, try DIO mode for the flash in boards.txt file . . .

lbernstone, 2018-06-16:
gpio12 and gpio2 are particularly notable on this: https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#note-about-gpio12

aniellod, 2018-06-18:
@lbernstone Thanks. The only one who might need to be shamed is myself. I might have done something wrong to the board which broke it. I ordered two and the second one is fine. Perhaps too much voltage on pin 0 and 2.

davidbono, 2018-06-30:
I had the same issue and fixed it by rotating 180°  lopy4 board when plugin into the expansion board ... shame on me.

pwpolar, 2018-08-02:
I'm having this issue with an ESP32 DevKitC_V4 board. Loads fine and then all of a sudden  get this message:
 Flashing binaries to serial port COM27 (app at offset 0x10000)...
esptool.py v2.1
Serial<id=0x24dcfec, open=True>(port='COM27', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False)
Connecting........_____....._____....._____....._____....._____....._____....._____.....
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

Sometimes power cycling board fixes problem other times does not. I can see ESP32 replying using scope on TXD0 but computer is not seeing response.

dcartman, 2018-08-22:
I had this same issue with an ESP32 Dev Module V1, to "fix it" I held down the boot button while compiling/uploading the sketch from Arduino.

Once the sketch was uploaded I power cycled by unplugging the USB and plugging back in. Uploads worked fine after that for a while.

Fuzzbinder, 2018-09-14:
> I take it the port shows and that it just doesn't upload. I've bought a couple of dev boards that they didn't design well. I have to hold down the reset but after esptool starts to try to send for [...]

This just fixed the same issue for me.  Thanks

nello, 2018-10-02:
I get this error whenever the tx0 pin is connected to my breadboard. Letting it float free is the only thing that allows me to upload new code. Very strange. My board is an ESP-WROOM-32.

nello, 2018-10-08:
FIXED: To be able to upload to the EP-WROOM-32 when connected to a breadboard, you need a 10K (or so) pull-down resistor between RX0 and GND.

Amaktronix, 2018-11-03:
@dcartman  Thanks.   yours solution gave a effect.
The problem was  solved for me 
My module is DOIT ESP32 DEVKIT V1    which is the same of  ESP32 Dev Module V1 .

fixed the same issue for me. Thanks

lemonbuzz, 2018-11-24:
I have a custom board with the same issue. I get the 'waiting for download' prompt but the upload always times out.

pjvander, 2019-01-07:
I ran into the same issue and couldn't resolve for the life of me.  Turns out I have been shipped (twice, now) an ESP8266 in an ESP32 packaging.  Configuring for the ESP8266 solved my issue, naturally.  Throwing this out there in case this same thing happened to others.

Stitch0, 2019-02-08:
Hi. In my case, pressing the boot button helped.

68gt500, 2019-02-23:
I had the same problem on my Windows 10 PC.
In my case installing CP210x USB to UART Bridge VCP Drivers solved the issue. 
Works perfectly, even at the highest baud rate of 921600.

Paelgin, 2019-03-09:
I had that Timeout problem, but it only just showed up when using my old Dell Inspiron 1525 (from 2006).  My newer laptop had done fine (but sadly, it is not on vacation with me, only my old one) On a hunch, I went out to Best Buy and bought a USB 3.0 powered hub, and with great joy, I found that programming it now works through that hub.

I'm using a "ESP32-01 Breakout and Development Board" from www.ezsbc.com to make Geoff Bunza's "Simplified WiFi Throttle" that he described in Model Railroad Hobbyist (https://model-railroad-hobbyist.com/node/35652).

Phil in gorgeous Young Harris, Georgia, USA

sravanth005, 2019-03-20:
Hi all,
I am trying to work with ESP32-CAM (https://robotzero.one/esp32-camera-module/#comment-3073) i can able to take a still with the camera but face detection and face reorganization is not happening, Do any one know for what reason it is not happening.please help me out to solve this issue.

dhcps: send_offer>>udp_sendto result 0
I (63186) camera_httpd: JPG: 6335B 1573ms
I (66456) camera_httpd: vflip = 1
dhcps: send_offer>>udp_sendto result 0
I (72716) camera_httpd: JPG: 7267B 118ms
I (78976) camera_httpd: face_detect = 1
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
I (1752586) wifi: max connection, deauth!
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
I (1873156) wifi: max connection, deauth!
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
I (2301736) wifi: station: b4:ef:fa:79:e6:e2 leave, AID = 1, bss_flags is 134243
I (2301736) wifi: new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:6
I (2301736) camera wifi: station:b4:ef:fa:79:e6:e2leave, AID=1
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0

thanks in advance

Gohith, 2019-05-02:
> ### Hardware:
> 
> Board: ESP32 LOLIN w/OLED - ESP-WROM32
> Core Installation/update date: (see below)
> IDE name: Arduino IDE 1.8.5
> Flash Frequency: 80Mhz
> Upload Speed: Tried various
> ### Description:
> 
> Unable to upload anything to this module.
> 
> Have another identical one and same sketch uploads and runs fine. Rules out cable, connection or computer. I tried to upload using esptool.py directly and also using FTDI but same problem.
> 
> How can I test if the board damaged? Any tips on how to revive it?
> ### Debug Messages:
> 
> (/dev/ttyUSB1)
> ets Jun 8 2016 00:22:57
> 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
> waiting for download
> 
> ```
> Sketch uses 638418 bytes (48%) of program storage space. Maximum is 1310720 bytes.
> Global variables use 38948 bytes (13%) of dynamic memory, leaving 255964 bytes for local variables. Maximum is 294912 bytes.
> esptool.py v2.3.1
> Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____
> 
> A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
> A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
> ```
> 
> Thanks

Just hold the BOOT switch while it says "Connecting......."
It solved for me.

Parth7030, 2019-05-11:
> I had the same problem on my Windows 10 PC.
> In my case installing CP210x USB to UART Bridge VCP Drivers solved the issue.
> Works perfectly, even at the highest baud rate of 921600.

Thanks, It worked for me...

VijayJaybhay, 2019-06-08:
> Make sure certain pins are not connected when uploading programs! In my case, pin 12 was connected to an INT signal from a sensor and could not upload. There may be other ESP32 pins that are sensitive. Also, try DIO mode for the flash in boards.txt file . . .

Thanks, it worked not sure why it is so

beegee-tokyo, 2019-06-08:
[Reserved GPIOs - special use GPIOs](https://desire.giesecke.tk/index.php/2018/07/06/reserved-gpios/)

daveyman123, 2019-06-08:
I am getting this error and no series of button presses will fix it. How do I start over?

nello, 2019-06-08:
@daveyman try a pull down resistor on RX0 (see my comment above).

mikkieg, 2019-07-27:
Had the same problem which seems to be fixed now with a 10uF cap between RST and GND

gpratt3151, 2019-09-20:
I found my problem was with the USB port on my laptop. If I use a USB 3.0 cable (USB-A -> USB 3) in the USB 2.0 port, I get the exact error messages above. When I moved the cable to the USB 3.0 port it worked just fine. I didn't realize my laptop had a single 2.0 port which is the one I naturally use due to it's location. As I understand, it should have worked with the USB 2.0 port, but it does not on my laptop either due to the cable or the port. Hope this helps someone.

mehtaSmitK, 2019-09-22:
Just make sure whether following conditions meet,
1. Check if GPIO 0(**IO0**)) is connected with Ground(**gnd**),
2. Press & Hold  **rst**  button  when you are uploading the code,
3. In my case, I had connected 5V output instead of 3 .3V output.

men0, 2019-11-02:
> 3\. In my case, I had connected 5V output instead of 3 .3V output.

#3 did work for me! thanks a lot.

night021, 2019-11-04:
> Just make sure whether following conditions meet,
> 
> 1. Check if GPIO 0(**IO0**)) is connected with Ground(**gnd**),
> 2. Press & Hold  **rst**  button  when you are uploading the code,
> 3. In my case, I had connected 5V output instead of 3 .3V output.

Where is GPIO0 on esp32 dev kit v1

lbernstone, 2019-11-04:
GIS: esp32 dev kit v1 pin map
There are about 100 images there

DavitecBrasil, 2019-11-04:
Hi Everybody!
I had this problems a few times when my projects had anything connected to ESP32-WROOM-32 pins RX0 and TXO. (A max485 chip, for example). It take long until I tried to disconnect the max485 from VCC  while uploading my sketches to de ESP32... And the problem was gone...
Sometimes it also happens when there is some other devices connected to the circuit as a multimeter or a logic analyzer, especially if they are connected to other power sources.

jhoughjr, 2019-11-09:
seeing this issue on mac os 10.15.1 latest Arduino. Been dead in the water for a month. ESP was a mistake. Maybe STM32 can actually be programmed reliably.

me-no-dev, 2019-11-09:
@jhoughjr would the thought that maybe the problem is in your computer ever passed you head? Obviously if ESP32 was not working on Mac that would have been a huge issue... not just for you. I use Mac too. Yet have no problem, like most other users.

jhoughjr, 2019-11-09:
That definitely crossed my mind but considering if one googles this they see the others having the same issue then its not likely to be my computer  but the unhandled exception.
Considering it works on the same computer on windows ten also makes me think it is not my computer.
Im just using windows for now.
Last I have seen is wait for apple to change the OS to not throw an exception. That is not my computer.

me-no-dev, 2019-11-09:
it is something environment related. I did not mean it's hardware related. If I ever get to experience it, I could look for the issue, but unless that can happen and unless the people that do get to experience it try to get to the bottom of it... it's kinda unfixable.

atanisoft, 2019-11-09:
Very likely a driver related issue. The fact that it works on Windows somewhat confirms that.

jvr-ks, 2019-11-12:
Pressing and holding (!) the boot-button during flashing the NodeMCU firmware solved it for me with newer Dev board.

JamesNewton, 2020-01-02:
On the off chance anyone else is as stupid as I am, the same issue from the Arduino IDE applies here: You must not have the serial monitor open when you try to upload. Unlike the program upload, the SPIFFs ("File System image") upload will not close it, use it's port, and then re-open it in the monitor when done. It just fails. It took me a while longer to figure this out, because I'm using the JTAG debugger and the SPIFFs upload tried to use it's port instead of telling me no port was available. 

Not sure if that's worth a feature request...

clzola, 2020-01-27:
I have recently acquired LoLin D32 board, and had a problem with uploading basic example:

```
int LED_PIN = 5;

void setup() {
  pinMode(LED_PIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_PIN, HIGH);
  delay(1000);
  digitalWrite(LED_PIN, LOW);
  delay(1000);
}
```

I solved my problem with changing **Tools > Upload Speed** to **230400**, I have not tried other values, but this one worked for me... Hope it helps.

jayaruvan, 2020-04-01:
I am having the same issue.

Paelgin, 2020-05-31:
Several times I have had problems.  Once when I was connected by USB to an old laptop that may have only been USB 1 ports.  I bought a powered USB hub to resolve that. Another time was when I unwittingly attached a switch to the Input IO0 (used for boot).  When the input was grounded, bad things happened during startup.

This should make this thread UNstale.
Phil

gadavidd, 2020-06-01:
I have the same problems, I changed this line:

//#define CAMERA_MODEL_WROVER_KIT
//#define CAMERA_MODEL_ESP_EYE
//#define CAMERA_MODEL_M5STACK_PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE
#define CAMERA_MODEL_AI_THINKER 

and It Worked!

maddisondesigns, 2020-06-18:
Just recently got a couple of ESP32 boards and changing the upload speed (to **230400**) as per @clzola's [comment above](https://github.com/espressif/arduino-esp32/issues/1497#issuecomment-578907056), worked for me.

adrianTNT, 2020-07-26:
I had this error today, while also noticing the **blue light was very low** when trying to upload:
```
Failed to connect to ESP32: Timed out waiting for packet header
```
**I moved it to a better USB port and it worked**, so it was most probably not having enough power in my case.

zatuliveter, 2020-11-28:
Thanks to @mikkieg
> Had the same problem which seems to be fixed now with a 10uF cap between RST and GND

Here is how I fixed firmware upload on ESP32-DevKitC V4 with ESP-WROOM-32 board
No need to hold `Boot` button during upload.

[image]

rhyslewis-aws, 2021-03-25:
I can't say for sure which of these are required, but I got it working when I:

- Changed the upload speed to 115200
- Held down the reset button while the sketch was compiling, then released it after the Arduino IDE said, "Connecting ..."
- Put a jumper between Gnd and 001

Once that was done I removed the jumper and USB, then reinserted USB, and the sketch was running fine.

quantumfluctuations, 2021-05-01:
I too had the same issue..
I was working with my laptop which had a single USB 3.0 port and dual USB 2.0 ports
I then switched to my old laptop in which all the ports were USB 2.
This error did not fire when I used the ESP32  WROOM 32 board on my old laptop
I suggest everyone  working on the ESP32 modules to use systems with uniform USB standard revisions

Aidenir, 2021-06-05:
Super strange, I was working on a program, uploading changes every half hour or so for like 4 hours, suddenly I'm getting this error out of nowhere... Tried multiple different modules, none work. Tried a Pro micro, that worked fine. What got me back up and running was googling "CP210x USB to UART Bridge VCP Drivers" like @68gt500 said, and now it works. Although now I do have to hold the boot button while it says connecting, which I didn't before :/

willz1200, 2021-07-31:
I had a similar issue to this when attempting to program an ESP32-S2 under esp-idf. Upon entering programming mode the following error occurred.
```
esptool.py v3.1-dev
Serial port COM5
Connecting.....
Chip is ESP32-S2
Features: WiFi, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz
MAC: ##:##:##:##:##:##
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 3072 bytes to 143...

A fatal error occurred: Timed out waiting for packet content
```
The issue turned out to be a bad solder joint between the ESP32-S2 and flash chip (even though I could id the esp and flash using esptool).
After reworking the chips everything programs fine.

b0xmaker, 2021-08-04:
Similar issue here. Using an Ubuntu VM through Windows 10.

`A fatal error occurred: Timed out waiting for packet content`

**I was able to resolve this issue by bumping up the  USB hub speed, it was set to `USB 1.1` but once changed to `USB 3.0` I got it to work.**
[image]

markotime, 2021-09-08:
TTGO-T display unit.  Would not communicate, port did not show up in WIN10 device manager.  Noticed that an UNO using CH340 did appear later, showing port number, so reconnected TTGO.  Nothing until I updated drivers, and all ok since.  Surprised that versions using CP21XX and CH340 would be available, without showing in spec.

HarborDesigns, 2021-10-18:
[image]
I have a custom board where I am using an ESp32 Wroom module. I am not able to flash code via serial communication. I am using FTDI232 USB to UART for serial communication. I am able to get serial messages in my serial monitor which mean there is no issue in serial wires. 

I have attached a capacitor and a pull up resistor on EN line. GPIO0 is also pulled to GND during flashing the code. Please see the attached rough schematic of my system and timing diagram for more information. I have jumped my EN and GPIO0 line from board to breadboard where I am manually jumping the EN & GPIO 0 wires to GND. This is just to be sure that I am pulling down the signals while booting and there is no noise or sudden voltage fluctuation while toggling the signals.

For extra precaution I have also pulled up or pulled low other strapping pins as shown in schematic. I am not sure where I am doing mistake. 

While uploading the code I pull down first GPIO 0 line and then EN line to GND and later when I see _'Connecting........_____' message on my Arduino IDE terminal I pull up the EN line and keep pressing the GPIO0 line until the Arduino IDE display error message: _'A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header'_

atanisoft, 2021-10-18:
@HarborDesigns refer to any of the DevKit schematics (ie: https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch-20180607a.pdf) for the DTR/RTS reset circuit requirements to have the board automatically enter download mode.

HarborDesigns, 2021-10-19:
Yes, i should be able to do that but it is not flashing new code on ESP32. Does my rough timing diagram looks correct in my previous picture?

I know I can enter my ESP32 into program mode as I see following message on my serial monitor. Whenever my GPIO 0 is low and then when I toggle the EN signal from 1 (high) to 0 (low) and then back to 1(high). It shows following message which is good 
_'rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download'_

Looks like it doesn't reads data over serial communication after that but my serial communication works. What I understand is that I can put ESP32 in program mode if my GPIO 0 is zero after reset and if my GPIO 0 is high after reset then it goes into normal mode. In my case, I can enter in and out of both modes and my serial communication prints that message but I can't flash new code because of packet header issue/ time out of the session.

Mathteus, 2021-12-16:
Resume

tenha uma placa esp32 ela estava compilando antes normamente, tanto que esta com o codigo funcionando dentro dela, mas ela parou de funcionar nao sei como resolver, alguem poderia me ajuda? 

wifi-esp-wroom-32
ide arduino 1.8.18

config

board: esp32 wrover module
uploaad speed: 921600
flash frequency 80MHz
flash mode QIO
partition schema: default 4Mb
core debug level: none 
port: ttyUSB0

things already tested

segurar, apertar varias vezes, apertar em tempos os botoes boot - reset - boot + reset durante a compilação, colocar um capator em GND e GIO 0, colocar um capacitor entre GND e 3v3

nothing went right with me will my plate burn?

DaveStu, 2022-01-07:
Hi all. I've had this board for a couple of months with a project I'm working on with someone else.
Its a TTGo display ESP32 with CP2104 USB to UART
I've been flashing .bin files with esphome flasher succesfully for the last few months, now suddenly it stopped taking a flash.
The screen will start to flash several times while esphome flasher tries to connect, but it fails, then boots back to previous code and is able to work with last upload, just no new flashing.
I've been all over forums and internet looking for solutions, tried to hold reset, button 1 and/or 2.
Tried different USB c data cables. I cannot disconnect board as its soldered in, to test out of circuit.
The port is correct and other variables, which were never changed but checked anyway

I'm now stuck with last code and cannot update our project. Can I bypass USB and flash via pins, like esp01 with USB flasher?
Up till now this board has been great, but this has now got me so frustrated.

Any clue anyone? Really strange all of a sudden. Thanks

jparshook, 2022-02-08:
For what it's worth: I had issues with an ESP32 not flashing, even while holding boot or using a resistor. The other 4 boards that came with this board were working fine. I used a small convection oven at 450 F for 2 minutes to reflow the solder. The board in question is flashing now without any issues.

cvcore, 2022-03-17:
Using `m5stack-timer-cam` here.
Setting `upload_speed = 115200` instead of the default value solved the problem for me.

thgfeller, 2022-04-24:
Same problem here with an ESP32-S  2AHMR  
I have it connected to an ESP32-CAM-MB. That should be the official programming board. Has both reset and IO0 but no matter how I fiddle with them I receive always : "Failed to connect to ESP32: Timed out waiting for packet header". Arduino IDE says it is connected to Com4. But serial monitor never shows anything, not even after I pressed reset.

Dasberacynthia, 2022-12-05:
I'm having the same problem with ESP-32 cam-S. Compiling is done perfectly but while uploading I am getting the error "failed to connect esp32:timed out waiting for packet header" repeatedly even when I am holding the reset button while uploading.
How can I find out if the camera module is working properly or not?If yes then what went wrong....if anyone have solution to my problem it'd be really helpful....

[6 more replies]

PR #8241 (2023-08-30): Doc update: Troubleshooting + SD libs
Troubleshooting:
- Updated flashing problems; closes #1497 
- Added paragraph describing S3 PSRAM options. Related discussion [#6426](https://github.com/espressif/arduino-esp32/discussions/6426), related issues #6980, ...  Related [article](https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/)
- Moved SPIFFS from FAQ to Troubleshooting
- SD card mount fail - closing #259 closing #7707; related: #8076 #963

SD libraries updates:
SD_MMC_Test.ino - added hint to read README
SD_MMC/README.md - new file, partial [... cut]
closed 2023-07-04 by PilnyTomas