espressif/arduino-esp32 #1501: FreeRTOS EventGroups

hreintke, 2018-06-15:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							Nano32
Core Installation/update date:		15/06/2018
IDE name:						Sloeber eclipse
Flash Frequency:					80Mhz
Upload Speed:					230400

### Description:
Experimenting with events groups from freertos. 
Now using an example from the freeRTOS documentation.
But this gets into the Guru Meditation Errors.

Any hints where I should be looking to solve this ?

### Sketch:
```cpp
extern "C"
{
#include "freertos\freeRTOS.h"
#include "freertos\task.h"
#include "freertos/event_groups.h"
}

void loop()
{
}

/* Definitions for the event bits in the event group. */
#define mainFIRST_TASK_BIT ( 1UL << 0UL ) /* Event bit 0, set by the first task. */
#define mainSECOND_TASK_BIT ( 1UL << 1UL ) /* Event bit 1, set by the second task. */
#define mainTHIRD_TASK_BIT ( 1UL << 2UL ) /* Event bit 2, set by the third task. */

/* Declare the event group used to synchronize the three tasks. */
EventGroupHandle_t xEventGroup;

static void vSyncingTask( void *pvParameters )
{
    const TickType_t xMaxDelay = 2000 ;// pdMS_TO_TICKS( 4000UL );
    const TickType_t xMinDelay = 500 ;//pdMS_TO_TICKS( 200UL );

    TickType_t xDelayTime;
    EventBits_t uxThisTasksSyncBit;
    const EventBits_t uxAllSyncBits = ( mainFIRST_TASK_BIT | mainSECOND_TASK_BIT | mainTHIRD_TASK_BIT );

    /* Three instances of this task are created - each task uses a different event bit in the synchronization.
     * The event bit to use is passed into each task instance using the task parameter. Store it in the uxThisTasksSyncBit variable. */
    uxThisTasksSyncBit = ( EventBits_t ) pvParameters;

    for( ;; )
    {
        /* Simulate this task taking some time to perform an action by delaying for a pseudo-random time.
         * This prevents all three instances of this task from reaching  the synchronization point at the same time,
         * and so allows the example's behavior to be observed more easily. */
 //       xDelayTime = ( rand() % xMaxDelay ) + xMinDelay;
    		xDelayTime = random(xMinDelay,xMaxDelay);
    	//        vTaskDelay( xDelayTime );
    	Serial.printf("%s delay is %d\r\n",pcTaskGetTaskName( NULL ),xDelayTime);
        delay(xDelayTime);
        /* Print out a message to show this task has reached its synchronization  point.
         * pcTaskGetTaskName() is an API function that returns the name assigned to the task when the task was created. */
        Serial.printf("%s reached sync point at : %d\r\n",pcTaskGetTaskName( NULL ),(int)millis());

        /* Wait for all the tasks to have reached their respective synchronization points. */
        xEventGroupSync( /* The event group used to synchronize. */ xEventGroup,
        		        /* The bit set by this task to indicate it has reached the synchronization point. */ uxThisTasksSyncBit,
						/* The bits to wait for, one bit for each task taking part in the synchronization. */ uxAllSyncBits,
						/* Wait indefinitely for all three tasks to reach the synchronization point. */ portMAX_DELAY );

        /* Print out a message to show this task has passed its synchronization point.
         * As an indefinite delay was used the following line will only be executed after all the tasks reached their respective synchronization  points. */
        Serial.printf("%s exited sync point at : %d\r\n",pcTaskGetTaskName( NULL ),(int)millis());
    }
}

void setup()
{
	Serial.begin(115200);
    /* Before an event group can be used it must first be created. */
    xEventGroup = xEventGroupCreate();

    /* Create three instances of the task. Each task is given a different name, which is later printed out to give a visual indication of which task is executing. The event bit to use when the task reaches its synchronization point is passed into the task using the task parameter. */

    xTaskCreatePinnedToCore( vSyncingTask, "Task 1", 1000, (void*)mainFIRST_TASK_BIT, 1, NULL,1);
    xTaskCreatePinnedToCore( vSyncingTask, "Task 2", 1000, (void*)mainSECOND_TASK_BIT, 1, NULL,1 );
    xTaskCreatePinnedToCore( vSyncingTask, "Task 3", 1000, (void*)mainTHIRD_TASK_BIT, 1, NULL,1 );
}

```

### Debug Messages:

Serial output is : 
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a5c
Task 1 delay is 1509
Task 3 delay is 751
Task 2 delay is 735
Guru Meditation Error: Core  0 panic'ed (StoreProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x40086e0d  PS      : 0x00060033  A0      : 0x80085dde  A1      : 0x3ffc0590
A2      : 0x3ffc5da8  A3      : 0x00000000  A4      : 0x00000005  A5      : 0x00000020
A6      : 0x00000020  A7      : 0x00000000  A8      : 0x00000002  A9      : 0xa5a5a5a5
A10     : 0x3ffc5fdc  A11     : 0x00000000  A12     : 0x00060021  A13     : 0x00000001
A14     : 0x00060021  A15     : 0x3ffc6814  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d
EXCVADDR: 0xa5a5a5ad  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x40086e0d  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40085645

Backtrace: 0x40086e0d:0x3ffc0590 0x40085ddb:0x3ffc05b0 0x400845d7:0x3ffc05d0 0x40086afd:0x3ffc05f0 0x400815d2:0x3ffc0600 0x400d2fab:0x00000000
```

Decoder output : 
```
Decoding 12 results
0x40086e0d: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 214
0x40086e0d: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 214
0x40085645: xTaskGetCurrentTaskHandle at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3529
0x40086e0d: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 214
0x40085ddb: xTaskIncrementTick at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3529
0x400845d7: xPortSysTickHandler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./port.c line 374
0x40086afd: _frxt_timer_int at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./portasm.S line 303
0x400815d2: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./xtensa_vectors.S line 1105
0x400d2fab: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./freertos_hooks.c line 85
```

stickbreaker, 2018-06-15:
@hreintke `pvParameters` is the problem, you are passing 1,2,4 thru `xtaskcreate` but your task is using it as an address reference. 

Chuck.

hreintke, 2018-06-16:
@stickbreaker 
The use of pvParameters is the usual way for tasks.
pvParameters is a void* and casted as needed.

`uxThisTasksSyncBit = ( EventBits_t ) pvParameters;`

The usage is somewhat hidden thru the comments but is : 

`
 xEventGroupSync(  xEventGroup, uxThisTasksSyncBit, uxAllSyncBits, portMAX_DELAY );
`

It defines the bit in the xEventgroup that this task is setting to indicate its syncing.

hreintke, 2018-06-16:
Found the problem.

I executed the same in esp-idf and got the errormessage : 
```
EventGroup test
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
tasks startedTask 2 delay is 2433
Task 2 reached sync point at
***ERROR*** A stack overflow in task Task 2 has been detected.
abort() was called at PC 0x400873f4 on core 1

Backtrace: 0x40087204:0x3ffb7690 0x400873db:0x3ffb76b0 0x400873f4:0x3ffb76d0 0x400840e4:0x3ffb76f0 0x40085ee4:0x3ffb7710 0x40085e9a:0x3ffb77b0
```

Increasing the stacksize to 8192 : 
`xTaskCreatePinnedToCore( vSyncingTask, "Task 1", 8192, (void*)mainFIRST_TASK_BIT, 1, NULL,1);`

Solved it

Is there a way to detect stack overflow when using arduino-esp32 ?

closed 2018-06-17 by hreintke