espressif/arduino-esp32 #1503: ESPNow example broken

thomasfla, 2018-06-16:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			commit 89859f7f4c07ce3e5d487a4b689dbda78f5d5c3a
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
ESPNow example seems to be broken, Multi-Slave/Master and Basic/Master

### Sketch:
ESPNow/Multi-Slave/Master Example

### Debug Messages:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90
ESPNow/Multi-Slave/Master Example
STA MAC: B4:E6:2D:B5:9F:85
ESPNow Init Success
E (264) wifi: esp_wifi_scan_start 967 wifi not start

Found -2 devices 
No Slave Found, trying again.
E (1269) wifi: esp_wifi_scan_start 967 wifi not start

Found -2 devices 
No Slave Found, trying again.
E (2269) wifi: esp_wifi_scan_start 967 wifi not start

lbernstone, 2018-06-20:
This should now be fixed. Can you update your local repo and test?

thomasfla, 2018-06-20:
It solves the issue, Thanks !

closed 2018-06-20 by thomasfla