espressif/arduino-esp32 #1507: Is it possible to put in precossesors for board #ifdef FEATHER_ESP32  #endif

wegunterjr, 2018-06-18:
I am looking all over for how to compile based on board type.
I have a feather32 and a devkit, which means some of the IO is different.
I would like to set up based on the IO without making multiple projects.

`#ifdef FEATHER_ESP32
#define BUILTIN_LED 16
#endif`

I have seen it where you can change it based on esp8266 or esp32, but not sure about this...
Thoughts?
Suggestions?

beegee-tokyo, 2018-06-18:
If you are using PlatformIO you can get the board type fromthe define `ARDUINO_BOARD`.  On ArduinoIDE I don't think it is possible.

wegunterjr, 2018-06-18:
I am using PlatformIO.  (LOVE IT!)  where is that define?  I am looking now...

beegee-tokyo, 2018-06-18:
The boards names are defined in **`[YOUR_PATH_TO_PLATFORMIO].platformio\platforms\espressif32\boards`**. Open the .json file for your boards and look for **_`"name": "Adafruit ESP32 Feather"`_**
For the **_esp32dev_** you find in **_esp32dev.json_** the name **_Espressif ESP32 Dev Module_**
for the **_featheresp32_** you find in **_featheresp32.json_** the name **_Adafruit ESP32 Feather_**

```
String thisBoard= ARDUINO_BOARD;
Serial.println(thisBoard);
if (thisBoard.compareTo("Espressif ESP32 Dev Module") == 0) {
  Serial.println("Compiled for Espressif ESP32 Dev Module");
  ledPort = 0;
} else if (thisBoard.compareTo("Adafruit ESP32 Feather") == 0) {
  Serial.println("Compiled for Adafruit ESP32 Feather");
  ledPort = 16;
} else {
  Serial.println("Compiled for another board");
  ledPort = 5;
}
```
Added to [ESP32 WIKI](https://desire.giesecke.tk/index.php/2018/06/18/get-board-name-in-your-code/)

wegunterjr, 2018-06-18:
@beegee-tokyo  thanks for the clarification. I used your suggestion, and can am seeing success.  But, this might be another question, the compiler will compile all boards in my platform.io, but the last one is the one that seems to win.   Here is my platformio.ini, and 
`
[env:featheresp32]
platform = espressif32
lib_extra_dirs = ~/Documents/Arduino/libraries
board = featheresp32
framework = arduino

lib_deps =
      PubSubClient
      ArduinoJson

board_build.partitions =partition_custom.csv

[env:esp32dev]
platform = espressif32
lib_extra_dirs = ~/Documents/Arduino/libraries
board = esp32dev
framework = arduino

lib_deps =
      PubSubClient
      ArduinoJson

board_build.partitions =partition_custom.csv
`

Here is my sample code to test blinking the led based on board.   The last one is the one that shows up on my feather32.

`String thisBoard= ARDUINO_BOARD;
int ledPin = 0;

void setup() {

  Serial.begin(115200);

  Serial.println(ARDUINO_BOARD);
  if (thisBoard.compareTo("Espressif ESP32 Dev Module") == 0)
  {
    Serial.println("Compiled for Espressif ESP32 Dev Module");

    //relayPort = 0;
  } else if (thisBoard.compareTo("Adafruit ESP32 Feather") == 0)
  {
    Serial.println("Compiled for Adafruit ESP32 Feather");
    // blink led
    **ledPin = 13;
    pinMode(ledPin, OUTPUT);**
  } else
  {
    Serial.println("Compiled for another board");
  }

    // put your setup code here, to run once:
}

void loop() {
    // put your main code here, to run repeatedly:
    digitalWrite(ledPin, HIGH);
    Serial.println();
    delay(500);
    digitalWrite(ledPin, LOW);
    delay(500);
}
`

beegee-tokyo, 2018-06-18:
You didn't define upload ports or upload IPs (in case you use ArduinoOTA). Therefor PlatformIO compiles all boards and uploads to each board. Then of course if you have only attached one board it will receive the update of the last board defined in your platformio.ini.
Check the [platformio docs]( http://docs.platformio.org/en/latest/projectconf/section_env_upload.html) how to define upload ports.
I, as I program one board after another added 
```
[platformio]
; env_default = featheresp32
; env_default = esp32dev
```
on top of my platformio.ini
The `env_default` that is not commented out will be compiled and flashed.

In addition/alternative you add 
`upload_port = xxxx`
to each board config to define the upload port. Then when running compile & upload each device will get its customized firmware over the specified upload port.

wegunterjr, 2018-06-18:
excellent!  Thanks a ton and I have read the documentation you linked.  
I am getting this error now:
i added:
`[platformio]
 env_default= featheresp32`

`ConfigParser.ParsingError: File contains parsing errors: C:\Users\wegun\Documents\PlatformIO\Projects\TestPinouts_myBoards\platformio.ini
        [line 11]: ' env_default = featheresp32\n'
 `

beegee-tokyo, 2018-06-19:
Looks good, now you have to find out why you get an error during compiling.

Please close the issue if it is solved for you.

closed 2018-06-19 by wegunterjr