espressif/arduino-esp32 #1517: ESP32 equivalent of the ESP8266 "custom_crash_callback" function?

Rob58329, 2018-06-19:
Board: NodeMCU-32S
SW: Arduino-IDE with latest "github.com/espressif/arduino-esp32" software (as at 19Jun18)

To run your own code whenever there is an exception, for the ESP8266 you can use the function:
"extern "C" void custom_crash_callback(struct rst_info * rst_info, uint32_t stack, uint32_t stack_end ) {...}"

This function actually seems to compile for the ESP32, but does not appear to do anything (ie. does not appear to be run when there is an exception such as a divide-by-zero).  Instead you just get the normal crash info in the IDE-Serial-Monitor and the ESP32 reboots (giving: rtc_get_reset_reason(1)=12 "SW_CPU_RESET").

[In my case, I am trying to use the callback-function to store crash-info (in the RTC-memory and EEPROM), which works for the ESP8266, but does not appear to work for the ESP32.]

Is there an equivalent working "custom_crash_callback" function for the ESP32?

closed 2019-08-15 by stale[bot]

atanisoft, 2019-10-25:
@voelkerb @Rob58329 there is no similar functionality and it is also not guaranteed to be possible. By the time the esp32 has "crashed" the esp32 is not able to do much more than try and print a stacktrace and reboot. It sometimes can't even do that fully. There is no possibility to hook into this flow to call user code unfortunately, at least not that I've been able to find anywhere.

Rob58329, 2019-10-26:
@atanisoft: Thank for your comment ("_There is no possibility to hook into this flow to call user code unfortunately_") - This is a pity, but what I had guessed might be the case.  Interesting that this works fine for the ESP8266 (and pretty reliably from my experience), but has not been implemented for the ESP32 - oh well!

atanisoft, 2019-10-26:
@Rob58329 agreed, but at the point of the crash handler almost everything is unavailable. There may be options in esp-idf but they won't work in arduino-esp32 today

vslinuxdotnet, 2021-07-28:
Any crash_callback available? or this option is dead ?

Rob58329, 2021-08-01:
@vslinuxdotnet:  Sadly "ESP32 ... custom_crash_callback" functionality is still not available in "github.com/espressif/arduino-esp32" (as far as I can tell), as discussed above.

MassimilianoMontagni98, 2022-01-24:
Hi,
any news?

Thanks

Rob58329, 2022-01-25:
@MassimilianoMontagni98: Sadly "ESP32 ... custom_crash_callback" functionality is still not available in "github.com/espressif/arduino-esp32" as at Jan22 (as far as I can tell), as discussed above.

FYI. In my projects I have got round this by saving “got_to” numbers in rtc-memory, and then after an unexpected reboot printing/displaying the “got_to” number so I know where it crashed.

Often I can then identify a pre-cursor to the crash (in my projects this is often memory issues caused by failures in SSL network access).  Where this is the case I check amount of free RAM/Stack/Heap before the crashing process, and if not enough do a controlled ESP32-reboot instead of allowing the crashing process to run.

atanisoft, 2022-02-02:
@MassimilianoMontagni98 @Rob58329 custom crash callback is not something that is available in ESP-IDF (the underlying library for arduino-esp32). 

However, the approach I've taken for tracking data across reboots is something like this:

```
...
#include <esp32/rom/rtc.h>

static constexpr uint32_t RTC_BOOL_FALSE = 0;

static uint32_t RTC_NOINIT_ATTR bootloader_request;

void init(uint8_t reset_reason)
{
  if (reset_reason == POWERON_RESET)
  {
    bootloader_request = RTC_BOOL_FALSE;
  }
}

void setup()
{
  init(rtc_get_reset_reason(PRO_CPU_NUM));
}

```