espressif/arduino-esp32 #1518: Encrypt message - Guru Mediation Error

Kazuya91, 2018-06-20:
I am trying to encrypt a message with the libraries provided by the arduino core. Obviously I am doing something wrong.... 

The message to encrypt is a this string (in JSON-Format):
{"mac-address":"30:AE:A4:00:00:00"}

I get a Guru Meditation Error: Core 1 paniced.

`Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x400e018b  PS      : 0x00060c30  A0      : 0x800d2951  A1      : 0x3ffd1940  
A2      : 0x3ffd19ab  A3      : 0x00000001  A4      : 0x0000003c  A5      : 0x3ffda178  
A6      : 0x3ffd199b  A7      : 0x3ffda9cc  A8      : 0x00000000  A9      : 0x3ffd1940  
A10     : 0x3ffd19ac  A11     : 0x3ffd199b  A12     : 0x00000010  A13     : 0x3ffd19bc  
A14     : 0x001b26f3  A15     : 0x0479721b  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x400e018b:0x3ffd1940 0x400d294e:0x3ffd1970 0x400d341b:0x3ffd19f0 0x400d3a56:0x3ffd1a30 0x40136993:0x3ffd1a70
`
My Code:
`void encrypt_message(String msg){
  void esp_aes_acquire_hardware();
  const unsigned char * const_str =  (const unsigned char *)(msg.c_str());
  unsigned char key[16] = "0123456789ABCDE";
  unsigned char iv[16];
  memset(iv,0,16); //fill IV with zeros
  
  esp_aes_context ctx;
  esp_aes_setkey(&ctx, key, 128); //set key and choose 128Bit-Encryption
  
  esp_aes_encrypt(&ctx, iv, key);

  //encrypt the message
  size_t len = msg.length();
  unsigned char  *ciphertext = (unsigned char  *) malloc(len); //cast it for esp_aes_crypt_cfb128
  memset(iv,0,12); //copies 0 to the first 12 characters of iv
  memcpy(iv+12,&msg_counter,4); //message_counter, 4byte = 32bit counter 
  esp_aes_setkey(&ctx, key, 128);
  
  int encrypt_res = esp_aes_crypt_cfb128(&ctx, ESP_AES_ENCRYPT, len, 0 ,iv, const_str, ciphertext);
  
  if (encrypt_res == 0){
    Serial.println("Encryption successful.");
  } else{
    Serial.println("Encryption FAILED");
  }
}`

Putting the backtrace into ESP Exception Decoder gives me this:
`Decoding stack results
0x400e018b: esp_aes_crypt_cfb128 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/hwcrypto/aes.c line 272
0x400d294e: encrypt_message(String) at /Users/xxx/Documents/Arduino/test/test.ino line 623
0x400d341b: createJson() at /Users/xxx/Documents/Arduino/test/test.ino line 383`

Does someone have an idea whats going on? I have the feeling that it has something to do with memory allocation etc. but I am not sure. There are also no examples out there.

lbernstone, 2018-06-21:
I don't like that I get different results, but it does run:
```
#include <hwcrypto/aes.h>
uint32_t msg_counter = 0;

void encrypt_message(String msg){
  const unsigned char * const_str =  (const unsigned char *)(msg.c_str());
  unsigned char key[16] = "0123456789ABCDE";
  char in[msg.length()];
  snprintf(in,sizeof(in),"%s",msg); 
  unsigned char out[16];

  esp_aes_context ctx;
  esp_aes_setkey(&ctx, key, 128); //set key and choose 128Bit-Encryption  
  esp_aes_encrypt(&ctx, (unsigned char*)in, out); //This function can only handle 16 chars
  Serial.println((char*)out);

  //encrypt the message
  size_t len = msg.length();
  unsigned char iv[16];
  size_t offset;
  unsigned char ciphertext[len]; //cast it for esp_aes_crypt_cfb128
  esp_aes_setkey(&ctx, key, 128);
  int encrypt_res = -1;
  encrypt_res = esp_aes_crypt_cfb128(&ctx, ESP_AES_ENCRYPT, len, &offset ,iv, (unsigned char*)in, ciphertext);
  if (encrypt_res == 0){
    Serial.println((char*)ciphertext);
  } else{
    Serial.println("Encryption FAILED");
  }
}

void setup() {
  Serial.begin(115200);
  String msg1="test_message_12";
  encrypt_message(msg1);
}

void loop() {vTaskDelete( NULL );}
```

Kazuya91, 2018-06-23:
@lbernstone 
Thx, your code helped me. Seems like I cant just give a zero as offset. My bad.

I have a question regarding the CBC-Algortihm. I also want to use the function below.

```
int esp_aes_crypt_cbc( esp_aes_context *ctx,
                       int mode,
                       size_t length,
                       unsigned char iv[16],
                       const unsigned char *input,
                       unsigned char *output );
```

The length of the message must be a 16/32/64/128 bytes etc. If its not its going to crash. Is there a padding function I can use? My message isnt always exactly one of those sizes above.

closed 2019-08-15 by stale[bot]