espressif/arduino-esp32 #1519: Adding the WiPy 3.0 board

korstiaanS, 2018-06-21:
Hi,

Can somebody help/add the WiPy3.0 board?
I do not know how to do it.

This is a nice board with internal/**external** antenna, WS2812 RGB led, 8MB flash, ...
(Do not use the expansion board to connect, it is not compatible with Arduino or ESP-IDF. Just use a separate usb convertor and connect the DTR and RTS lines))
[https://pycom.io/product/wipy-3](https://pycom.io/product/wipy-3/)
Pins, etc...
[https://docs.pycom.io/chapter/datasheets/development/wipy3.html](https://docs.pycom.io/chapter/datasheets/development/wipy3.html)

Thanks for helping.

copercini, 2018-07-01:
@korstiaanS please open a Pull request with this variant, it will certainly help other people with the same board =)

korstiaanS, 2018-07-03:
Is added!

closed 2018-07-03 by korstiaanS