espressif/arduino-esp32 #1521: The same sketch on ESP32 is much bigger than sketch on ESP8266

Deimos1994, 2018-06-21:
### Hardware:
Board:							ESP32 vs ESP8266
Core Installation/update date:			ESP32 git from 21/jun/2018 vs ESP8266 CoreV2.4.0)
IDE name:							Arduino IDE
Flash Frequency:					       40Mhz
Upload Speed:					115200

### Description:
Sketch on ESP32 is much bigger (about 2 times bigger).
Here is WiFiMulti as an example.

### Sketch ESP8266:
```cpp

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

ESP8266WiFiMulti wifiMulti;

void setup() {
    Serial.begin(115200);
    delay(10);
	
    WiFi.mode(WIFI_STA);
    wifiMulti.addAP("ssid_from_AP_1", "your_password_for_AP_1");
    wifiMulti.addAP("ssid_from_AP_2", "your_password_for_AP_2");
    wifiMulti.addAP("ssid_from_AP_3", "your_password_for_AP_3");

	Serial.println("Connecting Wifi...");
    if(wifiMulti.run() == WL_CONNECTED) {
        Serial.println("");
        Serial.println("WiFi connected");
        Serial.println("IP address: ");
        Serial.println(WiFi.localIP());
    }
}

void loop() {
    if(wifiMulti.run() != WL_CONNECTED) {
        Serial.println("WiFi not connected!");
        delay(1000);
    }
}
```
RESULT:
```
Sketch uses 254175 bytes (24%) of program storage space. Maximum is 1044464 bytes.
Global variables use 33328 bytes (40%) of dynamic memory, leaving 48592 bytes for local variables. Maximum is 81920 bytes
```

### Sketch ESP32:
```cpp

#include <WiFi.h>
#include <WiFiMulti.h>

WiFiMulti wifiMulti;

void setup()
{
    Serial.begin(115200);
    delay(10);

    wifiMulti.addAP("ssid_from_AP_1", "your_password_for_AP_1");
    wifiMulti.addAP("ssid_from_AP_2", "your_password_for_AP_2");
    wifiMulti.addAP("ssid_from_AP_3", "your_password_for_AP_3");

    Serial.println("Connecting Wifi...");
    if(wifiMulti.run() == WL_CONNECTED) {
        Serial.println("");
        Serial.println("WiFi connected");
        Serial.println("IP address: ");
        Serial.println(WiFi.localIP());
    }
}

void loop()
{
    if(wifiMulti.run() != WL_CONNECTED) {
        Serial.println("WiFi not connected!");
        delay(1000);
    }
}
```
RESULT:
```
Sketch uses 537078 bytes (40%) of program storage space. Maximum is 1310720 bytes.
Global variables use 38036 bytes (12%) of dynamic memory, leaving 256876 bytes for local variables. Maximum is 294912 bytes.
```
537078 bytes vs 254175 byte is pretty big difference.
I doubt that I the first one who noticed this, but I did not find any info about this issue.

lbernstone, 2018-06-22:
There is a "partition scheme" option in tools if you choose the esp32-dev board.  If you are done venting, please close the issue.

closed 2018-06-23 by Deimos1994

stickbreaker, 2018-06-26:
@Deimos1994 What is the chance that the reason for the increased memory footprint is that the propitiatory WiFi code is causing the blot.  And that the reason for this bloat is espionage? The WiFi code is a feeder system for Chinese Intelligence?  Everyone of these IOT devices could be used as spying devices. 
  Just like how Amazon's Alexa records and forwards conversations.

Chuck.