espressif/arduino-esp32 #1528: Gibberish spurted out onto COM4 when I press the EN button

14geronimo, 2018-06-24:
### Hardware:
Board:							ESP32 DEVKITV1 distributed by  JZK Express Network
Core Installation/update date:			24/nov/2018
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						921600

**Get board info** in Arduino IDE returns:
  BN: Unknown board
  VID: 10C4
  PID: EA60
  SN: Upload any sketch to obtain it

### Description:
I'm only getting gibberish on the Serial Monitor when I press EN pushbutton on the board. I've tried to swap the USB cable, same thing. This is such a basic issue I don't know where to go from here. Many thanks for any pointer you can give me.

The steps I'm following are, from the top:
1. Install Arduino IDE latest version 1.8.5
2. Install latest espressif libraries into Documents\Arduino\hardware\espressif\esp32
3. Restarted the IDE and switched board to ESP Dev Module
4. Connected board to USB on my Windows 10 laptop
5. Switched to COM4
6. Open the Serial Monitor console
7. Press EN... gibbrish galore prints out to Serial Monitor!

The photos and screenshots below are for:
- the board itself
- the options from the Tools menu on the Arduino IDE
- the Serial Monitor message
- the COM Port device information
(also available in [my Dropbox folder](https://www.dropbox.com/sh/ap8qqkg48p5666g/AAAo1RCP6MSIcEScO2yO9yAra?dl=0))

### Board:
[image]
[image]
[image]

### Screenshot (inc. of error):
[image]
[image]

### COM Port device info:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

stickbreaker, 2018-06-24:
@14geronimo set your Serial Monitor Baud rate to 115200

Chuck.

14geronimo, 2018-06-25:
Thank you. I've tried that (and also played with Flash size, frequency, etc.), it doesn't seem to make any change.

Not sure if it's relevant, but I've tried to load the Blink sketch, this is the trace I get, with this "Failed to connect to ESP32: Timed out waiting for packet header" message at the end:

`Error message when compiling/uploading the blinking LED sketch ("C:\Users\nicolas.daniau\Documents\Arduino\mysketches\sketch_101"):

Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, Default, QIO, 40MHz, 4MB (32Mb), 921600, Verbose"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\nicolas.daniau\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\nicolas.daniau\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:PartitionScheme=default,FlashMode=qio,FlashFreq=40,FlashSize=4M,UploadSpeed=921600,DebugLevel=verbose -ide-version=10805 -build-path C:\Users\NICOLA~1.DAN\AppData\Local\Temp\arduino_build_689899 -warnings=none -build-cache C:\Users\NICOLA~1.DAN\AppData\Local\Temp\arduino_cache_323656 -prefs=build.warn_data_percentage=75 -verbose C:\Users\nicolas.daniau\Documents\Arduino\mysketches\sketch_101\sketch_101.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\nicolas.daniau\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\nicolas.daniau\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:PartitionScheme=default,FlashMode=qio,FlashFreq=40,FlashSize=4M,UploadSpeed=921600,DebugLevel=verbose -ide-version=10805 -build-path C:\Users\NICOLA~1.DAN\AppData\Local\Temp\arduino_build_689899 -warnings=none -build-cache C:\Users\NICOLA~1.DAN\AppData\Local\Temp\arduino_cache_323656 -prefs=build.warn_data_percentage=75 -verbose C:\Users\nicolas.daniau\Documents\Arduino\mysketches\sketch_101\sketch_101.ino
Using board 'esp32' from platform in folder: C:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: C:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32
Detecting libraries used...
"C:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spiffs" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-IC:\Users\nicolas.daniau\Documents\Arduino\hardware\espressif\esp32/too [... cut ...] f\esp32/tools/sdk/bin/bootloader_qio_40m.bin 0x10000 C:\Users\NICOLA~1.DAN\AppData\Local\Temp\arduino_build_689899/sketch_101.ino.bin 0x8000 C:\Users\NICOLA~1.DAN\AppData\Local\Temp\arduino_build_689899/sketch_101.ino.partitions.bin 
esptool.py v2.3.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header`

14geronimo, 2018-06-26:
Hooray! Yes indeed changing the baud rate did printing civilised characters instead of gobbledigook, Got confused between 19200 and 115200 earlier.

And secondly holding BOOT at upload time got me through the upload sequence, so victory on all fronts! My LED blinks at will, the world is my oyster.

Thanks to you both @Oh-Doo-Yong and Chuck @stickbreaker 👍 probably trivial stuff when you know but you wouldn't believe how many hours I got stuck scratching my head on this...

closed 2018-06-26 by 14geronimo