espressif/arduino-esp32 #1531: Rebuilding esp-idf libraries for Arduino

HKGK, 2018-06-25:
I need to improve the TCP/WiFi throughput of my ESP32 Arduino project. 

It would seem that I need to make some changes to the esp32/tools/sd/sdkconfig file.

The esp-idf libraries for Arduino are pre compiled. 

How do I recompile these libraries to reflect any changes I make to sdkconfig?

Thanks

lbernstone, 2018-06-25:
https://github.com/espressif/arduino-esp32/issues/1142

lbernstone, 2018-06-26:
You need to copy any libraries that you modify.  That issue was regarding clock speed, which lives only in libesp32.a. You will need liblwip.a if you modify the stack settings.  If you follow those instructions (and compile with the exact same esp-idf files as was used for the latest arduino-esp32), it will not hurt anything to copy the whole set.

ifrew, 2018-06-26:
What's the hold up in getting the idf 3.0+ components into ardunio? Last idf update was over 3 months ago. Anybody know if @me-no-dev is still working on this project? Thanks for any info.

ifrew, 2018-06-28:
Idf just updated yesterday..super!

lbernstone, 2018-06-28:
@ifrew what does your comment have to do with this issue?  Please keep general bitching and moaning on Reddit.
@HKGK If this has solved your issue, please close it.

HKGK, 2018-06-29:
@lbernstone Thanks a lot, got it working. Great care required to synch the arduino and idf setups.

closed 2018-06-29 by HKGK

megazero1316, 2018-09-03:
hello , i am trying to rebuild the idf library but seems like i have some difficults understand how to do i refered to this issues :
[1142](https://github.com/espressif/arduino-esp32/issues/1142) and [1723](https://github.com/espressif/arduino-esp32/issues/1723)

and after i recompile and copy all lib*.a files from build folder, i get an error and can`t compile my code 

i am trying to disable CONFIG_SW_COEXIST_ENABLE for some testing

```
Archiving built core (caching) in: C:\Users\Mohamed\AppData\Local\Temp\arduino_cache_864432\core\core_espressif_esp32_esp32_PSRAM_disabled,PartitionScheme_min_spiffs,FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_c9e2fc0cf67c87988654826a638c7a35.a
Linking everything together...
"C:\\Users\\Mohamed\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc" -nostdlib "-LC:\\Users\\Mohamed\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/lib" "-LC:\\Users\\Mohamed\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\sketch\\advanled-firmware-esp32.ino.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\WiFi\\ETH.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\WiFi\\WiFi.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\WiFi\\WiFiAP.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\WiFi\\WiFiClient.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\WiFi\\WiFiGeneric.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\WiFi\\WiFiMulti.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\WiFi\\WiFiSTA.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\WiFi\\WiFiScan.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\WiFi\\WiFiServer.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\WiFi\\WiFiUdp.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\ESPmDNS\\ESPmDNS.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\EEPROM\\EEPROM.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\async-mqtt-client-master\\AsyncMqttClient.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\async-mqtt-client-master\\AsyncMqttClient\\Packets\\ConnAckPacket.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\async-mqtt-client-master\\AsyncMqttClient\\Packets\\PingRespPacket.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\async-mqtt-client-master\\AsyncMqttClient\\Packets\\PubAckPacket.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\async-mqtt-client-master\\AsyncMqttClient\\Packets\\PubCompPacket.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\async-mqtt-client-master\\AsyncMqttClient\\Packets\\PubRecPacket.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\async-mqtt-client-master\\AsyncMqttClient\\Packets\\PubRelPacket.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\async-mqtt-client-master\\AsyncMqttClient\\Packets\\PublishPacket.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\async-mqtt-client-master\\AsyncMqttClient\\Packets\\SubAckPacket.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\async-mqtt-client-master\\AsyncMqttClient\\Packets\\UnsubAckPacket.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\AsyncTCP-master\\AsyncTCP.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\DNSServer\\DNSServer.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\ESPAsyncWebServer-master\\AsyncEventSource.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\ESPAsyncWebServer-master\\AsyncWebSocket.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\ESPAsyncWebServer-master\\SPIFFSEditor.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\ESPAsyncWebServer-master\\WebAuthentication.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\ESPAsyncWebServer-master\\WebHandlers.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\ESPAsyncWebServer-master\\WebRequest.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\ESPAsyncWebServer-master\\WebResponses.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\ESPAsyncWebServer-master\\WebServer.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\FS\\FS.cpp.o" "C:\\Users\\Mohamed\\AppData\\Local\\Temp\\arduino_build_736639\\libraries\\FS\\vfs_api.cpp.o" "C:\\Users\\Mohamed\\AppData\\ [... cut ...]
C:\Users\Mohamed\AppData\Local\Temp\arduino_build_736639/arduino.ar(Esp.cpp.o): In function `EspClass::flashWrite(unsigned int, unsigned int*, unsigned int)':

C:\Users\Mohamed\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Esp.cpp:249: undefined reference to `spi_flash_write'

C:\Users\Mohamed\AppData\Local\Temp\arduino_build_736639/arduino.ar(Esp.cpp.o): In function `EspClass::flashRead(unsigned int, unsigned int*, unsigned int)':

C:\Users\Mohamed\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Esp.cpp:254: undefined reference to `spi_flash_read'
```

human890209, 2018-09-09:
@megazero1316 @lbernstone I got the same problem as megazero1316.
Hope an expert like lbernstone could help us.

Ben79543, 2020-05-08:
Hi, I am also trying to do the same and all my rebuilt libraries do not work in arduino. 
I need to try what human890209 suggested about the directory locations of .a files.
Meanwhile can someone confirm about issue 1142: 
I checked git log in the arduino I have in the component directory and find several logs about idf version:

`commit 1977370e6fc069e93ffd8818798fbfda27ae7d99
Author: Me No Dev <me-no-dev@users.noreply.github.com>
Date:   Mon Jan 20 22:07:04 2020 +0200
    IDF release/v3.3 (#3339)
    * IDF release/v3.3 46b12a560
    * fix build
    * IDF release/v3.3 367c3c09c`

and also:
`
commit 0d163a1ce274cf389f8a263d66daaaa36b757d6f
Author: Me No Dev <me-no-dev@users.noreply.github.com>
Date:   Thu Sep 5 11:08:45 2019 +0300
    Update IDF 3.2 to 7dd492319 + WiFi fix (#3177)
`
and also: 
`
commit 0acf19af8f5e9e459f9cc39142138d2f5c351146
Author: Me No Dev <me-no-dev@users.noreply.github.com>
Date:   Sun May 12 18:52:23 2019 +0300
    Update IDF to v3.2 977854975 (#2771)
    * Update IDF to v3.2 977854975
    * Update app_httpd.cpp
`
So I am confused, is this arduino at IDF 3.2 or 3.3. So far all my attempts were with an IDF 3.2 but maybe I am wrong? To what commit should I checkout?

Ben79543, 2020-05-08:
So I did a gitcheckout for esp-idf to thte level of "Update IDF 3.2 to 7dd492319 + WiFi fix (#3177)"

After I recompile a "blink project" I got: 
`C:/esp/blink/components/arduino/libraries/WiFi/src/WiFiGeneric.h:31:39: fatal error: wifi_provisioning/manager.h: No such file or directory
compilation terminated.
`

This seems to be related to commit in arduino component, which came AFTER "Update IDF 3.2 to 7dd492319" 

What should I do?

lbernstone, 2020-05-08:
Don't hijack closed issues.

Ben79543, 2020-05-08:
Yes sorry I realised that: I started there as someone else had the same issue: https://github.com/espressif/arduino-esp32/issues/3971