espressif/arduino-esp32 #1534: seek(pos, SeekEnd) fails

mmmilan, 2018-06-25:
Hi,
I've tracked down a file seeking issue to this:

ESP32, vfs_api.cpp:
```
bool VFSFileImpl::seek(uint32_t pos, SeekMode mode)
{
    if(_isDirectory || !_f) {
        return false;
    }
    auto rc = fseek(_f, pos, mode);
    return rc == 0;
}
```
ESP8266, spiffs_api.h:
```
    bool seek(uint32_t pos, SeekMode mode) override
    {
        CHECKFD();

        int32_t offset = static_cast<int32_t>(pos);
        if (mode == SeekEnd) {
            offset = -offset;
        }
        auto rc = SPIFFS_lseek(_fs->getFs(), _fd, offset, (int) mode);
        if (rc < 0) {
            DEBUGV("SPIFFS_lseek rc=%d\r\n", rc);
            return false;
        }

        return true;
    }
```
The ESP32 implementation is missing the position to offset conversion (negation) in SeekEnd mode. Could you fix it, please?

mmmilan, 2019-08-09:
The problem still persists.

```
#ifndef ESP8266
#include "SPIFFS.h"
#endif
#include "FS.h"

const char* testFileName = "/test.txt";

void setup()
{
  Serial.begin(115200);
  Serial.println();
  while (Serial.available())
    Serial.read();
  Serial.println(F("setup begin"));

#ifdef ESP8266
  SPIFFS.begin();
#else
  if (!SPIFFS.begin(true))
    Serial.println(F("begin() failed"));
#endif

  File f = SPIFFS.open(testFileName, "w");
  if (f)
  {
    f.print("abc");
    f.close();
  }
  else
    Serial.println(F("failed to open test file for writing"));
  File r = SPIFFS.open(testFileName, "r");
  if (r)
  {
    r.seek(1, SeekEnd);
    Serial.print(F("position="));
    Serial.println(r.position());
    r.close();
  }
  else
    Serial.println(F("failed to open test file for reading"));
  Serial.println(F("setup end"));
}

void loop()
{  
}
```

The code above results position=2 on ESP8266 and position=3 on ESP32.

atanisoft, 2019-08-09:
@mmmilan please open an issue against [ESP-IDF](https://github.com/espressif/esp-idf) as the arduino side depends on the VFS layer provided by ESP-IDF which you can see the SPIFFS_seek call [here](https://github.com/espressif/esp-idf/blob/release/v3.2/components/spiffs/esp_spiffs.c#L501-L502). Once ESP-IDF has a fix merged for it and it reaches a stable release version it can be picked up by arduino-esp32.

mmmilan, 2019-08-09:
You're referencing code where the offset has a signed type (off_t), not uint32_t, like the Arduino side. It is fine, if it expects negative values for SeekEnd.

atanisoft, 2019-08-09:
@mmmilan The arduino side is a wrapper for the underlying VFS layer, nothing more, nothing less. It will require a fix on the IDF side regardless.

mmmilan, 2019-08-09:
Why isn't the VFS layer supposed to work like that? "The arduino side is a wrapper for the underlying VFS layer" - then again, the error is in the arduino layer, where the offset is defined unsigned.

atanisoft, 2019-08-09:
If you can fix it in the Arduino VFS wrapper send in a PR...

closed 2019-10-22 by stale[bot]