espressif/arduino-esp32 #1540: E (103) spiram: SPI RAM enabled but initialization failed. Bailing out.

tueddy, 2018-06-27:
### Hardware:
Board:							Lolin D32
Core Installation/update date:			27/jun/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
With latest revision i get an error on serial debug output:

**E (103) spiram: SPI RAM enabled but initialization failed. Bailing out.**

### Sketch:
Simple Blink demo

### Debug Messages:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
E (103) spiram: SPI RAM enabled but initialization failed. Bailing out.

Is it an error or can i just ignore the message??
Best regards
Dirk

me-no-dev, 2018-06-27:
just ignore it ;) will change later in IDF to warning so this do not show up.

closed 2018-06-27 by me-no-dev

tueddy, 2018-06-27:
OK, thank's for fast response!
Best Dirk

luc-github, 2018-07-02:
Sorry @me-no-dev issue is closed but not fixed yet - I am correct ? Or is there a workaround ?(other than revert to previous commits) 

With today Git my ESP32 is rebooting in loop with this error so I do not see how to "ignore" it

PrimeTime416, 2021-04-07:
Hello, is there a plan to fix this message : E (620) spiram: SPI RAM enabled but initialization failed. Bailing out