espressif/arduino-esp32 #1542: latest IDF unhandled exception if not esp_wifi_init

tablatronix, 2018-06-27:
esp32 doit

If I call
```
  wifi_config_t conf;
  esp_wifi_get_config(WIFI_IF_STA, &conf);
```

and wifi was not init
Instead of just getting a return ESP_ERR_WIFI_NOT_INIT, I get a exception

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4010cfc6  PS      : 0x00060530  A0      : 0x800d1c99  A1      : 0x3ffb1d30  
A2      : 0x3ffc55b0  A3      : 0x3ffb1dd4  A4      : 0x00000000  A5      : 0x00000003  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb1d20  
A10     : 0x00000002  A11     : 0x3f40108b  A12     : 0x3ffc1074  A13     : 0x00000003  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000138  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

Backtrace: 0x4010cfc6:0x3ffb1d30 0x400d1c96:0x3ffb1d60 0x4012d0b3:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6704
entry 0x400802e4
E (194) spiram: SPI RAM enabled but initialization failed. Bailing out.
```

tablatronix, 2018-06-28:
same for everything I tested

esp_wifi_disconnect() == ESP_OK;

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4011771c  PS      : 0x00060930  A0      : 0x800d86ce  A1      : 0x3ffb1d30  
A2      : 0x3ffc586c  A3      : 0x00000003  A4      : 0x3f4020a1  A5      : 0x3f40106c  
A6      : 0x3ffc3d74  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffb1d00  
A10     : 0x3ffb1e50  A11     : 0x00000003  A12     : 0x3f4020a1  A13     : 0x3f40106c  
A14     : 0x3ffc3d74  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000138  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x4011771c:0x3ffb1d30 0x400d86cb:0x3ffb1d60 0x400dc8f5:0x3ffb1d80 0x400d2dbd:0x3ffb1dc0 0x40136e17:0x3ffb1fa0

Rebooting...

```

JonSilver, 2018-11-06:
Is there any recommended workaround to prevent this exception occurring on `esp_wifi_get_config`?

tablatronix, 2018-11-06:
make sure wifi is init first, so do something that does it, setmode etc

copercini, 2018-11-06:
you can check if the the wifi was init before do something:

```cpp
   wifi_mode_t currentMode;
   if(!esp_wifi_get_mode(&currentMode) == ESP_ERR_WIFI_NOT_INIT)
   {
      //do something
   }
```

But as @lbernstone said, this is an IDF code

JonSilver, 2018-11-06:
@tablatronix 

> make sure wifi is init first, so do something that does it, setmode etc

Tried `esp_wifi_set_mode(WIFI_MODE_NULL)` and `esp_wifi_set_mode(WIFI_MODE_STA)` before the call but with no difference. I'll go ask the IDFers.

closed 2019-08-15 by stale[bot]