espressif/arduino-esp32 #1550: Data on Serial2, even though not initialized

thomasklingbeil, 2018-06-28:
### Hardware:
Board:							DoIT ESP32
Core Installation/update date:			27/jun/2018 (a59eafbc9dfa3ce818c110f996eebf68d755be24)
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
For our application, we have used RX2 and TX2 on the DoIT ESP32 board (GPIO 16&17) for other purposes. This worked until recently, but not it seems, there is serial data being sent on those pins, even though I haven't initialized anything (i.e. I use an empty sketch).

I understand that there is the boot information on the first serial port, with which I am fine. But is there any way to prevent action on the other serial ports?

### Sketch:
```cpp
void setup() {
}

void loop() {
}
```

hex86, 2018-06-28:
I tried updating my arduino-esp32 yesterday and also had significant problems with the hardwareserial 2 not working under conditions like the USB being connected but would immediately start working when unplugged and running off battery power on the adafruit feather. Had to roll back to a much older version for my projects to work as expected again.

thomasklingbeil, 2018-06-29:
I did another little test today. Connected two LEDs to RX2/TX2 (i.e. GPIO16/17) and 3.3V with the sketch below.

On the most recent version RX2 seems to stay `HIGH` and TX2 `LOW` (only during boot both seem to be high(ish)).

[image]

With an older version of arduino-esp32 everything works as I expected it to work.

[image]

```cpp
void setup() {
  pinMode(16, OUTPUT);
  pinMode(17, OUTPUT);
}

void loop() {
  digitalWrite(16, HIGH);
  digitalWrite(17, LOW);

  delay(500);
  digitalWrite(16, LOW);
  digitalWrite(17, HIGH);

  delay(500);
  }
```

me-no-dev, 2018-06-29:
hmm... I will look into this and see what IDF guys think of it

thomasklingbeil, 2018-07-02:
Thank you! @me-no-dev: Is there anything I can do to support?

closed 2019-08-15 by stale[bot]