espressif/arduino-esp32 #1555: How to disable SPI RAM?

epicface2304, 2018-06-28:
Hello, I am having some issues with my ESP32.

I am trying to use it with a 7.5" waveshare ePaper/eInk display and it is not working with the example code, I have tried the code with the 2 boards I have and both get the same error during boot. The code shows debug messages like it is working, but the display is not updating so I think this error has to do with it. 

`load:0x40080000,len:6696
entry 0x400802e4
E (129) spiram: SPI RAM enabled but initialization failed. Bailing out.`

I am using the ESP32 DOIT DevKit v1 with the Arduino IDE on Windows.

I did some research and it looks like I need to disable the SPI RAM. How do I disable it?

lbernstone, 2018-06-28:
Nope, this error has nothing to do with it.  It should be a warning, not an error.  At the moment, it will come up on every device that doesn't have PSRAM (the WROVER kit). 
Make sure you have the right driver for your particular device, they are not all the same.

epicface2304, 2018-06-28:
Wait... but my board has the ESP-WROOM-32 module on it. It isn't one of the WROVERs.

closed 2018-06-28 by epicface2304