espressif/arduino-esp32 #1559: issues on Update IDF to aaf1239

dsptech, 2018-06-28:
### Hardware:
Board:							Heltec_Wifi_kit32?
Core Installation/update date:			28/06/18
IDE name:							other
Flash Frequency:					40Mhz
Upload Speed:						115200?

### Description:  Compiler error after the last update.

The previouos update was done two weeks ago and everything was working fine. Few minutes ago I downloaded and installed the last update (idf update included) and now I'm not able to compile my projects due an undefined refererence in the precompiled library. This is the message produced by the linker:
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 
'Starting combiner'
"C:/Program Files (x86)/Arduino/hardware/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc" -nostdlib "-LC:/Program Files (x86)/Arduino/hardware/esp32/tools/sdk/lib" "-LC:/Program Files (x86)/Arduino/hardware/esp32/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group  -Xlinker -Map=maps.map   .\ESP32_A_BMP280.cpp.o .\ESP32_A_CommonInterface.cpp.o .\ESP32_A_Display.cpp.o .\ESP32_A_FlashSettings.cpp.o .\ESP32_A_NetReceive.cpp.o .\ESP32_A_TCP_Serv.cpp.o .\ESP32_A_WiFi.cpp.o .\ESP32_A_remoteBmp280.cpp.o .\ESP_MPU9250.cpp.o .\MyEspSerial.cpp.o .\NetClient.cpp.o .\NetServer.cpp.o .\sloeber.ino.cpp.o  .\libraries\gitBMP280_driver\bmp280.c.o  .\libraries\git-esp8266-oled-ssd1306\OLEDDisplay.cpp.o .\libraries\git-esp8266-oled-ssd1306\OLEDDisplayUi.cpp.o  .\libraries\WiFi\src\ETH.cpp.o .\libraries\WiFi\src\WiFi.cpp.o .\libraries\WiFi\src\WiFiAP.cpp.o .\libraries\WiFi\src\WiFiClient.cpp.o .\libraries\WiFi\src\WiFiGeneric.cpp.o .\libraries\WiFi\src\WiFiMulti.cpp.o .\libraries\WiFi\src\WiFiSTA.cpp.o .\libraries\WiFi\src\WiFiScan.cpp.o .\libraries\WiFi\src\WiFiServer.cpp.o .\libraries\WiFi\src\WiFiUdp.cpp.o  

_(omitted a big private libreries list to reduce the size of this message)_

 .\libraries\EEPROM\EEPROM.cpp.o   "C:\MyPRJ\slob420\ESP32_A/Release/arduino.ar" -lgcc -lopenssl -lbtdm_app -lfatfs -lwps -lcoexist -lwear_levelling -lhal -lnewlib -ldriver -lbootloader_support -lpp -lmesh -lsmartconfig -ljsmn -lwpa -lethernet -lphy -lapp_trace -lconsole -lulp -lwpa_supplicant -lfreertos -lbt -lmicro-ecc -lcxx -lxtensa-debug-module -lmdns -lvfs -lsoc -lcore -lsdmmc -lcoap -ltcpip_adapter -lc_nano -lrtc -lspi_flash -lwpa2 -lesp32 -lapp_update -lnghttp -lspiffs -lespnow -lnvs_flash -lesp_adc_cal -llog -lexpat -lm -lc -lheap -lmbedtls -llwip -lnet80211 -lpthread -ljson -lstdc++ -Wl,--end-group -Wl,-EL -o "C:\MyPRJ\slob420\ESP32_A/Release/ESP32_A.elf" C:/MyPRJ/slob420/ESP32_A/Release/arduino.ar
C:/Program Files (x86)/Arduino/hardware/esp32/tools/sdk/lib\libesp32.a(wifi_os_adapter.o):(.iram1.literal+0xc): undefined reference to `sc_ack_send'
C:/Program Files (x86)/Arduino/hardware/esp32/tools/sdk/lib\libesp32.a(wifi_os_adapter.o): In function `semphr_take_from_isr_wrapper':
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/wifi_os_adapter.c:221: undefined reference to `sc_ack_send'
C:/Program Files (x86)/Arduino/hardware/esp32/tools/sdk/lib\libesp32.a(wifi_os_adapter.o):(.data.g_wifi_osi_funcs+0x17c): undefined reference to `sc_ack_send_stop'
collect2.exe: error: ld returned 1 exit status
makefile:96: recipe for target 'ESP32_A.elf' failed
make: *** [ESP32_A.elf] Error 1

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 
Is someone experiencing this issue ?
Regards

me-no-dev, 2018-06-28:
make clean?

dsptech, 2018-06-28:
Fully cleaned before recompiling (any object files folder and related was also manually deleted and also all temporary files).

For testing purpose, I also rolled back to the commit 7abd586 and my projects now are succefully compiling again.

dsptech, 2018-06-29:
Update:
Now I applied only the update: a59eafbc9dfa3ce818c110f996eebf68d755be24 and the issue now is on again.
I also see (from idf sources) that the missing reference error came from the same binary library where the symbols should be declared (libesp32.a) , so, it's very strange.

RosCstr, 2018-06-29:
I also updated the /arduino-esp32 libs and I get the same errors at compiling:

.../arduino-esp32/tools/sdk/lib\libesp32.a(wifi_os_adapter.o):(.iram1.literal+0xc): undefined reference to `sc_ack_send'
.../arduino-esp32/tools/sdk/lib\libesp32.a(wifi_os_adapter.o): In function `semphr_take_from_isr_wrapper':
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/wifi_os_adapter.c:221: undefined reference to `sc_ack_send'
.../arduino-esp32/tools/sdk/lib\libesp32.a(wifi_os_adapter.o):(.data.g_wifi_osi_funcs+0x17c): undefined reference to `sc_ack_send_stop'
collect2.exe: error: ld returned 1 exit status
makefile:84: recipe for target 'test2.elf' failed
make: *** [test2.elf] Error 1

dsptech, 2018-06-29:
Hi,
Seems to me that the smartconfig source cose is not included in the precompiled library. The missing symbols are declared here.

william-ferguson-au, 2018-07-02:
This is a problem. Master is currently broken because of this.

me-no-dev, 2018-07-04:
ok will look into the libs today and see where smartconfig went.

me-no-dev, 2018-07-04:
I just checked the examples and we do have a smartconfig example in the repo. If it's missing as you say, how did that example compile? Can you give some minimal example sketch that triggers this issue?

dsptech, 2018-07-04:
Hi,
I'm not using smartconfig at all. I had only seen that the missed functions sc_ack_send and sc_ack_send_stop are declared in smartconfig_ack.h and implemented in smartconfig_ack.c.

I'm not in my developing machine now (and I'm unable to perform checks today) but, regarding networking, my project is using only : WiFi.mode(WIFI_STA),  WiFi.scanNetworks, WiFi.disconnect, WiFi.begin,  esp_wifi_set_country and the directly the lwIP netconn APIs (for UDP packets). Any reconnect or autoconnect feature is disabled.

No other networking APIs are used (could this cause the issue ?).

I rolled back to the previous commit again, so, creating an example sketch will require a bit of time. I hope to post it tomorrow in the evening.

.... I will try also to compile the smartconfig example (just in case... )

Thank you.

dsptech, 2018-07-05:
Hi @me-no-dev ,
I can confirm the missing smartconfig, but it is not a library issue. It is a my IDE issue.
I performed several checks and I discovered that my IDE (ecplise based - sloeber) does not recognize  the changes of the include list in platform.txt (even forcing a refresh and cleaning any compiler created files too)
So, the new directory smartconfig_ack was not handled in compile and the linker error is produced.

Seems that the only way to force a new platform.txt scan is to choose another board family (example: Arduino Nano and, after that, choose again the proper ESP32 board).
Other users that are using Ardino IDE have posted similar messages, so, I suspect that the problem is the same.

... I'm sorry to have bothered you for this non-issue.

closed 2018-07-12 by dsptech