espressif/arduino-esp32 #1560: HTTPClient can't work in AP mode

Cooper-WJH, 2018-06-29:
### Hardware:
Board:							ESP32-PICO-KIT V4
Core Installation/update date:		Null
IDE name:						Eclipse
Flash Frequency:					40Mhz
Upload Speed:					921600

### Description:
Hi

I can use HTTPClient GET & POST in STA mode. But when ESP32 work in AP mode, I can't sent GET & POST request to the STA node which connects to AP and got IP.

Can't HTTPClient GET & POST work in AP mode?

everslick, 2018-06-29:
AFAIK, there should be absolutely no difference for HTTPCLient. Are you sure you connect to the right IP (URL)? Is the webserver running on the ESP that connects to your ESP in AP mode?

copercini, 2018-07-01:
What is your test scenario ?

sysizlayan, 2018-07-03:
If your case is just AP mode, it is natural because in AP mode, esp32 waits for some client to connect itself.

Try AP_STA mode instead and be sure you have supplied necessary SSID and password for internet connection.

closed 2019-08-15 by stale[bot]