espressif/arduino-esp32 #1567: about esp32 in fritzing.org

dimension1991, 2018-07-01:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

copercini, 2018-07-01:
what is your question?

dimension1991, 2018-07-01:
Sorry .. Actually my question is how do I get library parts of esp32 30 pin
 for fritzing .. I did search for the library and only found the esp32 with
38 pin ..

lbernstone, 2018-07-02:
http://fritzing.org/learning/tutorials/creating-custom-parts/

closed 2019-08-15 by stale[bot]