espressif/arduino-esp32 #1571: 'clone or download'-button  not downloading all files

TTKKu, 2018-07-02:
Hello,

I have now used the arduino-esp32 with succes for the past few months. However, after a new install, it did not seem to work anymore. It is my feeling that when I click the 'clone or download'-button in github, not all files are downloaded. for example, at least the following folders are empty: 
arduino-esp32-master\libraries\AzureIoT
arduino-esp32-master\libraries\BLE

A friend of mine had the same, but when he used the git-program to download it worked. I can't use that however.

Although I could do something wrong, I don't think I am (I had it working several times in the past).
Is this something wrong with Github? Can somebody explain to me how/where I can download correctly?

regards,

T

dsptech, 2018-07-02:
Hi,
I experienced the same issue about AzureIoT

per1234, 2018-07-02:
It's not a problem with GitHub or with this repository. Those folders are Git [submodules](https://blog.github.com/2016-02-01-working-with-submodules/). 'clone or download' does not populate submodules. That's why it worked for your friend and why the instructions tell you to use Git.
https://github.com/espressif/arduino-esp32#installation-instructions

If you absolutely can't follow the official installation instructions then you will need to download each of the submodules separately and then copy their files to the correct places.

dsptech, 2018-07-02:
Hi,
I was totally unaware of that. 
Thank you

TTKKu, 2018-07-02:
I did not know that either. thank you for explaining. However, still strange that I got it working in the past, without using git-gui or similar.

everslick, 2018-07-05:
The repo only contains submodules since a few days. That's why it used to work before.

closed 2018-07-05 by everslick