espressif/arduino-esp32 #1575: espressif: unknown package

hashimalkhamis, 2018-07-02:
I have done all the procedure to install the library and board of esp32 on my PC but whenever I try to compile the sketch it gives my this error

espressif: unknown package

what can I do to solve this issue ?

lbernstone, 2018-07-02:
Installation instructions will be changing very soon (the next day or two).  If you are familiar with Arduino, you can now add https://dl.espressif.com/dl/package_esp32_dev_index.json into the board manager.

beegee-tokyo, 2018-07-02:
Finally!
Will try tomorrow and if it works will add it to the [Wiki](https://desire.giesecke.tk/index.php/2018/01/30/esp32-wiki-entries/)

lbernstone, 2018-07-02:
It worked for me.  No real need for Wiki, since it will be the primary install method, but if you want to write the doco, it will be needed very soon.

hashimalkhamis, 2018-07-02:
works perfect thnx

closed 2018-07-02 by hashimalkhamis