espressif/arduino-esp32 #1577: Missing global instance of 'Serial1'; is one needed?

mrwgx3, 2018-07-03:
### Basic Infos

#### Hardware

Adafruit HUZZAH32 - ESP32 Feather Board  
https://www.adafruit.com/product/3405

Adalogger FeatherWing - RTC + SD Add-on For All Feather Boards  
https://www.adafruit.com/product/2922

Adafruit Ultimate GPS FeatherWing  
https://www.adafruit.com/product/3133

FeatherWing Tripler Mini Kit - Prototyping Add-on For Feathers  
https://www.adafruit.com/product/3417

#### OS, IDE, and Commit
``` 
OS:       Windows 10 Professional
IDE:      1.8.5
COMMIT?:  via 'git branch -v'
          * master b855eb2 * arduino-esp32 release mgmt. update (#1557)
```
### Settings in IDE
```
Module:            Adafruit ESP32 Feather
CPU Frequency:     80Mhz  
Core Debug Level:  None
```

### Description

While bringing up the forementioned hardware assembly, my colleague and I found that the [basic GPS Rx/Tx test sketch][rx_tx_test] would not compile for the ESP32. The cause of the problem is due to a missing global instance of 'HardwareSerial' named 'Serial1' (see adafruit/Adafruit_GPS#74).

The comments regarding the rx/tx test sketch implies that a global 'Serial1' should exist for multiple Arduino boards; they do exist within the Arduino/ESP8266 repository in core files ['HardwareSerial.cpp'][esp8266_hws_cpp] and ['HardwareSerial.h'][esp8266_hws_h].

The corresponding 'Serial1' definitions, however, cannot be found within the Arduino/ESP32 repository. [(a local instance can be found HERE)][serial1_instance].

Does a global instance of 'Serial1' need to be defined for the Arduino/ESP32 repository? Are there good reasons for not doing so?

[rx_tx_test]:
https://learn.adafruit.com/adafruit-ultimate-gps-featherwing/basic-rx-tx-test "Basic Rx/Tx GPS test"

[esp8266_hws_cpp]:
https://github.com/esp8266/Arduino/blob/3a110aa698dd17e0d80704db47c8e43c986b0f86/cores/esp8266/HardwareSerial.cpp#L111#L116 "ESP8266 'HardwareSerial.cpp'"

[esp8266_hws_h]:
https://github.com/esp8266/Arduino/blob/3a110aa698dd17e0d80704db47c8e43c986b0f86/cores/esp8266/HardwareSerial.h#L187#L196 "ESP8266 'HardwareSerial.h'"

[serial1_instance]:
https://github.com/espressif/arduino-esp32/blob/432bcf5a0aa0d234346c22598bea8e248c156c79/libraries/WiFi/examples/WiFiTelnetToSerial/WiFiTelnetToSerial.ino#L32#L36  "Instance for'Serial1'"

mrwgx3, 2018-07-03:
Just noted there are (3) UARTs defined for the ESP32. I do see the potential naming conflicts...

lbernstone, 2018-07-03:
The global instance is just Serial, and is tied to UART0.  It lives on GPIO 1 & 3.  To use other HardwareSerial instances, you will typically want to specify the pins used.  Since many esp32 projects are not going to use additional serial comms, why allocate the limited memory to that?

mrwgx3, 2018-07-03:
@lbernstone 
```
"Since many esp32 projects are not going to use additional serial comms,
 why allocate the limited memory to that?"
```
Elaborating, does the Arduino development environment expect global instances be defined for multiple serial ports if they exist in hardware? Given that the ESP32 has (3) uarts, I would have expected a global "Serial', 'Serial1', and 'Serial2' to have be defined in order to maintain consistency with other board realizations.  

As  most boards having multiple UARTs only have (2), 'Serial1' by default seems to be designator for the auxiliary serial port. As the Adafruit ESP32 feather uses UART2 for its aux. serial port, I would find it confusing (esp. as a beginner) to see the following code:
```
HardwareSerial   Serial  ( 0 );      // Default port     Hence my comment
HardwareSerial   Serial1 ( 2 );      // 1st aux. port    about potential naming
HardwareSerial   Serial2 ( 1 );      // 2nd aux. port    conflicts
```
My impression is that tradition would dictate global instances, but current coding does not do so. Believing that this is not a simple oversight, I'm asking if there is a compelling (beyond potential naming conflicts) to do so. As I'm not versed in the design philosophy behind the Arduino environment, I cannot answer the question for myself.

me-no-dev, 2018-07-03:
Done: https://github.com/espressif/arduino-esp32/commit/4e5cbdaa7f4ae6d4f76a378e010d8979da9532bf

closed 2018-07-03 by me-no-dev