espressif/arduino-esp32 #1578: arduino How to upgrade the compiled firmware

1163167506, 2018-07-03:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
arduino How to upgrade the compiled firmware 
like test.ino.esp32.bin

karan6190, 2018-07-03:
So are you talking about an upgrade to bin file or what ??

lbernstone, 2018-07-03:
The correct answer: Get the source code and compile/flash.
How I would do it: `python esptool.py write_flash 0x10000 ./test.ino.esp32.bin`
There are a ton of assumptions in that, mainly that you are using the default partition scheme and the bootloader on your device matches what that bin expects.

closed 2018-07-17 by 1163167506