espressif/arduino-esp32 #1585: cannot find -lstdc++; MACOS

js-bone21, 2018-07-03:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:	HELTEC LORA 32				?ESP32 Dev Module?
Core Installation/update date:03/Jul/2018	?11/jul/2017?
IDE name:	Arduino					?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:	N/A				?40Mhz?
Upload Speed:	N/A					?115200?

### Description:
On attempting to compile in Arduino IDE (without flashing), gets through everything except linking...

Error shows cannot find -lstdc++, which I've been digging around and cannot find a satisfactory answer for macos.

What I've been able to gather is that it's an issue with the compiler mixing 32bit and 64bit code? Not clear enough on what is causing it or why...Thank you in advance

### Sketch:
```cpp

/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleServer.cpp
    Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class MySecurity : public BLESecurityCallbacks {

	uint32_t onPassKeyRequest(){
        ESP_LOGI(LOG_TAG, "PassKeyRequest");
		return 123456;
	}

	void onPassKeyNotify(uint32_t pass_key){
        ESP_LOGI(LOG_TAG, "On passkey Notify number:%d", pass_key);
	}

	bool onSecurityRequest(){
	    ESP_LOGI(LOG_TAG, "On Security Request");
		return true;
	}

	void onAuthenticationComplete(esp_ble_auth_cmpl_t cmpl){
		ESP_LOGI(LOG_TAG, "Starting BLE work!");
		if(cmpl.success){
			uint16_t length;
			esp_ble_gap_get_whitelist_size(&length);
			ESP_LOGD(LOG_TAG, "size: %d", length);
		}
	}
};

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("ESP32");
[... 150 lines cut ...]
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/wiring_pulse.c.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/wiring_shift.c.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/libb64/cdecode.c.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/libb64/cencode.c.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/Esp.cpp.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/HardwareSerial.cpp.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/IPAddress.cpp.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/IPv6Address.cpp.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/MD5Builder.cpp.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/Print.cpp.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/Stream.cpp.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/StreamString.cpp.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/WMath.cpp.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/WString.cpp.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/base64.cpp.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/cbuf.cpp.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar" cru  "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/arduino.ar" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/core/main.cpp.o"
Archiving built core (caching) in: /var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_cache_198384/core/core_espressif_esp32_heltec_wifi_lora_32_FlashFreq_80,UploadSpeed_115200_f1a0cf8bd51ee50ad93ea98889069a9c.a
Linking everything together...
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc" -nostdlib "-L/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/lib" "-L/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority  -u __cxa_guard_dummy -u __cxx_fatal_exception  -Wl,--start-group "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/sketch/BLE_server_passkey.ino.cpp.o" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/libraries/BLE/BLE2902.cpp.o" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/libraries/BLE/BLE2904.cpp.o" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/libraries/BLE/BLEAddress.cpp.o" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/libraries/BLE/BLEAdvertisedDevice.cpp.o" "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/libraries/BLE/BLEAdvertisin [... cut ...] lders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_118254/BLE_server_passkey.ino.elf"
/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lstdc++
collect2: error: ld returned 1 exit status
Using library BLE at version 0.4.12 in folder: /Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/libraries/BLE 
exit status 1
Error compiling for board Heltec_WIFI_LoRa_32.```

me-no-dev, 2018-07-03:
libstdc++ is in the toolchain.maybe something went wrong during install? you can rerun `python get.py` again to reinstall it. Maybe delete the `tools/dist` folder first to clear the cache.

js-bone21, 2018-07-04:
Removed and re-installed. New error...along with a lot of warnings (not clear why it's occurring...)

Had to truncate some of the lesser-important lines because of max comment length (65536 characters)

(Almost) Full output:
```
Arduino: 1.8.5 (Mac OS X), Board: "Heltec_WIFI_LoRa_32, 80MHz, 115200"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/joshsoucie/Library/Arduino15/packages -hardware /Users/joshsoucie/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/joshsoucie/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/joshsoucie/Documents/Arduino/libraries -fqbn=espressif:esp32:heltec_wifi_lora_32:FlashFreq=80,UploadSpeed=115200 -ide-version=10805 -build-path /var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_568049 -warnings=all -build-cache /var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_cache_225798 -prefs=build.warn_data_percentage=75 -verbose /Users/joshsoucie/Documents/Arduino/esp32-snippets/cpp_utils/tests/BLETests/Arduino/security/BLE_server/BLE_server_passkey/BLE_server_passkey.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/joshsoucie/Library/Arduino15/packages -hardware /Users/joshsoucie/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/joshsoucie/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/joshsoucie/Documents/Arduino/libraries -fqbn=espressif:esp32:heltec_wifi_lora_32:FlashFreq=80,UploadSpeed=115200 -ide-version=10805 -build-path /var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_568049 -warnings=all -build-cache /var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_cache_225798 -prefs=build.warn_data_percentage=75 -verbose /Users/joshsoucie/Documents/Arduino/esp32-snippets/cpp_utils/tests/BLETests/Arduino/security/BLE_server/BLE_server_passkey/BLE_server_passkey.ino
Using board 'heltec_wifi_lora_32' from platform in folder: /Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32
Detecting libraries used...
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-

...

DF_CPU=240000000L -DARDUINO=10805 -DARDUINO_Heltec_WIFI_LoRa_32 -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="Heltec_WIFI_LoRa_32" -DARDUINO_VARIANT="heltec_wifi_lora_32"  -DESP32 -DCORE_DEBUG_LEVEL=0  "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/cores/esp32" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/variants/heltec_wifi_lora_32" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/libraries/BLE/src" "/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/libraries/BLE/src/BLEServer.cpp" -o "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_568049/libraries/BLE/BLEServer.cpp.o"
/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/libraries/BLE/src/BLEServer.cpp:26:20: warning: 'LOG_TAG' defined but not used [-Wunused-variable]
 static const char* LOG_TAG = "BLEServer";
                    ^
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid/api" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_trace" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_adc_cal" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_http_client" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/esp-tls" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/heap" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/jsmn" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/inclu [... cut ...] cie/Documents/Arduino/hardware/espressif/esp32/cores/esp32/cbuf.cpp" -o "/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_568049/core/cbuf.cpp.o"
"/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32

stickbreaker, 2018-07-04:
@js-bone21 
>"/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_568049/BLE_server_passkey.ino.elf"
>
>/var/folders/04/dkvrmgp53qgfpkfv5fyz1ksh0000gn/T/arduino_build_568049/sketch/BLE_server_passkey.ino.cpp.o:(.rodata._ZTV10MySecurity[vtable for MySecurity]+0x20): undefined reference to `BLESecurityCallbacks::onConfirmPIN(unsigned int)'
>
>collect2: error: ld returned 1 exit status
Using library BLE at version 0.4.16 in folder: /Users/joshsoucie/Documents/Arduino/hardware/espressif/esp32/libraries/BLE 
exit status 1
Error compiling for board Heltec_WIFI_LoRa_32.

This looks to be the problem:
> undefined reference to `BLESecurityCallbacks::onConfirmPIN(unsigned int)'

Chuck.

stickbreaker, 2018-07-04:
@js-bone21 The warnings are popping up because you have:
> -warnings=all

in your compile options

And you have a few unused variables.

Chuck.

js-bone21, 2018-07-04:
@stickbreaker

Sure, I understand that this line indicates the problem. I’m not sure how to rectify the issue in a gcc temporary compiler file...or, if it’s solvable in ‘BLE_server_passkey.ino’/the cpp file which I obviously determined (perhaps incorrectly, and if so, my apologies in advance) that an issue in ‘BLE_server_passkey.ino’ was an issue in this repo. 

As far as warnings, I understand why, in the grand scheme of things, warnings are showing and that this variable appears to be unused and that it’s not really a big deal...just throwing annoying orange text into the otherwise wonderful compiler stream. I’m aware that I’ll have to bring it to the attention of nkolban in the esp-snippets repo. Can’t say the same about the other file.

stickbreaker, 2018-07-04:
@js-bone21 Most of those warning probably result from conditional compile issues. Those variable declarations need to be  bounded by:
```c++
#if ARDUHAL_LOG_LEVEL >= ARDUHAL_LOG_LEVEL_ERROR
// variable only used if CORE DEBUG LEVEL greater than or equal to ERROR
#endif
```

You'll have to track down the source files and create a PR with fixes if the warning message bother you.

Chuck.

js-bone21, 2018-07-04:
> it’s not really a big deal

I'm more interested in getting this example to compile properly on my machine. But thanks.

js-bone21, 2018-07-23:
I think I fixed it...not sure what changed...but I reinstalled both the Arduino IDE and arduino-esp32 core. Also removed esp32-related files in ~/Users/[username]/Library/Arduino15

1. ```sudo rm -rf ~/Documents/Arduino/hardware/espressif```
2. remove all esp32-related files in ```~/Users/[username]/Library/Arduino15```
3. remove arduino
4. Reinstall Arduino IDE
5. Reinstall arduino-esp32 following directions here: [https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/mac.md](url)

Still getting warnings about unused variables but that's fine. Not my code.

closed 2018-07-23 by js-bone21