espressif/arduino-esp32 #1586: Incorrect 'SS' pin assignment, variant file for Adafruit ESP32 Feather Board' ?

mrwgx3, 2018-07-03:
### Basic Infos

#### Hardware

Adafruit HUZZAH32 - ESP32 Feather Board  
https://www.adafruit.com/product/3405

Adalogger FeatherWing - RTC + SD Add-on For All Feather Boards  
https://www.adafruit.com/product/2922

Adafruit Ultimate GPS FeatherWing  
https://www.adafruit.com/product/3133

FeatherWing Tripler Mini Kit - Prototyping Add-on For Feathers  
https://www.adafruit.com/product/3417

#### OS, IDE, and Commit
``` 
OS:       Windows 10 Professional
IDE:      1.8.5
COMMIT?:  via 'git branch -v'
          * master b855eb2 * arduino-esp32 release mgmt. update (#1557)
```
### Settings in IDE
```
Module:            Adafruit ESP32 Feather
CPU Frequency:     80Mhz  
Core Debug Level:  None
```

### Description

While bringing up the forementioned hardware assembly, my colleague and I found we could not compile the 'CardInfo' example sketch contained within the Arduino 'SD' library:
```
Arduino: 1.8.5 (Windows 10), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None"
CardInfo:27: error: 'Sd2Card' does not name a type
 Sd2Card card;
...
...
```

Pressing onward, we next tried the ['SD_Test.ino'][sd_test_ino] example sketch within the SD(esp32) library. The sketch compiled OK, but when ran produced a 'Card Mount Failed' message. Suspecting an incorrect SD card chip-select assignment, we next looked at the ['pins_arduino.h' variant file][sd_ss_variant] defined for the 'ESP32 Feather' board:
```
static const uint8_t SS = 2;
```
The [Adafruit website for the RTC/SD board][using_sd_card] identifies GPIO pin 33 as the required pin, and changing the variant file's SS pin definition to this value resolved the issue.

Is the 'SS=33' assignment for the RTC/SD/ESP32 feather board combination a once-only assignment, or can the change be made permanent within the variant file?

Copy Author: @ladyada

[sd_test_ino]:
https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/examples/SD_Test/SD_Test.ino

[using_sd_card]:
https://learn.adafruit.com/adafruit-adalogger-featherwing/using-the-sd-card  "Using the SD card"

[sd_ss_variant]:
https://github.com/espressif/arduino-esp32/blob/master/variants/feather_esp32/pins_arduino.h#L23#L23 "Default SS pin assignment"

me-no-dev, 2018-07-03:
It works either way :) but since we are talking about a wing, it makes no sense to define this in the board's pin definition.
In your code you should use `SD.begin(33)` to mount SD :)

Oh and the original Arduino SD library does not work here.

closed 2018-07-03 by me-no-dev

mrwgx3, 2018-07-04:
Thanks @me-no-dev  for your comments. Copying @ladyada ...
```
Oh and the original Arduino SD library does not work here.
```
We were just following the instructions given us on the Adafruit website to verify card functioning. See ['Get Card Info' section on this webpage][using_sd_card]
```
It works either way :) but since we are talking about a wing, it makes no
sense to define this in the board's pin definition.
```
After studying the schematics for the wings supporting an SD card, I agree with you; the SD_CS pin jumps around. The current SS definition given in the variant file (GPIO2), however, happens to be a no-connect at the ESP32 processor. This would be the pin I'd pick to "do no harm", but it guarantees the example sketches 'SD_Test.ino' and 'SD_time.ino' won't function with the ESP32 Feather as coded.
```
In your code you should use SD.begin(33) to mount SD :)
```
That would be one way to get 'SD_Test.ino' and 'SD_time.ino' working again:
```
#define   mySDSS   SS      // Default, change as required
...
    if (!SD.begin( mySDSS )) {
        Serial.println("Card Mount Failed");
        return;
}
```
Another alternative would be for Adafruit to write an 'CardInfo' sketch for the ESP32.

[using_sd_card]:
https://learn.adafruit.com/adafruit-adalogger-featherwing/using-the-sd-card  "Using the SD card"

me-no-dev, 2018-07-04:
While it made sense to use the standard SD lib on ESP8266, it does not make much sense to do it on ESP32 (we have more integrated file system, sdmmc and more). If "GetCardInfo" is an issue, then maybe yes, writing a compatible sketch is a good idea.
If SS is defined as IO2 in the feather definitions and that pin is not broken out, then sure it's not a bad idea to be changed to a pin that is actually exposed.

mrwgx3, 2018-07-13:
```
While it made sense to use the standard SD lib on ESP8266, it does not make much
sense to do it on ESP32 (we have more integrated file system, sdmmc and more).
If "GetCardInfo" is an issue, then maybe yes, writing a compatible sketch is a good idea.
If SS is defined as IO2 in the feather definitions and that pin is not broken out, then
sure it's not a bad idea to be changed to a pin that is actually exposed.
```
@ladyada,
Let me know about any changes you want to make to the 'pins_arduino.h' variant file for the feather-esp32 board (SS = 33,for example). Thnks