espressif/arduino-esp32 #1589: i can't update it

Master811129, 2018-07-04:
hi i tried to update my arduino core according from Installation Instructions for windows but it didn't update.
it didn't add any new file to my computer (OTAWebUpdater) or new board (WiPy 3.0) in IDE.
please help.

me-no-dev, 2018-07-04:
what did you do exactly? Also you can look up on google for "git pull latest changes" or similar.

alopexc0de, 2018-07-05:
This seems related to the problem I'm experiencing.

I am trying to install this library into Arduino on my Mac, and each step is successful, up until I have to run "get.py". The error message that I ultimately recieve is "IOError: [Errno socket error] [Errno 60] Operation timed out"

In Arduino preferences, my sketchbook path is indeed ~/Documents/Arduino.

Here is the command in my terminal:

> [c0de@c0defox-MacBook] [20:21:52]
>  [~]> mkdir -p ~/Documents/Arduino/hardware/espressif && \
> cd ~/Documents/Arduino/hardware/espressif && \
> git clone https://github.com/espressif/arduino-esp32.git esp32 && \
> cd esp32 && \
> git submodule update --init --recursive && \
> cd tools && \
> python get.py
> Cloning into 'esp32'...
> remote: Counting objects: 8466, done.
> remote: Compressing objects: 100% (46/46), done.
> remote: Total 8466 (delta 14), reused 25 (delta 8), pack-reused 8409
> Receiving objects: 100% (8466/8466), 136.35 MiB | 7.30 MiB/s, done.
> Resolving deltas: 100% (4913/4913), done.
> git clone https://github.com/espressif/arduino-esp32.git esp32  17.05s user 7.57s system 104% cpu 23.500 total
> Submodule 'libraries/AzureIoT' (https://github.com/VSChina/ESP32_AzureIoT_Arduino) registered for path 'libraries/AzureIoT'
> Submodule 'libraries/BLE' (https://github.com/nkolban/ESP32_BLE_Arduino.git) registered for path 'libraries/BLE'
> Cloning into '/Users/c0de/Documents/Arduino/hardware/espressif/esp32/libraries/AzureIoT'...
> Cloning into '/Users/c0de/Documents/Arduino/hardware/espressif/esp32/libraries/BLE'...
> Submodule path 'libraries/AzureIoT': checked out '67dfa4f31ef88b0938dd87d955612100dea5562e'
> Submodule path 'libraries/BLE': checked out '7951347ed68313d75c367e1f2cce763cb56d1eb2'
> System: Darwin, Info: Darwin-18.0.0-x86_64-i386-64bit
> Platform: x86_64-apple-darwin
> Downloading xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz

Here is the Trackback I get from get.py:

```
Traceback (most recent call last):
  File "get.py", line 148, in <module>
    get_tool(tool)
  File "get.py", line 103, in get_tool
    urlretrieve(url, local_path, report_progress)
  File "/usr/local/Cellar/python/2.7.13_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 98, in urlretrieve
    return opener.retrieve(url, filename, reporthook, data)
  File "/usr/local/Cellar/python/2.7.13_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "/usr/local/Cellar/python/2.7.13_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "/usr/local/Cellar/python/2.7.13_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 443, in open_https
    h.endheaders(data)
  File "/usr/local/Cellar/python/2.7.13_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1038, in endheaders
    self._send_output(message_body)
  File "/usr/local/Cellar/python/2.7.13_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 882, in _send_output
    self.send(msg)
  File "/usr/local/Cellar/python/2.7.13_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 844, in send
    self.connect()
  File "/usr/local/Cellar/python/2.7.13_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1255, in connect
    HTTPConnection.connect(self)
  File "/usr/local/Cellar/python/2.7.13_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 821, in connect
    self.timeout, self.source_address)
  File "/usr/local/Cellar/python/2.7.13_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/socket.py", line 575, in create_connection
    raise err
IOError: [Errno socket error] [Errno 60] Operation timed out`
```

It seems that the website hosting **xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz** is not available right now.

lbernstone, 2018-07-11:
If your problem is resolved, please close the issue.

Master811129, 2018-07-12:
ok but there is one more question!
can i download and install the core by using the "clone or download" button?

lbernstone, 2018-07-12:
Yes.  That will give you a static version of the core.  The preferred method for that will now be to use the Arduino boards manager. https://github.com/lbernstone/arduino-esp32/blob/pr1/docs/arduino-ide/boards_manager.md

Master811129, 2018-07-13:
thank you for your help guys
why don't you add this to the installation Instructions section?
bye 😉

closed 2018-07-13 by Master811129