espressif/arduino-esp32 #1594: btStart(): BT Enable failed

chegewara, 2018-07-04:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
E (224) spiram: SPI RAM enabled but initialization failed. Bailing out.
[E][esp32-hal-bt.c:45] btStart(): BT Enable failed
```
Even if ble starts my app crash when trying to connect. It is possible its my app fault, but it has been work about 2 weeks ago.

me-no-dev, 2018-07-05:
you are not running the latest code :) please update your esp32-arduino and submodules

chegewara, 2018-07-06:
Hi @me-no-dev 
i was in rush because i got business meeting and i needed my app running.
Here is log with latest code (today git pull):
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
[E][esp32-hal-bt.c:45] btStart(): BT Enable failed
```

GithubReinart, 2018-07-06:
haven same issue with freshly installed esp-idf and Arduino-esp-idf files

chegewara, 2018-08-26:
Still no change in here? I have new arduino-ide installation and esp32 support installed with board manager and i still have the same issue:
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
FREE stack: 7424
FREE heap: 4377012
[E][esp32-hal-bt.c:45] btStart(): BT Enable failed
```
Its log from wrover module but it does not matte it wrover or wroom, still the same issue.

ankgt, 2018-10-11:
Hi @chegewara 

Could you ever fix this issue. I am trying the same thing (release bluetooth classic memory), and have the exact same error. Thanks.

chegewara, 2018-10-11:
Sorry, its not my job to do it. Its esp-idf dependent and i cant update esp32 library.

czuvich, 2018-11-15:
I'm getting the following log with the latest ESP32 Arduino pull when trying to call esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT) before init().

08:17:46.704 -> [E][BLEScan.cpp:206] start(): esp_ble_gap_set_scan_params: err: 259, text: Invalid state

Any updates would be great!

chegewara, 2018-11-26:
Its not a bug, we just cant release memory anymore in arduino because both, ble and esp classic are turned on in sdkconfig.