espressif/arduino-esp32 #1595: [E][WiFiGeneric.cpp:641] hostByName(): DNS Failed for CE

wegunterjr, 2018-07-04:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			version 3.5.4/ says most up to date (Seems wrong?)
IDE name:							Platform.io
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Not sure what is happening, but this is the error I am getting:

`
[E][WiFiGeneric.cpp:641]` hostByName(): DNS Failed for CE
`

I think it has to do with calling WiFi.begin(sid, pwd), but wanted to verify..

jontay81, 2018-07-30:
**Hardware:**
Boards:	Adafruit Feather ESP32, HiLetgo ESP32 Dev Board
Core Installation/update date: July 7. 2018 install date
IDE name: VSCode / Arduino Extension
Flash Frequency:	80Mhz
Upload Speed:	115200

**Description:**
Having this same error when scanning for BLE signals then issuing a POST request over WiFi to a LAN server with DNS setup through my router.

It occurs intermittently - somewhere between 1 and 4 hours after reset. After one occurrence, the error occurs every time the device attempts a POST request. 

Haven't tested on an internet DNS lookup or using a LAN IP address yet.

`[V][HTTPClient.cpp:140] beginInternal(): url: http://dev-wifi:3100/api/beacon
`
` 
[D][HTTPClient.cpp:181] beginInternal(): host: dev-wifi port: 3100 url: /api/beacon
`
`
[E][WiFiGeneric.cpp:659] hostByName(): DNS Failed for dev-wifi 
`
`
[D][HTTPClient.cpp:846] connect(): failed connect to dev-wifi:3100 
`
`
[W][HTTPClient.cpp:1124] returnError(): error(-1): connection refused 
`

I suppose there's a way to catch the error and reset the wifi connection that might work. I'll update this if I find a workaround.

copercini, 2018-10-12:
I'm also getting it during or after BLE scans

copercini, 2018-10-12:
This seems to happen here when a task has a priority too high (over 4) or may the wifi code isn't pinned to arduino core

closed 2018-12-15 by me-no-dev

roger-zacharczyk, 2019-02-13:
Having the same issue despite the update. The message "STA IP..., GW... " is logged, though, so it gets all the addresses needed.

roger-zacharczyk, 2019-02-13:
Nevermind that. I passed "http://hostname" instead of just "hostname".