espressif/arduino-esp32 #1596: unable to remove obsolete platformio version

wegunterjr, 2018-07-04:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			???
IDE name:							Platform.io ATOM
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
I am getting the following warning:
```
Obsolete PIO Core v3.5.4 is used (previous was 3.6.0a5)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
```

I am not sure what I did. I was using the 3.6.0a4, and tried to upgrade to the 3.6.0a5 - I may have skipped a step.
I am using Windows 10 and have tried to use the CMD prompt to remove it, but that isn't working for me:
http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system
`pip uninstall platformio`

copercini, 2018-07-04:
As it is a specific PIO question, I think the best place to get support about it is https://github.com/platformio/platform-espressif32

wegunterjr, 2018-07-04:
ah dang...yeah, wrong place.  closing now.

closed 2018-07-04 by wegunterjr