espressif/arduino-esp32 #1598: no matching function for call (BLE server & rc522)

azzaniadtya2, 2018-07-05:
### Hardware:
Board:							Wemos Lolin32 Lite
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
i have a problem when i want to send the UID of card from RC522 to client from server, please help me

### Sketch:
//server buat bis

#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

String simpan[25];
int counter = 25;
#include<SPI.h>
#include<MFRC522.h>
#define rst 0
#define ss 5

#define idsize 7
#define timeoutK 1000
#define timerP 15

bool readuid() ;
String printHex(byte *buffer, byte bufferSize);

MFRC522 reader(ss, rst);

String idold, idnew;
unsigned long last_record;

BLECharacteristic *pCharacteristic;
std::string rxValue;
String rxValueArduino ="";
bool deviceConnected = false;
#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"
class MyServerCallbacks: 
public BLEServerCallbacks {
  void onConnect(BLEServer* pServer) {
    deviceConnected = true;
  };

  void onDisconnect(BLEServer* pServer) {
    deviceConnected = false;
  }
};
class MyCallbacks: 
public BLECharacteristicCallbacks {
  void onWrite(BLECharacteristic *pCharacteristic) {
    rxValue = pCharacteristic->getValue();
    rxValueArduino ="";
    for (int i = 0; i < rxValue.length(); i++) {
      rxValueArduino += rxValue[i];
    }
  }
};
String txStringArduino ="";
String getValue(String data, char separator, int index)
{
  int found = 0;
  int strIndex[] = { 
    0, -1   };
  int maxIndex = data.length() - 1;
  for (int i = 0; i <= maxIndex && found <= index; i++) {
    if (data.charAt(i) == separator || i == maxIndex) {
      found++;
      strIndex[0] = strIndex[1] + 1;
      strIndex[1] = (i == maxIndex) ? i+1 : i;
    }
  }
  return found > index ? data.substring(strIndex[0], strIndex[1]) : "";
}

void setup()
{

  
  int x;
  for (x=0;x<25;x++)
  {
    simpan[x] = "";
  }
  SPI.begin();
  reader.PCD_Init();
  idnew = ""; idold = "";
  
  BLEDevice::init("ESP32");
  // Membuat BLE Server
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Membuat BLE Servis
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Membuat Karakteristik BLE
  pCharacteristic = pService->createCharacteristic(
  CHARACTERISTIC_UUID_TX,
  BLECharacteristic::PROPERTY_NOTIFY
    );

  pCharacteristic->addDescriptor(new BLE2902());

  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
  CHARACTERISTIC_UUID_RX,
  BLECharacteristic::PROPERTY_WRITE
    );
  pCharacteristic->setCallbacks(new MyCallbacks());
  // Memulai Servis
  pService->start();

  // Memulai advertising
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->addServiceUUID(pService->getUUID());
  pAdvertising->start();
  Serial.println("BUS server Menunggu koneksi Halte client untuk notify...");
  Serial.begin(115200);
  //Serial2.begin(115200);

}

void loop()
{
  if (readuid()) {
    last_record = millis();
  }
  //txStringArduino += String(  random( 0 , 10 )) +'|';
  //txStringArduino += String(  random( 20 , 30 )) +'|';
  //txStringArduino += String(  random( 40 , 50 )) +'|';
  //txStringArduino += String ( char (A)) +'|';
  
  

  
  if (deviceConnected) {
    uint8_t txLen = txStringArduino.length();
    char txString[ txLen ];
    for(uint8_t i=0; i<txLen; i++){
      txString[i] = txStringArduino[i];
    }
    pCharacteristic->setValue(readuid());
    pCharacteristic->notify(); // Send the value to the app!
  }
  txStringArduino ="";
  //Serial.print(getValue(rxValueArduino,'|', 0).toInt());
  //Serial.print(" ");
  //Serial.print(getValue(rxValueArduino,'|', 1).toInt());
  //Serial.print(" ");
  //Serial.print(getValue(rxValueArduino,'|', 2).toInt());
  //Serial.print(" ");
  //Serial.print(getValue(rxValueArduino,'|', 3).toInt());
  //Serial.print(" ");
  Serial.println();
}

bool readuid() {
  if (!reader.PICC_IsNewCardPresent()) return false;
  if (!reader.PICC_ReadCardSerial())return false;
  idnew = printHex(reader.uid.uidByte, reader.uid.size);
  if(idold!=idnew||last_record+timeoutK<millis()){
    last_record = millis();
//    Keyboard.write(idnew);
    idold = idnew;
  }
  reader.PICC_HaltA();
  reader.PCD_StopCrypto1();
  return true;
}

String printHex(byte *buffer, byte bufferSize) {
  int i;
  int a=0;
  int tunjuk;
  String temp = "";
  for (byte i = bufferSize; i < idsize; i++ )
    temp += "00";
  for (byte i = 0; i < bufferSize; i++) {
    if (buffer[bufferSize - i - 1] < 0x10)temp += "0";
    temp += String(buffer[bufferSize - i - 1], HEX);
  }
  temp.toUpperCase();
  for (i=0;i<25;i++)
  {
    if (temp == simpan[i])
    {
      a = 1;
      tunjuk = i;
      break;
    }
  }
  if (a == 1)
  {
    simpan[tunjuk] = "";
    Serial.print("RFID ke ");
    Serial.print(tunjuk);
    Serial.print(" telah dikosongkan");
    Serial.println();
    counter = counter +1;
  }
  else if (a == 0)
  {
    for (i=0;i<25;i++)
    {
      if (simpan[i]=="")
      {
        simpan[i] = temp;
         Serial.print("RFID ke ");
        Serial.print(i);
        Serial.print(" adalah ");
        Serial.print(simpan[i]);
        Serial.println();
        counter = counter - 1;
        break;
      }
    }
  }
  Serial.print("Sisa tempat duduk adalah ");
  Serial.print(counter);
  Serial.println();
  return temp;
}

### Debug Messages:
C:\Users\AdityaCHIKO\Documents\Arduino\coba1\coba1.ino: In function 'void loop()':

coba1:140: error: no matching function for call to 'BLECharacteristic::setValue(bool)'

     pCharacteristic->setValue(readuid());

                                        ^

In file included from C:\Users\AdityaCHIKO\Documents\Arduino\libraries\ESP32_BLE_Arduino-0.4.7\src/BLEServer.h:19:0,

                 from C:\Users\AdityaCHIKO\Documents\Arduino\libraries\ESP32_BLE_Arduino-0.4.7\src/BLEDevice.h:18,

                 from C:\Users\AdityaCHIKO\Documents\Arduino\coba1\coba1.ino:3:

C:\Users\AdityaCHIKO\Documents\Arduino\libraries\ESP32_BLE_Arduino-0.4.7\src/BLECharacteristic.h:75:7: note: candidate: void BLECharacteristic::setValue(uint8_t*, size_t)

  void setValue(uint8_t* data, size_t size);

       ^

C:\Users\AdityaCHIKO\Documents\Arduino\libraries\ESP32_BLE_Arduino-0.4.7\src/BLECharacteristic.h:75:7: note:   candidate expects 2 arguments, 1 provided

C:\Users\AdityaCHIKO\Documents\Arduino\libraries\ESP32_BLE_Arduino-0.4.7\src/BLECharacteristic.h:76:7: note: candidate: void BLECharacteristic::setValue(std::__cxx11::string)

  void setValue(std::string value);

       ^

C:\Users\AdityaCHIKO\Documents\Arduino\libraries\ESP32_BLE_Arduino-0.4.7\src/BLECharacteristic.h:76:7: note:   no known conversion for argument 1 from 'bool' to 'std::__cxx11::string {aka std::__cxx11::basic_string<char>}'

exit status 1
no matching function for call to 'BLECharacteristic::setValue(bool)'

chegewara, 2018-07-16:
You have error in your code here:
`pCharacteristic->setValue(readuid());`
readuid function returns bool type value, and there is no setValue() function that can be used to write bool value into characteristic.

ardhinugroho, 2018-12-07:
Maaf menggunakan Bahasa Indonesia
Boleh minta konfigurasi pin dari rc522 ke lolin32 lite
Karena saya googling minim referensi
Kebetulan mau saya pasang juga RTC DS3231 
Mohon saran
Terimakasih

closed 2019-08-15 by stale[bot]