espressif/arduino-esp32 #1599: base64 Encoding, undefined reference to base64_encode(...)

Kazuya91, 2018-07-05:
I want to encode my ciphertext and send it via JSON to a server. Unfortunately the Arduino-IDE tells me, that  there is a undefined reference. I dont know what I am doing wrong.

I want to use the functions of this file:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/wpa_supplicant/crypto/base64.h

My Code:

```
#include "crypto/base64.h"

String msg = "hello"
size_t plaintext_length = msg.length();
size_t ciphertext_length = plaintext_length;
size_t out_len = 0;
unsigned char ciphertext[plaintext_length];
//code to fill "ciphertext"-array

unsigned char *base64_ciphertext = base64_encode(ciphertext, ciphertext_length, &out_len);
```

I get this error message:
```
sketch/test.ino.cpp.o:(.literal._Z16encrypt_message16String+0x38): undefined reference to `base64_encode(unsigned char const*, unsigned int, unsigned int*)'
sketch/test.ino.cpp.o: In function `encrypt_message1(String)':
/Users/Kazuya91/Documents/Arduino/acaMail/acaMail.ino:444: undefined reference to `base64_encode(unsigned char const*, unsigned int, unsigned int*)'
```

Why does it say its an unsigned int? I am passing the arguments which the function wants from me.

Can someone tell me whats wrong? Or alternatively provide an example how to use the function?

Thanks in Advance!

lbernstone, 2018-07-05:
Looks like that function isn't included in the compiled libraries, even though the header is.  The mbedtls function works:
```
#include "mbedtls/base64.h"
void setup() {
  Serial.begin(115200);
  String msg = "hello";
  size_t out_len = 0;
  mbedtls_base64_encode(NULL,0,&out_len,(unsigned char*)msg.c_str(), msg.length());
  unsigned char b64_cipher[out_len]; 
  mbedtls_base64_encode(b64_cipher,out_len,&out_len,(unsigned char*)msg.c_str(), msg.length());
  Serial.println((char*)b64_cipher);
}
void loop() {}
```

Kazuya91, 2018-07-05:
@lbernstone 
Thank you very much! I was trying for hours what I am doing wrong. Works now.

lbernstone, 2018-07-05:
Please close the issue.

milanbhatia, 2020-01-01:
> @lbernstone
> Thank you very much! I was trying for hours what I am doing wrong. It works now.

@Kazuya91 - Can you please share what is it that you were doing wrong and how you fixed it? I ask because I am getting the same undefined reference to `base64_encode(unsigned char const*, unsigned int, unsigned int*)' error when using base64_encode to convert my image from the esp32 camera to base64.

nh719, 2020-01-29:
@lbernstone I am also getting the same error as  (undefined reference to `base64_encode(unsigned char const*, unsigned int)') I am trying to fix this but not working out for me. I want to convert a jpeg to base64 string. How can I do this ? what files should i add to my project?

I refered this code -
                string encoded_jpg;
		Mat Stream_image;              // Load an image here
		vector<uchar> buf;
		cv::imencode(".jpg", Stream_image, buf);
		auto base64_jpg = reinterpret_cast<const unsigned char*>(buf.data());
		encoded_jpg = "data:Stream_image/jpeg;base64," + base64_encode(base64_jpg, buf.size());			
		cout << "Encoded Image string" << encoded_jpg << endl;