espressif/arduino-esp32 #1601: HardwareSerial transmit not working as expected

arcturial, 2018-07-05:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		4/July/2018
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
I have a raspberry PI connected to my dev board on port 16/17 and I am using those ports for serial communication. Everything worked fine for a while and then suddenly after a reboot the HardwareSerial connection stopped transmitting messages to the PI RX port and instead the messages being printed appear as part of the `read()` call. Sending data from the PI seems fine, I just can't get data from the dev board to transmit.

If I use the normal RX and TX port on the dev board then everything works fine, but since that is used for flashing and debug it doesn't suit my needs.

Can this be a hardware fault? Any suggestions welcome.

### Sketch:
```cpp
HardwareSerial OmronSerial(2);

void setup() {
    OmronSerial.begin(9600, SERIAL_8N1, 16, 17);
    Serial.begin(115200);
    delay(2000);
}

void loop() {
    Serial.println("Tick");
    OmronSerial.println("Second serial message");

    
    Serial.println("READING!");
    while (OmronSerial.available())
    {
      char c = OmronSerial.read();
      Serial.print(c);
    }
    Serial.println("");

    vTaskDelay(5000 / portTICK_RATE_MS);
}
```

### Serial Monitor Output:
```
READING!
Second serial message

Tick
READING!
Second serial message

Tick
READING!
Second serI`l m@⸮sage

Tick
READING!
Second serI⸮l i⸮rsaga	

Tick
READING!
Sebond serI⸮l I⸮rsafa
```

On the PI side it doesn't receive or send any messages. I also am not sure why the text is sometimes read correctly and sometimes read with garbage characters.

stickbreaker, 2018-07-05:
@arcturial They made a change to the latest release that initiates `Serial()`, `Serial1()`, and `Serial2()`.

instead of defining your own with:
```c++
 HardwareSerial OmronSerial(2);

void setup() {
    OmronSerial.begin(9600, SERIAL_8N1, 16, 17);
    Serial.begin(115200);
    delay(2000);
}
```
Just use the preexisting `Serial2()`:
```c++
// HardwareSerial OmronSerial(2); // no longer necessary

void setup() {
    Serial2.begin(9600, SERIAL_8N1, 16, 17);
    Serial.begin(115200);
    delay(2000);
}
```

Chuck.

arcturial, 2018-07-05:
Thank you, this seems to have solved my problem.

closed 2018-07-05 by arcturial

kbickham, 2019-08-01:
Not for me...

atanisoft, 2019-08-01:
@kbickham please open a new issue with code example etc reproducing your issue.