espressif/arduino-esp32 #1602: Latest Firmware

usmanshahid001, 2018-07-05:
hi i have updated the arduino core for esp32 after that i am having some issues related to WIFI_AP_STA mode if i use this mode and use wifi.reset() ap will never connect with my phone and AP only working fine in WIFI_AP mode in any other case its being connecting and disconnecting...

lbernstone, 2018-07-05:
Set core debug level to verbose and post some logs while you try to connect the phone.

usmanshahid001, 2018-07-05:
I (2266) wifi: wifi driver task: 3ffb7328, prio:23, stack:4096, core=0
I (4592) wifi: wifi firmware version: 4fd6a2c
I (4592) wifi: config NVS flash: enabled
I (4595) wifi: config nano formating: disabled
I (4609) wifi: Init dynamic tx buffer num: 32
I (4609) wifi: Init data frame dynamic rx buffer num: 10
I (4609) wifi: Init management frame dynamic rx buffer num: 10
I (4614) wifi: Init static rx buffer size: 1600
I (4618) wifi: Init static rx buffer num: 4
I (4622) wifi: Init dynamic rx buffer num: 10
I (4683) wifi: mode : sta (b4:e6:2d:86:c0:01) + softAP (b4:e6:2d:86:c0:02)

usmanshahid001, 2018-07-05:
I (6556) wifi: ap channel adjust o:7,1 n:9,2
I (6556) wifi: n:9 0, o:7 0, ap:9 2, sta:9 0, prof:7
I (7289) wifi: state: init -> auth (b0)
I (7292) wifi: state: auth -> assoc (0)
I (7298) wifi: state: assoc -> run (10)
I (7315) wifi: connected with GRESHAM EASTERN, channel 9
I (7361) wifi: pm start, type: 1

usmanshahid001, 2018-07-05:
when i try to connect my AP i receive this verbose

I (59995) wifi: n:9 0, o:9 0, ap:9 2, sta:9 0, prof:9
I (59995) wifi: station: 24:77:03:6d:79:ec join, AID=1, bgn, 20
but it refuse to connect

lbernstone, 2018-07-05:
I'm unable to reproduce.  Can you connect to the AP with another device?

usmanshahid001, 2018-07-05:
are u using the latest commit?

usmanshahid001, 2018-07-05:
so i used the basic authentication code here with ap_sta

#include <WiFi.h>
#include <WiFiClient.h>
#include "includes.h"
#include "WiFiAP.h"
#include "esp_log.h"
HardwareSerial hmi(1);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress DNS(192, 168, 1, 249);
WiFiClient client[my_client];
 
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;
///////////////////////////////////////////////////////////////////////// 
void IRAM_ATTR onTimer(){
        
        portENTER_CRITICAL_ISR(&timerMux);
        if(info.start_totalizer==1)
        {
        info.temp+=info.curr_flow;
        if((uint16_t)info.temp>=1)
        {
          Total_s.Total+=(uint16_t)info.temp;
          
          if(Total_s.Total>4294967290)
          Total_s.Total=0;
          info.temp-=(uint16_t)info.temp;
        }
        }
        if(mywdt>6)
        esp_restart_noos();
        mywdt++;
        digitalWrite(led_pin,link1);
        if(link1==0)
        link1=1;
        else
        link1=0;
        portEXIT_CRITICAL_ISR(&timerMux);
        
        
}

void setup() {
      Wire.begin(SCL,SDL,FREQUENCY );
 
      delay(1000);
      Serial.begin(115200);
      esp_log_level_set("*", ESP_LOG_VERBOSE);
      My_functions.feedback();
      app_init();
      My_functions.feedback();
      delay(10);
      My_functions.feedback();
      delay(1000);
      Time_Date.read_time();
      My_functions.feedback();delay(10);
      My_functions.feedback();
      initialization();
      analogReadResolution(12);
      analogSetPinAttenuation(A0, ADC_0db);
      analogSetPinAttenuation(A1, ADC_0db);
      xMutex = xSemaphoreCreateMutex();
      xMutex_client=xSemaphoreCreateMutex();
      
      
      xTaskCreate(client_tasks,"client_tasks",10000,NULL,(configMAX_PRIORITIES - 4),&client_Tasks);
      xTaskCreate(flow_cal_task,"flow_cal_task",4096,NULL,(configMAX_PRIORITIES - 2),&flow);
      xTaskCreate(time_take,"time_take",4096,NULL,(configMAX_PRIORITIES - 3),&time_taker);
[... 302 lines cut ...]
        WiFi.config(ip, gateway, subnet, DNS);
        got_static_ip=1;
        }
        Serial.println(WiFi.localIP());  */
        My_functions.feedback();delay(10);My_functions.feedback();
       initialize_server();
         
        delay(1000);
        startTimer();
        
}
void hmi_print()
{
            hmi.write(255);
            hmi.write(255);
            hmi.write(255 );
}

void startTimer()
{
      timer = timerBegin(0, 80, true); 
      timerAttachInterrupt(timer, &onTimer, true); 
      timerAlarmWrite(timer, 1000000, true); 
      timerAlarmEnable(timer);
}

usmanshahid001, 2018-07-05:
whenever i try to connect i receive these verbose but i am unable to connect

I (55155) wifi: n:9 0, o:9 0, ap:9 2, sta:9 0, prof:9
I (55155) wifi: station: 24:77:03:6d:79:ec join, AID=1, bg, 20

usmanshahid001, 2018-07-05:
problem occurs when wifi.begin() is call

usmanshahid001, 2018-07-05:
system says password is incorrect but it works when i do not call wifi.begin

stickbreaker, 2018-07-05:
@usmanshahid001 The ESP32 stores the successful connection info in NVS so without you specifically calling `Wifi.Begin()` it reuses the last successful credentials. Automagically logging in.

Chuck.

lbernstone, 2018-07-05:
Note that the radio can only use one channel.  If you force the AP to use channel 6, and your STA needs a different channel, it may disconnect a client when it switches to the "correct" channel.

usmanshahid001, 2018-07-05:
actually if i do not mention the channel AP never connects...actually i had working codes at previous commits may be 4 month old...today i updated my esp32 firmware now they are causing problem......

usmanshahid001, 2018-07-12:
does anyone facing this problem

zzpaf, 2018-11-18:
I am having the same exact problem. Removing WiFi.begin(ssid, password); AP mode works correctly, once WiFi.begin(ssid, password); is called, AP mode does not allow connection

zzpaf, 2018-11-21:
> Note that the radio can only use one channel. If you force the AP to use channel 6, and your STA needs a different channel, it may disconnect a client when it switches to the "correct" channel.

This is a different behavior as in ESP8266, actually if client needs a different channel, AP will change channel, disconnecting current connections but will allow you to connect again, what happens now is that once you call Wifi.Begin() no matter what channel the client is using, the AP refuses any connections.

I would expect the same behavior as in ESP8266 where this work perfectly but as of now I can't get ESP32 connected both as AP and STA

me21, 2019-01-15:
It's supposed to be fixed now in the latest release.

pablodeymo, 2019-04-02:
I have the same problem that @usmanshahid001 described, using the ESP32 Dev Module.
I've downloaded the latest version of this repo, and I am using the files of _libraries/WiFi/src_ folder, and Arduino IDE version 1.8.9.

When ESP runs Wifi.begin(...), the AP **disappear** from the list of SSIDs.

Do you know if this problem is already fixed?

asetyde, 2019-04-02:
WiFi has many problems also for me, I hope depends from update of ESP-IDF , hope is updated at last

usmanshahid001, 2019-04-02:
i faced that problem when wifi.begin() failed to connect SSID due to wrong
password etc.if it connects successfully then i think AP connects..

closed 2019-08-15 by stale[bot]