espressif/arduino-esp32 #1604: Unable to use pin 17 for UART 0, 1, 2 TX

thethereza, 2018-07-06:
### Hardware:
Board:							Custom
CPU:                                                         **ESP32-D2WD**
Core Installation/update date:		?Jan 12, 2018? 
IDE name:						?Arduino IDE? 
Flash Frequency:					?80Mhz?
Upload Speed:					?921600?

### Description:
Cannot get any uart to output to GPIO17 (even though this is the default for UART2).  I've also tried with uart0 and uart1.  When I set it to uart0, I can get the console debug output sent to the uart but it's just a guru meditation error with the registers.  when i use uart 0 or uart 1 I get gibberish out.  When looking at the output when set to uart2, this is what it looks like:

[image]

Our device needs to output serial on pin 17, and just about every other pin but 17 seems to work.  Any thoughts on how to proceed?

### Sketch:
```cpp
HardwareSerial Serial2(2);

void setup() {
  //Serial2.begin(115200);
  Serial2.begin(115200, SERIAL_8N1, 16, 17);
  Serial2.print("ESP32_BOOTED\r\n"); 
}

void loop() {  
 Serial2.printf("Hello, world\r\n");
}
```

### Debug Messages:
```

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400d0b38  PS      : 0x00060530  A0      : 0x800e2f62  A1      : 0x3ffca620  
A2      : 0x3ffc2204  A3      : 0x00000000  A4      : 0x00060023  A5      : 0x3ffc7d7c  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d0894  A9      : 0x3ffca5f0  
A10     : 0x3ffc2204  A11     : 0x3f400ef0  A12     : 0x0800001c  A13     : 0x00000010  
A14     : 0x00000011  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000080  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d0b38:0x3ffca620 0x400e2f5f:0x3ffca640

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400d0b38  PS      : 0x00060530  A0      : 0x800e2f62  A1      : 0x3ffca620  
A2      : 0x3ffc2204  A3      : 0x00000000  A4      : 0x00060023  A5      : 0x3ffc7d7c  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d0894  A9      : 0x3ffca5f0  
A10     : 0x3ffc2204  A11     : 0x3f400ef0  A12     : 0x0800001c  A13     : 0x00000010  
A14     : 0x00000011  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000080  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d0b38:0x3ffca620 0x400e2f5f:0x3ffca640

Rebooting...
ets Jun  8 2016 00:22:57

```

thethereza, 2018-07-06:
> @thethereza The new core per-allocates Serial(), Serial1(), and Serial2().
> So, you don't need to do it.
> Chuck.

Has anyone been able to use any uart on pin 17 and not have it fault?  

Reza

thethereza, 2018-07-06:
So I can't even control GPIO17 with digitalwrite.  I think there is an issue with this pin on the ESP32-D2WD.

```
void setup() {
  pinMode(17, OUTPUT);
}

void loop() {  
 digitalWrite(17, HIGH);
 delay(100);
 digitalWrite(17, LOW);
 delay(100);
}
```

This is what the output looks like running that code

[image]

thethereza, 2018-07-06:
Ugh.  The latest version of their datasheet added this text

> 
> • ESP32-D2WD’s pins GPIO16, GPIO17, SD_CMD, SD_CLK, SD_DATA_0 and SD_DATA_1 are used for connecting
> the embedded flash, and are not recommended for other uses.

The version that was out when we designed the board lacked that documentation.  Very frustrating.

Redferne, 2018-07-19:
Arrrgh. Also have a design using GPIO17 as DigitalIO. Any way to free it in the framework? Compile flag? Enable/Disable PSRAM?

1163167506, 2018-07-20:
#include <HardwareSerial.h>

HardwareSerial MySerial(1);

void setup() 
{
    MySerial.begin(115200, SERIAL_8N1, 16, 17);
        //Write something like that
    MySerial.println("wait your date...");
}

void loop() 
{

    #if 1
    while (MySerial.available() > 0) 
    {
        // Do something
        uint8_t byteFromSerial = MySerial.read();
        MySerial.print("I GET IT:");
        MySerial.println(byteFromSerial, DEC);
    }
    #endif
}

--------------------------------------------
this is HardwareSerial MySerial(1);
-------------------------------------------

axwell, 2018-07-30:
I have the same issue, this worked before i updated the arduino-esp32.
How can i fix it ?

lbernstone, 2018-07-30:
Are you certain you have the same issue?  Does `python esptool.py chip_id` show that you have a D2WD? If not, make sure you are using the latest version of code (release 1.0.0) and disable PSRAM in the tools menu.

kbickham, 2019-08-01:
I specifically have this issue. You are saying you CAN get any serial output off the board to anything but the usb virtual connection to pc? I'll go switch around some wire...

closed 2019-10-14 by stale[bot]