espressif/arduino-esp32 #1606: Enabling BLE is causing WiFi connection to fail - any suggestions?

wegunterjr, 2018-07-06:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			July 2018
IDE name:							Platform.io
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
I am enabling BLE Server with notify to be available for client to connect and that appears to work, but when I try to enable WiFi later and connect to a SSID, then connect to an MQTT broker securely, I get back that it is failing - 
```

BLEDevice::init("iotDevice");
BLEServer *pServer = BLEDevice::createServer();

BLEService *pService = pServer->createService(SERVICE_UUID);

pCharacteristic = pService->createCharacteristic(
                    CHARACTERISTIC_UUID,
                    BLECharacteristic::PROPERTY_READ   |
                    BLECharacteristic::PROPERTY_WRITE  |
                    BLECharacteristic::PROPERTY_NOTIFY |
                    BLECharacteristic::PROPERTY_INDICATE
                                     );
BLEDescriptor *desc1= new BLEDescriptor(BLEUUID((uint16_t)0x2908));
const uint8_t desc1_val[] = {0x01, 0};
desc1->setValue((uint8_t*)desc1_val, 1);
pCharacteristic->addDescriptor(desc1);
pCharacteristic->addDescriptor(new BLE2902());

pCharacteristic->setCallbacks(new MyCallbacks());

char hex[] = {0xee,0xef};
pCharacteristic->setValue(hex);
pService->start();

BLEAdvertising *pAdvertising = pServer->getAdvertising();
pAdvertising->start();

while (status != WL_CONNECTED)
    {
        Serial.print("Attempting to connect to SSID: ");
        Serial.println(WIFI_SSID);
        // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
        status = WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

        // wait 5 seconds for connection:
        delay(5000);
    }

    Serial.println("Connected to wifi");
if(grillDevice.connect(HOST_ADDRESS,CLIENT_ID)== 0) // Connect to AWS using Host Address and Cliend ID
    {
        Serial.println("Connected to AWS");
        delay(1000);
    }
    else
    {
        Serial.println("AWS connection failed, Check the HOST Address");
        while(1);
    }

```

If I don't try and enable BLE before I do the WiFi connect,

chegewara, 2018-07-16:
Hi,
its very hard to make it both work together. Its because ble is using a lot of ram and mbedTLS (ssl) needs a lot of ram for handshake. You can make it work togheter, but you need to free as much heap as you can before you even start, then connect with aws mqtt and when you are connected then you can finally connect to ble peer device.
In my app i am starting with over 200kB free heap and when i am connected to ble device i have about 20kB free heap.

kapollo, 2018-07-24:
@chegewara how can you do that on Arduino, I mean free the heap before starting?

chegewara, 2018-07-24:
https://github.com/chegewara/esp32-snippets/issues/2#issuecomment-405275150

closed 2019-08-15 by stale[bot]