espressif/arduino-esp32 #1608: EEPROM.readString(address) reads string with junk characters at the end

avinashs24, 2018-07-06:
### Hardware:
Board:	ESP-WROOM-32
Core Installation/update date:	01/JUN/2018
IDE name:	Arduino IDE
Flash Frequency:	80Mhz
Upload Speed:	921600

### Description:
I am trying to write some string to EEPROM and retrieve it later. Whenever I try to read the string using EEPROM.readString(address), one or two junk characters like (`, ?, ") are at the end of the string. Even in the example eeprom_extra, the last two operations write and read string. The output for the string operations are,

```
.
.
I love ESP32."
Thank You Espressif!`

```

### Sketch:
```cpp

/*
  ESP32 eeprom_extra example with EEPROM library

  This simple example demonstrates using other EEPROM library resources

  Created for arduino-esp32 on 25 Dec, 2017
  by Elochukwu Ifediora (fedy0)
*/

#include "EEPROM.h"

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\nTesting EEPROM Library\n");
  if (!EEPROM.begin(64)) {
    Serial.println("Failed to initialise EEPROM");
    Serial.println("Restarting...");
    delay(1000);
    ESP.restart();
  }

  int address = 0;                                  // Same address is used through the example

  EEPROM.writeByte(address, -128);                  // -2^7
  Serial.println(EEPROM.readByte(address));

  EEPROM.writeChar(address, 'A');                   // Same as writyByte and readByte
  Serial.println(char(EEPROM.readChar(address)));

  EEPROM.writeUChar(address, 255);                  // 2^8 - 1
  Serial.println(EEPROM.readUChar(address));

  EEPROM.writeShort(address, -32768);               // -2^15
  Serial.println(EEPROM.readShort(address));

  EEPROM.writeUShort(address, 65535);               // 2^16 - 1
  Serial.println(EEPROM.readUShort(address));

  EEPROM.writeInt(address, -2147483648);            // -2^31
  Serial.println(EEPROM.readInt(address));

  EEPROM.writeUInt(address, 4294967295);            // 2^32 - 1
  Serial.println(EEPROM.readUInt(address));

  EEPROM.writeLong(address, -2147483648);           // Same as writeInt and readInt
  Serial.println(EEPROM.readLong(address));

  EEPROM.writeULong(address, 4294967295);           // Same as writeUInt and readUInt
  Serial.println(EEPROM.readULong(address));

  int64_t value = -9223372036854775808;             // -2^63
  EEPROM.writeLong64(address, value);
  value = 0;                                        // Clear value
  value = EEPROM.readLong64(value);
  Serial.printf("0x%08X", (uint32_t)(value >> 32)); // Print High 4 bytes in HEX
  Serial.printf("%08X\n", (uint32_t)value);         // Print Low 4 bytes in HEX

  uint64_t  Value = 18446744073709551615;           // 2^64 - 1
  EEPROM.writeULong64(address, Value);
  Value = 0;                                        // Clear Value
  Value = EEPROM.readULong64(Value);
  Serial.printf("0x%08X", (uint32_t)(Value >> 32)); // Print High 4 bytes in HEX
  Serial.printf("%08X\n", (uint32_t)Value);         // Print Low 4 bytes in HEX

  EEPROM.writeFloat(address, 1234.1234);
  Serial.println(EEPROM.readFloat(address), 4);

  EEPROM.writeDouble(address, 123456789.123456789);
  Serial.println(EEPROM.readDouble(address), 8);

  EEPROM.writeBool(address, true);
  Serial.println(EEPROM.readBool(address));

  String sentence = "I love ESP32.";
  EEPROM.writeString(address, sentence);
  Serial.println(EEPROM.readString(address));

  char gratitude[] = "Thank You Espressif!";
  EEPROM.writeString(address, gratitude);
  Serial.println(EEPROM.readString(address));

  // See also the general purpose writeBytes() and readBytes() for BLOB in EEPROM library
  // To avoid data overwrite, next address should be chosen/offset by using "address =+ sizeof(previousData)"
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages:
```
Testing EEPROM Library

128
A
255
-32768
65535
-2147483648
4294967295
-2147483648
4294967295
0x8000000000000000
0xFFFFFFFFFFFFFFFF
1234.1234
123456789.12345679
1
I love ESP32."
Thank You Espressif!`

```

lbernstone, 2018-07-06:
Yup.  I'll put in a patch.

closed 2018-07-07 by me-no-dev

MustafaMontaser, 2020-05-29:
For example if I wrote "I love ESP32" in address 0  or in any address, How can I know the number of addresses which be taken to store the previous string  ?

lbernstone, 2020-05-29:
This issue is closed.  This site is not for tutorials.
EEPROM requires you to keep track of the address.  Use Preferences instead.

PR #1609 (2018-07-07): EEPROM.readstring was returning an extra character
Fixes #1608