espressif/arduino-esp32 #1610: Identify Version

stickbreaker, 2018-07-06:
Can we set the name and version info in the `platform.txt` file to some identifiable values?

Currently `platfrom.txt` just defaults to:
>name=ESP32 Arduino
version=0.0.1

I would recommend it be changed for each version: 
>name=ESP32 Arduino RC1 
version=1.0.1rc

>name=ESP32 Arduino RC2
version=1.0.2rc

Chuck.

lbernstone, 2018-07-06:
package.json has a spot for the version number as well.

stickbreaker, 2018-07-06:
maybe even a new define:
>name=ESP32 Arduino RC2
version=1.0.0rc2
corename=ESP32_Arduino_RC2

I modified the `platform.txt`, changing the `build.extra_flags` to:
>build.extra_flags=-DESP32 -DCORE_DEBUG_LEVEL={build.code_debug} -DESP32_CORE_NAME="{corename}" -DESP32_CORE_VERSION="{version}"

`{name}` by itself was replaced with the `{board}.name` value, and embedded spaces also caused problems.
 
If we do this then we can easily identify which version is causing problems with:
```c++
Serial.printf("%s %s\n",ESP32_CORE_NAME,ESP32_CORE_VERSION);
```
Chuck.

stickbreaker, 2018-07-07:
@me-no-dev I don't know how to inject it into `platform.txt`.  The reason I want it is that This name is displayed as the group header inside the Arduino Boards Selection pulldown.  I currently have 3 ESP32 sets of boards, My modified library, the Current Repo and RC2.  When they all said ESP32 is was a pain to use the correct "board/repo'.

How automated is your package process? Can we 'just' add a python script to your packaging process?

Chuck.

me-no-dev, 2018-07-07:
you do not need to inject anything. there is a header in the package that you need to include. it's generated by the build script. Question is how to generate this when you are using git version and not a package

me-no-dev, 2018-07-08:
can you elaborate a bit on this?

lbernstone, 2018-07-10:
google knows far more than me. You could use python to modify a file when you pull/fetch from upstream, or even just sed.
https://www.startpage.com/do/dsearch?query=git+hook+modify+file+on+pull

me-no-dev, 2018-07-10:
that is something else though. version in platform.txt will be updated now BTW (merged this morning).

stickbreaker, 2018-07-11:
That is a good step.

 Can you append the version code to `name` field? that way under board manager it would be easy to see. 

I just noticed that the text in `platform.txt` is only uses `lf` termination. I hadn't noticed this before.

Chuck.

closed 2018-08-06 by stickbreaker