espressif/arduino-esp32 #1611: how to use block read in SDMMC？

wx85105157, 2018-07-08:
my tf card test on pc have 20MB/s read。
and I use the SDMMC_Text.ino to test，only 0.7MB/s。

By reading the source code，maybe file.read(buf, toRead); in SDMMC_Text.ino just use sdmmc_io_read_bytes()，not use sdmmc_io_read_blocks()

how to change the code to get fast read?

wx85105157, 2018-07-09:
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
SD_MMC Card Type: SDSC
SD_MMC Card Size: 1908MB
67108864 bytes written for 16832 ms
3.802281 Mbytes/S write
67108864 bytes read for 59760 ms
1.070950 Mbytes/S read
Total space: 1907MB
Used space: 64MB
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
SD_MMC Card Type: SDHC
SD_MMC Card Size: 3796MB
67108864 bytes written for 13809 ms
4.634659 Mbytes/S write
67108864 bytes read for 75129 ms
0.851868 Mbytes/S read
Total space: 3795MB
Used space: 64MB

ArTyGill, 2018-07-23:
The same problem, and could not solve. A file of 5 mb is written for 15 seconds.

RevoluPowered, 2019-06-07:
Having the same issue.

RevoluPowered, 2019-06-08:
I'm digging into this right now.

So I am using ESP32-IDF and having much better results using tests. 

**I have decreased the read time** 
1048576 bytes read for 1640 ms
1048576 bytes written for 6506 ms - **previous result was 7300 ms**
Total space: 14786MB
Used space: 3MB

FIrst changes I made were to this code as results from the ESP32 sd library were much better.
I believe the bottleneck is still in the fat library configured for the SD lib. 

```cpp
bool SDMMCFS::begin(const char * mountpoint, bool mode1bit)
{
    if(_card) {
        return true;
    }
    //mount
    sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
   // removed custom configuration
sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    
    // overrided for my settings from the esp32 test - these do not apply to all boards... you have been warned.
    host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;
    slot_config.width = 1;
    slot_config.flags = SDMMC_HOST_FLAG_1BIT;

    sdmmc_host_set_bus_ddr_mode(SDMMC_HOST_SLOT_1, true);
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = false,
        .max_files = 5,
        .allocation_unit_size = 64 * 1024
    };
     // static sdcard path made a difference in the path, I am unsure why but will know more later.
    esp_err_t ret = esp_vfs_fat_sdmmc_mount("/sdcard", &host, &slot_config, &mount_config, &_card);
    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            log_e("Failed to mount filesystem. If you want the card to be formatted, set format_if_mount_failed = true.");
        } else if (ret == ESP_ERR_INVALID_STATE) {
            _impl->mountpoint(mountpoint);
            log_w("SD Already mounted");
            return true;
        } else {
            log_e("Failed to initialize the card (%d). Make sure SD card lines have pull-up resistors in place.", ret);
        }
        _card = NULL;
        return false;
    }
    _impl->mountpoint(mountpoint);
    return true;
}

```

RevoluPowered, 2019-06-08:
I think the next issue lies in the SDMMC's IO functions.

In the ESP32-IDF library the test cases use a lot of very standard c code. Going to look into the implementation of file.write and file.read in the arduino-esp32 package as most of the latency seems to be in there.

The other thing, depending on the sd card, the ideal allocation size for reads vs writes in the esp32-idf exxamples seems to be 8192 bytes per block (for my sd card), so will also try and statically set that, then establish how to implement something which performs, but also runs on other sd cards optimally, perhaps might require adding a few functions, to set the 'block size' as they yeild different results and would be beneficial if you could configure them, depending on what you write to the flash.

RevoluPowered, 2019-06-08:
Using bare c problem is much less of an issue.

Results are as follows. Let me know your results if you can.

> 1048576 bytes read for 1611 ms
> 1048576 bytes written for 6488 ms
> Wrote 1048576 bytes (block size 16384) in 2235.549ms (0.447 MB/s)
> Read 1048576 bytes (block size 16384) in 274.089ms (3.648 MB/s)
> Total space: 14786MB
> Used space: 4MB
> 

Arduino ESP32 Script.
```cpp
/*
 * Connect the SD card to the following pins:
 *
 * SD Card | ESP32
 *    D2       12
 *    D3       13
 *    CMD      15
 *    VSS      GND
 *    VDD      3.3V
 *    CLK      14
 *    VSS      GND
 *    D0       2  (add 1K pull up after flashing)
 *    D1       4
 */

#include "FS.h"
#include "SD_MMC.h"
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <time.h>
#include <unistd.h>

#include "esp_log.h"
#include "esp_system.h"
#include "esp_vfs.h"
#include "esp_vfs_fat.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/sdmmc_host.h"
#include "driver/sdmmc_defs.h"
#include "sdmmc_cmd.h"
#include "diskio.h"
#include "ff.h"

bool test_rw_speed(const char* filename, void* buf, size_t buf_size, size_t file_size, bool is_write)
{
    const size_t buf_count = file_size / buf_size;

    FILE* f = fopen(filename, (is_write) ? "wb" : "rb");
    if(f == NULL) return false;

    struct timeval tv_start;
    gettimeofday(&tv_start, NULL);
    for (size_t n = 0; n < buf_count; ++n) {
        if (is_write) {
            write(fileno(f), buf, buf_size);
        } else {
            if (read(fileno(f), buf, buf_size) != buf_size) {
                printf("reading at n=%d, eof=%d", n, feof(f));
                return false;
            }
        }
    }

    struct timeval tv_end;
[... 193 lines cut ...]
    removeDir(SD_MMC, "/mydir");
    listDir(SD_MMC, "/", 2);
    writeFile(SD_MMC, "/hello.txt", "Hello ");
    appendFile(SD_MMC, "/hello.txt", "World!\n");
    readFile(SD_MMC, "/hello.txt");
    deleteFile(SD_MMC, "/foo.txt");
    renameFile(SD_MMC, "/hello.txt", "/foo.txt");
    readFile(SD_MMC, "/foo.txt");
    testFileIO(SD_MMC, "/test.txt");

    const size_t buf_size = 16 * 1024;
    uint32_t* buf = (uint32_t*) calloc(1, buf_size);
    esp_fill_random(buf, buf_size);
    const size_t file_size = 1 * 1024 * 1024;
    
    test_rw_speed("/sdcard/test.bin", buf, buf_size, file_size, true);
    test_rw_speed("/sdcard/test.bin", buf, buf_size, file_size, false);
    Serial.printf("Total space: %lluMB\n", SD_MMC.totalBytes() / (1024 * 1024));
    Serial.printf("Used space: %lluMB\n", SD_MMC.usedBytes() / (1024 * 1024));
}

void loop(){

}
```

wx85105157, 2019-06-11:
> Using bare c problem is much less of an issue.
> 
> Results are as follows. Let me know your results if you can.
> 
> **block size 16k**
> 
> > 1048576 bytes read for 1611 ms
> > 1048576 bytes written for 6488 ms <--- original library code
> > Wrote 1048576 bytes (block size 16384) in 2235.549ms (0.447 MB/s)
> > Read 1048576 bytes (block size 16384) in 274.089ms (3.648 MB/s) <--- pure c write/read fat32
> 
> **block size 8k, note im using 1 wire, non spi. (olimex-evb)**
> 
> > 1048576 bytes read for 1612 ms
> > 1048576 bytes written for 5588 ms
> > Wrote 1048576 bytes (block size 8192) in 655.021ms (1.527 MB/s)
> > Read 1048576 bytes (block size 8192) in 269.950ms (3.704 MB/s)
> 
> **block size 4k**
> 
> > 1048576 bytes read for 1623 ms
> > 1048576 bytes written for 6915 ms
> > Wrote 1048576 bytes (block size 4096) in 1263.701ms (0.791 MB/s)
> > Read 1048576 bytes (block size 4096) in 332.474ms (3.008 MB/s)
> 
> **block size 2k**
> 
> > 1048576 bytes read for 1622 ms
> > 1048576 bytes written for 7050 ms
> > Wrote 1048576 bytes (block size 2048) in 1681.870ms (0.595 MB/s)
> > Read 1048576 bytes (block size 2048) in 444.073ms (2.252 MB/s)
> 
> Arduino ESP32 Script.
> 
> ```c++
> /*
>  * Connect the SD card to the following pins:
>  *
>  * SD Card | ESP32
>  *    D2       12
>  *    D3       13
>  *    CMD      15
>  *    VSS      GND
>  *    VDD      3.3V
>  *    CLK      14
>  *    VSS      GND
>  *    D0       2  (add 1K pull up after flashing)
>  *    D1       4
>  */
> 
> #include "FS.h"
> #include "SD_MMC.h"
> #include <stdio.h>
> #include <stdlib.h>
> #include <string.h>
> #include <time.h>
> #include <unistd.h>
> 
> #include "esp_log.h"
> #include "esp_system.h"
> #include "esp_vfs.h"
> #include "esp_vfs_fat.h"
> #include "freertos/FreeRTOS.h"
> #include "freertos/task.h"
> #include "driver/sdmmc_host.h"
> #include "driver/sdmmc_defs.h"
> #include "sdmmc_cmd.h"
> #include "diskio.h"
> #include "ff.h"
> 
[... 219 lines cut ...]
>     readFile(SD_MMC, "/hello.txt");
>     deleteFile(SD_MMC, "/foo.txt");
>     renameFile(SD_MMC, "/hello.txt", "/foo.txt");
>     readFile(SD_MMC, "/foo.txt");
>     testFileIO(SD_MMC, "/test.txt");
> 
> 
>     const size_t buf_size = 16 * 1024;
>     uint32_t* buf = (uint32_t*) calloc(1, buf_size);
>     esp_fill_random(buf, buf_size);
>     const size_t file_size = 1 * 1024 * 1024;
>     
>     test_rw_speed("/sdcard/test.bin", buf, buf_size, file_size, true);
>     test_rw_speed("/sdcard/test.bin", buf, buf_size, file_size, false);
>     Serial.printf("Total space: %lluMB\n", SD_MMC.totalBytes() / (1024 * 1024));
>     Serial.printf("Used space: %lluMB\n", SD_MMC.usedBytes() / (1024 * 1024));
> }
> 
> void loop(){
> 
> }
> ```

very good. I'm outside home,when i back a week later,i will test your code.
can you test large file read?like 64MB or bigger?

RevoluPowered, 2019-06-11:
Thanks will try when I get some time free to do that will probably be tomorrow evening.

closed 2019-08-24 by stale[bot]