espressif/arduino-esp32 #1614: EEPROM does not compile

bdimir, 2018-07-09:
### Hardware:
Board:							ESP32 DevC Module
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
I'm trying to connect gravity TDS sensor on ADC14. The code does not compile. On Arduino UNO compiles fine.

### Sketch:
#include <EEPROM.h>
#include "GravityTDS.h"

#define TdsSensorPin A14 // gpio 13
GravityTDS gravityTds;

float temperature = 25,tdsValue = 0;

void setup()
{
    Serial.begin(115200);
    gravityTds.setPin(TdsSensorPin);
    gravityTds.setAref(5.0);  //reference voltage on ADC, default 5.0V on Arduino UNO
    gravityTds.setAdcRange(1024);  //1024 for 10bit ADC;4096 for 12bit ADC
    gravityTds.begin();  //initialization
}

void loop()
{
    //temperature = readTemperature();  //add your temperature sensor and read it
    gravityTds.setTemperature(temperature);  // set the temperature and execute temperature compensation
    gravityTds.update();  //sample and calculate 
    tdsValue = gravityTds.getTdsValue();  // then get the value
    Serial.print(tdsValue,0);
    Serial.println("ppm");
    delay(1000);
}

### Debug Messages:
```
Build options changed, rebuilding all
In file included from C:\Users\myuser\OneDrive\Documents\Arduino\libraries\GravityTDS-master\GravityTDS.cpp:17:0:

C:\Users\myuser\OneDrive\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM/EEPROM.h:91:5: error: 'float_t' does not name a type

     float_t readFloat(int address);

     ^

C:\Users\myuser\OneDrive\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM/EEPROM.h:92:5: error: 'double_t' does not name a type

     double_t readDouble(int address);

     ^

C:\Users\myuser\OneDrive\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM/EEPROM.h:95:5: error: 'String' does not name a type

     String readString(int address);

     ^

C:\Users\myuser\OneDrive\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM/EEPROM.h:110:36: error: 'float_t' has not been declared

     size_t writeFloat(int address, float_t value);

                                    ^

C:\Users\myuser\OneDrive\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM/EEPROM.h:111:37: error: 'double_t' has not been declared

     size_t writeDouble(int address, double_t value);

                                     ^

C:\Users\myuser\OneDrive\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM/EEPROM.h:114:37: error: 'String' has not been declared

     size_t writeString(int address, String value);

                                     ^

C:\Users\myuser\OneDrive\Documents\Arduino\libraries\GravityTDS-master\GravityTDS.cpp: In constructor 'GravityTDS::GravityTDS()':

C:\Users\myuser\OneDrive\Documents\Arduino\libraries\GravityTDS-master\GravityTDS.cpp:25:17: error: 'A1' was not declared in this scope

     this->pin = A1;

                 ^

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\myuser\OneDrive\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
Error compiling for board ESP32 Dev Module.
```

per1234, 2018-07-09:
There is a bug in the ESP32 EEPROM library. EEPROM.h does not `#include` Arduino.h. This is not noticeable if you only use the library from a sketch because the Arduino IDE automatically adds `#include <Arduino.h>` before the sketch's `#include <EEPROM.h>` but when the EEPROM library is used from another library (as in this case) this is not guaranteed. The solution is to simply move the `#include` directive [from EEPROM.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/EEPROM/EEPROM.cpp#L28) to EEPROM.h.

As for the A1 error, that's because ESP32 does not have an A1 pin. That's a problem with the GravityTDS, not the ESP32 core.

closed 2018-07-16 by me-no-dev

nguyenphuctinh, 2024-05-19:
I still have that problem :(